ARMmbed/mbed-os #10641: TARGET_PSA, TARGET_MBED_SPM and TARGET_TFM folders under COMPONENTS folder

devran01, 2019-05-23:
### Description

Should the folders TARGET_PSA, TARGET_MBED_SPM and TARGET_TFM under COMPONENTS folder be renamed to COMPONENT_* or should they be moved under targets folder?

@alzix's comment:
https://github.com/ARMmbed/mbed-os/pull/9221#issuecomment-495087104

### Issue request type

    [ x] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-08-13:
@bulislaw please review

bulislaw, 2019-08-14:
The original idea behind the components and features was different than the practice:
* Components was supposed to be for drivers for external HW modules
* Features was for software features
Of course the reality showed it's not as clear division and it wasn't understood very well.

My preference is that PSA would be in features as it's not an external HW component. As for TARGET_PSA, TARGET_MBED_SPM and TARGET_TFM it's how they are defined in targets.json, again it would make sense to have it as FEATURE based on the directory the code is in, but from the other point of view TARGET_PSA makes sense (not in targets directory though) as it's an inherent trait of HW architecture. For TARGET_MBED_SPM and TARGET_TFM as far as i understand it would make sense to be FEATURE as it's a software thing.

closed 2019-10-25 by ciarmcom