ARMmbed/mbed-os #10648: Should the online compiler be able to export projects on 64kb targets when using the 'bare-metal' profile?

hierophect, 2019-05-23:
### Description
The mbed online compiler does not allow for the compilation or export of any mbed-os project on a 64kb target, even if the `bare-metal` profile is used. These projects successfully compile when using the mbed-cli and mbed studio. Is this an oversight? 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-05-24:
> The mbed online compiler does not allow for the compilation or export

What do you get, an error ? Please provide details

hierophect, 2019-05-24:
Sorry, I misunderstood that the `bare-metal` profile was supposed be accessed via a compile macro as per `tickless` and other settings, based on a mixup in a previous conversation. However, I cannot alter the mbed_app.json file to use this profile in the way recommended for mbed CLI in the documentation. I've updated the issue title and description to represent this.

hierophect, 2019-05-24:
Issue was that the mbed library was very old (as per older issues), and the online compilers "update all" tool *doesn't update it* - you must click the mbed library and update it specifically. That's a bit confusing! Found issue after manually checking mbed version. 

Closing as I doubt the "update all" tool is really worth looking into. Sorry for all the misdirection here.

closed 2019-05-24 by hierophect

hierophect, 2019-05-24:
Very minor, but it might be nice to add the line:
 "Target not supported by MBED OS 5 without bare-metal profile. The bare-metal profile can be set in mbed_app.json"

To the default error produced by the online compiler when using OS5 with 64kb targets.

linlingao, 2019-05-24:
You're correct, "update all" doesn't work. It's a known issue.