ARMmbed/mbed-os #10650: M2351: Compile failed on GCC with atomic

ccli8, 2019-05-24:
### Description

Nuvoton's NUMAKER_PFM_M2351 target is M23 based. Currently, it only supports ARMC6 toolchain. Its GCC support has merged into master but is not enabled due to some reason. Testing on recent master and GCC, it compiles failed with:

<pre>

[Warning] @,: Compiler version mismatch: Have 7.3.1; expected version >= 6.0.0 and < 7.0.0
Building project mbed-os-example-blinky (NUMAKER_PFM_M2351, GCC_ARM)
Scan: mbed-os-example-blinky
Using ROM region application in this build.
  Region application: size 0x40000, offset 0x10040000
Using RAM region application_ram in this build.
  Region application_ram: size 0x10000, offset 0x30008000
Compile [  0.1%]: mbed_tz_context.c
Compile [  0.3%]: MCR20Drv.c
Compile [  0.4%]: at24mac.cpp
Compile [  0.5%]: NanostackRfPhyAtmel.cpp
Compile [  0.6%]: NanostackRfPhyMcr20a.cpp
Compile [  0.8%]: main.cpp
Compile [  0.9%]: rf_configuration.c
Compile [  1.0%]: fslittle_test.c
Compile [  1.1%]: at24mac_s2lp.cpp
Compile [  1.3%]: NanostackRfPhys2lp.cpp
Compile [  1.4%]: FlashIAPBlockDevice.cpp
[ERROR] C:\Users\ccli8\AppData\Local\Temp\ccq605pf.s: Assembler messages:
<b>
C:\Users\ccli8\AppData\Local\Temp\ccq605pf.s:898: Error: instruction not supported in Thumb16 mode -- `subs r0,r4,r1'
C:\Users\ccli8\AppData\Local\Temp\ccq605pf.s:961: Error: instruction not supported in Thumb16 mode -- `adds r1,r2,#1'
</b>

</pre>

It seems the error comes from **mbed-os/platform/internal/mbed_atomic_impl.h**. GCC doesn't translate in UAL syntax. If I roll back mbed-os to 5.12.4, it compiles OK.

#### Steps to reproduce

**GCC**:
GNU Tools for Arm Embedded Processors 7-2018-q2-update

**mbed-os-example-blinky**:
0b8957a (HEAD -> master, origin/master, origin/HEAD) Merge pull request #173 from jamesbeyond/test

**mbed-os**:
541dee4531 (HEAD -> master, origin/master, origin/HEAD) Merge pull request #10553 from korjaa/semihosting_docs

1. Enable NUMAKER_PFM_M2351 GCC support by add `GCC_ARM` in **mbed-os/targets/targets.json** for NUMAKER_PFM_M2351  target:

    <pre>
    "supported_toolchains": ["ARMC6", <b>"GCC_ARM"</b>],
    </pre>

1. Uner mbed-os-example-blinky root directory, compile by running:
    ```
    mbed compile -m NUMAKER_PFM_M2351 -t GCC_ARM
    ```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-05-24:
> Nuvoton's NUMAKER_PFM_M2351 target is M23 based. Currently, it only supports ARMC6 toolchain. Its GCC support has merged into master but is not enabled due to some reason. 

Can you find out why it is not enabled? I can see only ARMC6 enabled in targets.json - is this because we still are on GCC 6?

This is something to check once we switch to newer GCC and enable this target.

cc @kjbracey-arm

ccli8, 2019-05-24:
It is GCC toolchain issue. Per check on latest GCC 8-2018-q4, it still has errors in epilogue of NSC function.

cc @deepikabhavnani

0xc0170, 2019-05-24:
> cc @deepikabhavnani

@ARMmbed/mbed-os-core

kjbracey-arm, 2019-05-24:
>  I can see only ARMC6 enabled in targets.json - is this because we still are on GCC 6?

I assume so.

> "epilogue of NSC function"

Sorry, lost me there.

I'll look at the GCC/ARMv8-M atomic_impl.h issue.

PR #10705 (2019-06-09): Atomics: GCC fix for M23 (ARMv8-M baseline)

### Description

Add unified syntax directives to make the atomic assembler work when GCC is building for M23.

GCC actually uses unified syntax when compiling C code, but puts `.syntax divided` before each piece of inline assembly when targetting Thumb-1 type devices like M0 and M23 for backwards compatibility. We can overcome this with our own `.syntax unified`.  The command-line option `-masm-syntax-unified` intended to override this globally has been broken from GCC 6 to 8.0.

Fixes #10650

### Pull request type

    [X] Fix
    [ ] Refactor
    [... cut]
closed 2019-06-09 by 0xc0170