ARMmbed/mbed-os #10655: NRF52840 Bare Metal - Linker Script not found

DBS06, 2019-05-24:
### Description

Hi,
I already build successfully a custom bootloader with mbed-dev, but to maintain another repository is expensive in time and costs, I tried to build a bare-metal bootloader application with mbed-os 5.12.3, but the compilation fails with an error which says no linker script found.

```
mbed compile --profile .\profiles\bootloader.json -m NRF52840_DK --build BUILD_RELEASE -t GCC_ARM --app-config .\_____\_____\cfg\mbed_app.json
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Working path "P:\________" (program)
Building project _________-app (NRF52840_DK, GCC_ARM)
Scan: _________-app
Using ROM regions application, post_application in this build.
  Region application: size 0x6000, offset 0x0
  Region post_application: size 0xfa000, offset 0x6000
No Linker Script found

Could not compile for NRF52840_DK: No Linker Script found
[mbed] ERROR: "c:\program files (x86)\python37-32\python.exe" returned error.
       Code: 1
       Path: "P:\_____________"
       Command: "c:\program files (x86)\python37-32\python.exe -u P:\_____________\mbed-os\tools\make.py -t GCC_ARM -m NRF52840_DK --profile .\profiles\bootloader.json --source . --build BUILD_RELEASE --app-config .\_______\bootloader-app\cfg\mbed_app.json"
       Tip: You could retry the last command with "-v" flag for verbose output
---
[mbed] WARNING: Using Python 3 with Mbed OS 5.8 and earlier can cause errors with compiling, testing and exporting
```
If I remove
```
    "requires": [
        "bare-metal"
    ],
```
from my `mbed_app.json`, `mbed compile` finds the Linker Script for the `NRF52840`.
What could be the reason for this?
With a different target (i.e. EFM32PG12_STK3402) it works.
```
mbed compile --profile .\profiles\bootloader.json -m EFM32PG12_STK3402 --build BUILD_RELEASE -t GCC_ARM --app-config .\_______\bootloader-app\cfg\mbed_app.json --clean
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Working path "P:\_____________" (program)
Building project _________-app (EFM32PG12_STK3402, GCC_ARM)
Scan: ng-master-app
Using ROM regions application, post_application in this build.
  Region application: size 0x6000, offset 0x0
  Region post_application: size 0xfa000, offset 0x6000
Compile [  0.4%]: except.S
Compile [  0.9%]: mbed_tz_context.c
Compile [  1.3%]: mbed_fault_handler.c
Compile [  1.8%]: Ethernet.cpp
Compile [  2.2%]: CAN.cpp
...
```
I looked inside the target folders and configuration files for this two targets and compared them, but I didn't found obvious differences which could explain this.
I would be really grateful if somebody could look inside this issue!

### Issue request type

    [X] Question
    [ ] Enhancement
    [X] Bug

DBS06, 2019-05-28:
Any idea where I can start to search inside the mbed tools?

0xc0170, 2019-05-28:
build_api.py has this there

```
        # Change linker script if specified
        if linker_script is not None:
            resources.add_file_ref(FileType.LD_SCRIPT, linker_script, linker_script)
        if not resources.get_file_refs(FileType.LD_SCRIPT):
            notify.info("No Linker Script found")
            raise NotSupportedException("No Linker Script found")
```

I would check why there is no .ld file found in the resources object (just print it to check) . Check paths if your target is getting included (TARGET_ missing - label?).

It should be fairly simple to find out once you get resources object

DBS06, 2019-05-28:
I added
```
        # Change linker script if specified
        notify.info("linker_script: " + str(resources.get_file_refs(FileType.LD_SCRIPT)))
[...]
```
Result:
```
    mbed compile -m NG_NRF52840_BOOT_DK --app-config p:\ng-master-app\ng-apps\bootloader-app\cfg\mbed_app.json --profile p:\ng-master-app\profiles\release.json --build p:\ng-master-app\BUILD_RELEASE\bootloader-app\NG_NRF52840_BOOT_DK -vv 
	env: None
Building project ng-master-app (NG_NRF52840_BOOT_DK, GCC_ARM)
Scan: ng-master-app
Using ROM regions application, post_application in this build.
  Region application: size 0x6000, offset 0x0
  Region post_application: size 0xfa000, offset 0x6000
linker_script: []
No Linker Script found
```
without
```
    "requires": [
        "bare-metal"
    ],
```
Result:
```
    mbed compile -m NG_NRF52840_BOOT_DK --app-config p:\ng-master-app\ng-apps\bootloader-app\cfg\mbed_app.json --profile p:\ng-master-app\profiles\release.json --build p:\ng-master-app\BUILD_RELEASE\bootloader-app\NG_NRF52840_BOOT_DK -vv 
	env: None
Building project ng-master-app (NG_NRF52840_BOOT_DK, GCC_ARM)
Scan: ng-master-app
Using ROM regions application, post_application in this build.
  Region application: size 0x6000, offset 0x0
  Region post_application: size 0xfa000, offset 0x6000
linker_script: [FileRef(name='mbed-os\\targets\\TARGET_NORDIC\\TARGET_NRF5x\\TARGET_NRF52\\TARGET_MCU_NRF52840\\device\\TOOLCHAIN_GCC_ARM\\NRF52840.ld', path='.\\mbed-os\\targets\\TARGET_NORDIC\\TARGET_NRF5x\\TARGET_NRF52\\TARGET_MCU_NRF52840\\device\\TOOLCHAIN_GCC_ARM\\NRF52840.ld')]
Macros: -DTARGET_NAME=NG_NRF52840_BOOT_DK -DDEVICE_I2C=1 -DTARGET_SDK_14_2 -DMBED_BUILD_TIMESTAMP=1559045413.0244005 -DTARGET_FF_ARDUINO -DDEVICE_QSPI=1 -DDEVICE_SPI_ASYNCH=1 -DBOARD_PCA10056 -DTARGET_RELEASE -DMBED_MPU_CUSTOM -DMBED_TICKLESS -DTARGET_NRF52840 -DCOMPONENT_QSPIF=1 -DTARGET_MCU_NRF52840 -DDEVICE_PWMOUT=1 -DTARGET_NORDIC -DTARGET_NG_NRF52840_BOOT_DK -DCOMPONENT_NSPE=1 -DTARGET_NRF52840_DK -DTARGET_LIKE_CORTEX_M4 -D__FPU_PRESENT=1 -DARM_MATH_CM4 -DDEVICE_SYSTICK_CLK_OFF_DURING_SLEEP=1 -DDEVICE_INTERRUPTIN=1 -DWSF_MAX_HANDLERS=10 -DDEVICE_PORTINOUT=1 -DTARGET_SOFTDEVICE_NONE -D__CORTEX_M4 -DGPIOTE_CONFIG_NUM_OF_LOW_POWER_EVENTS=8 -DCOMPONENT_PSA_SRV_EMUL=1 -DTARGET_CORTEX_M -DDEVICE_FLASH=1 -DNRF52840_XXAA -DDEVICE_PORTIN=1 -DTOOLCHAIN_GCC -DDEVICE_SERIAL_FC=1 -DDEVICE_I2C_ASYNCH=1 -DDEVICE_ANALOGIN=1 -DCONFIG_GPIO_AS_PINRESET -DDEVICE_ITM=1 -DCMSIS_VECTAB_VIRTUAL -DDEVICE_SERIAL=1 -DCMSIS_VECTAB_VIRTUAL_HEADER_FILE="cmsis_nvic.h" -D__MBED__=1 -DTOOLCHAIN_GCC_ARM -DTARGET_M4 -DDEVICE_SPI=1 -D__CMSIS_RTOS -DDEVICE_PORTOUT=1 -D__MBED_CMSIS_RTOS_CM -DTARGET_NRF5x -DTARGET_LIKE_MBED -DTARGET_CORTEX -DCOMPONENT_PSA_SRV_IMPL=1 -DNRF52_ERRATA_20 -DDEVICE_SERIAL_ASYNCH=1 -DTARGET_RTOS_M4_M7 -DTARGET_NRF52 -DDEVICE_SLEEP=1 -DDEVICE_LPTICKER=1 -DDEVICE_USTICKER=1 -DDEVICE_TRNG=1
Compile [  0.2%]: mbed_tz_context.c
```

0xc0170, 2019-05-28:
`LINKER_EXT` is used to search for linker script file. `mbed_toolchain.py` - line 714 should be it - where it gets linker script file and use it for linking.

There might a bug in bare metal configuration - your path looks fine : `mbed-os\\targets\\TARGET_NORDIC\\TARGET_NRF5x\\TARGET_NRF52` , cant see why it would remove this target path for bare-metal (it requires platform + drivers, targets should be there by default).

requires is checked in `add_config_files` method inside `config/__init__.py` file. Can you check resource objects prior invoking config what is there ? Somewhere your target paths are being removed.

DBS06, 2019-05-28:
I added a test repo at https://github.com/DBS06/bare_metal_test
and write also the mbed commands which I used to test into readme.

DBS06, 2019-05-28:
@0xc0170 I think I know why this happens, because `'.\\mbed-os\\targets\\TARGET_NORDIC\\TARGET_NRF5x\\TARGET_NRF52'` is inside the "excluded_lib_paths" from `mbed-os\tools\resources\__init__.py` method: `def _include_file(self, ref)`

see:
[image]

0xc0170, 2019-05-28:
This is definitely a bug . How did it get to exclude list then?

DBS06, 2019-05-28:
@0xc0170 I just made a quick fix inside `mbed-os\tools\resources\__init__.py` method: `def _include_file(self, ref)` just to verify if this would be the error/bug:
```
[...]
        _, path = ref
        cur_dir = dirname(path)
        included_lib_paths = [dirname(e.path) for e in self._libs_filtered]
        excluded_lib_paths = [dirname(e.path) for e in self._excluded_libs]

        nrf52_tool_path = '.\\mbed-os\\targets\\TARGET_NORDIC\\TARGET_NRF5x\\TARGET_NRF52'
        if nrf52_tool_path in excluded_lib_paths:
            excluded_lib_paths.remove(nrf52_tool_path)
[...]
```
But it is definitely no solution for production usage! 😉

0xc0170, 2019-05-29:
cc @ARMmbed/mbed-os-tools

DBS06, 2019-06-05:
@ARMmbed/mbed-os-tools @0xc0170 
I looked again in this issue, because it seems mbed-os-tools-team does not react.
I think the Linker Script shouldn't be added to `all_library_refs` in function `mbed-os\tools\resources\__init__.py`
```
def filter_by_libraries(self, libraries_included):
        """
        Call after completely done scanning to filter resources based on
        libraries
        """
        self._libs_filtered = set(libraries_included)
        all_library_refs = set(
            ref for ref in self._file_refs[FileType.JSON]
            if ref.name.endswith(MBED_LIB_FILENAME)
        )
        self._excluded_libs = all_library_refs - self._libs_filtered
        if self._collect_ignores:
            self.ignored_dirs += [
                dirname(n) or "." for n, _ in self._excluded_libs
            ]
```
Because if I build for the EFM32PG12 the Linker script isn't added.

0xc0170, 2019-06-05:
@DBS06 tools team was moved recently. To progress here, can you send PR with proposed fix? I'll make sure tools team review , me as well.

mark-edgeworth, 2019-06-05:
Yes, thanks for that @0xc0170 :-) We're still coming up-to-speed with all this...

DBS06, 2019-06-05:
@0xc0170 I would like to, but actually I "just" know whats happening, but not why... Right now I am looking now into this issue, if I am successful, I will create a PR of course ;-)

DBS06, 2019-06-07:
Mhm I still didn't found the reason and I have no further ideas where to search deeper to find the bug...

statropy, 2019-07-11:
@DBS06 You need to add "nordic" to the requires list in the mbed_app.json file. Thanks to the debug info you included, I was able to track down why bare-metal will work with other targets but not the NRF52840. I tried the NUCLEO-F103RB board and that built without issue. The difference is the TARGET_NRF52 is in the excluded list because there is a mbed_lib.json file in that directory. In the while loop of _include_file(), the linker file will eventually have it path shortened to the TARGET_NRF52 directory and return false. With the Nucleo board, there is no mbed_lib.json file in any of the target subdirectories, so the while loop completes and and returns true.
I also had to remove CRYPTOCELL310 as that needed to include "platform_alt.h".

`{
    "requires": ["bare-metal", "nordic"],
    "target_overrides": {
        "NRF52840_DK": {
            "target.features_remove": ["CRYPTOCELL310"]
        }
    }
}`

DBS06, 2019-07-11:
@statropy nice done! to add `nordic` to the `requires` is a nice and suitable solution!
Many thanks for looking into this issue! Maybe this should be documented for the bare-metal-build.
BTW: Should I close this issue?

0xc0170, 2019-08-13:
@statropy Nice find. I don't understand why though it is excluded

cc @ARMmbed/mbed-os-tools 

I'll close this as resolved, but still would like to  understand why one board works and other not. might be good to track it in the new issue with details.

0xc0170, 2019-09-05:
Still valid and should be fixed. Nordic targets should not define mbed_lib.json file, but use target configuration.

0xc0170, 2019-09-05:
Fix: https://github.com/ARMmbed/mbed-os/pull/11418

PR #11418 (2019-09-06): NRF52: fix config
### Description

remove lib config and use target configuration instead. To avoid duplication of symbols, etc.

Fixes #10655

It could be simplified but I noticed not all targets from NRF52 contain  overrides neither have one common parent (two MCU), therefore I only moved the config already there to the target config. No other change.

### Pull request type

    [X] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

### Reviewers

@SeppoTakalo @evedon 

### Release Notes
commit 530f1e9a (2019-09-06): NRF52: fix config

remove lib config and use target configuration instead. To avoid duplication of symbols, etc.

Fixes #10655
commit c51d452d (2019-09-13): NRF52: fix config

remove lib config and use target configuration instead. To avoid duplication of symbols, etc.

Fixes #10655
closed 2019-09-06 by 0xc0170