ARMmbed/mbed-os #10658: Cellular: Multiple URCs issue

mudassar-ublox, 2019-05-24:
When setting different RAT of the modem, the ublox modem needs to deregister from the network. This is done by sending AT+COPS=2 which in response generates the following URCs:

+CREG: 0
+CGREG: 0
+CGEV: ME DETACH

However, the `ATHandler` processes only the first URC `(+CREG)` and the other URCs are missed consequently resulting in `at_overflow` and return error -3012 `(_last_err)`. 

[00042148][INFO][CELL]: AT TX ( 9): AT+COPS=2
[00042149][INFO][CELL]: AT TX ( 1): <cr>
[00043596][DBG ][CELL]: AT OoB readable 1, len 0
[00043596][INFO][CELL]: AT RX ( 6): <cr><ln>OK<cr><ln>
[00043597][INFO][CELL]: AT RX ( 5): <cr><ln>+CR
[00043597][INFO][CELL]: AT RX ( 5): EG: 0
[00043598][INFO][CELL]: AT RX ( 5): <cr><ln><cr><ln>+
[00043660][INFO][CELL]: AT RX ( 2): CG
[00043715][DBG ][CELL]: +CREG: NotRegistered, LAC -1, cell -1, RAT unknown
[00043808][INFO][CELL]: AT RX (13): REG: 0<cr><ln><cr><ln>+CG
[00043887][INFO][CELL]: AT RX (13): EV: NW DETACH
[00043954][ERR ][CELL]: AT overflow
[00044006][INFO][CELL]: AT RX ( 2): <cr><ln>
[00044067][INFO][CELL]: AT ERR (29): +: EV: NW DETACH<00><00><00><1D><00><00><00><00><00><00><00><00><00>
[00044568][DBG ][CELL]: AT error -3012
[00044568][DBG ][CELL]: AT OoB done

Is there any solution to parse multiple URCs. 
### Description

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

mudassar-ublox, 2019-05-24:
Not sure but may be this is related to this issue [#10549](https://github.com/ARMmbed/mbed-os/issues/10549)

mudassar-ublox, 2019-05-28:
@fahim-ublox @RobMeades

ccchang12, 2019-05-28:
The default size of serial RX buf is 256, which may be not enough for different module. You can try to add below data to mbed_app.json to enlarge it.

"drivers.uart-serial-rxbuf-size": 1024,

RobMeades, 2019-05-28:
Have you tried removing or compiling-out the debug prints?  There's a good chance that printing out debug in the URC is the cause of the overflow.  URCs are time sensitive.

mudassar-ublox, 2019-05-29:
By disabling debug traces still getting same result, also increased "drivers.uart-serial-rxbuf-size": 1024, as @ccchang12  mention but no use.

RobMeades, 2019-05-29:
There's something very odd going on then as you're plainly not getting 1024 characters back from the modem in just those few URCs, not even 256 in fact.  Are you sure that the `+CGEV` URC handler is constructed such that it picks up the expected content, rather than waiting for something which does not appear?  Is it possible that there is character loss so that the `\r\n` sequence which terminates the URC has gone missing?

mirelachirica, 2019-06-12:
Change from this PR: https://github.com/ARMmbed/mbed-os/pull/10812#pullrequestreview-248631139 should be tried if it helps this case.

AriParkkila, 2019-07-22:
@mudassar-ublox Is this still reproducing after PR #10812 was merged.

mudassar-ublox, 2019-07-22:
Yes, Issue resolved. Closing this issue. Thanks

closed 2019-07-22 by mudassar-ublox