ARMmbed/mbed-os #10662: Argument list too long

janhaa, 2019-05-24:
I consistently get the following when trying to use mbed through platform.io.
I suppose this is a problem with mbed-os rather than platform.io.

```
Processing nrf52_dk (platform: nordicnrf52; board: nrf52_dk; framework: mbed)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nordicnrf52/nrf52_dk.html
PLATFORM: Nordic nRF52 > Nordic nRF52-DK
HARDWARE: NRF52832 64MHz 64KB RAM (512KB Flash)
DEBUG: CURRENT(cmsis-dap) ON-BOARD(cmsis-dap, jlink) EXTERNAL(blackmagic, stlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 5 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <mbed-os>
Compiling .pioenvs/nrf52_dk/src/main.o
Compiling .pioenvs/nrf52_dk/libd6b/mbed-os_ID2499/TESTS/events/queue/main.o
Compiling .pioenvs/nrf52_dk/libd6b/mbed-os_ID2499/TESTS/integration/basic/main.o
*** [.pioenvs/nrf52_dk/src/main.o] Argument list too long
*** [.pioenvs/nrf52_dk/libd6b/mbed-os_ID2499/TESTS/events/queue/main.o] Argument list too long
Compiling .pioenvs/nrf52_dk/libd6b/mbed-os_ID2499/TESTS/mbed_drivers/c_strings/main.o
*** [.pioenvs/nrf52_dk/libd6b/mbed-os_ID2499/TESTS/integration/basic/main.o] Argument list too long
*** [.pioenvs/nrf52_dk/libd6b/mbed-os_ID2499/TESTS/mbed_drivers/c_strings/main.o] Argument list too long
```

gcc -v:
```
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/7/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 7.4.0-1ubuntu1~18.04' --with-bugurl=file:///usr/share/doc/gcc-7/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-7 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 7.4.0 (Ubuntu 7.4.0-1ubuntu1~18.04) 
```

0xc0170, 2019-08-13:
I am not certain where this `Argument list too long` comes from. Quick search : remove command ? Is this coming from mbed-cli ? I don't think it does as haven't seen the report like this one. But could.

@janhaa Has there been any update for this issue?

janhaa, 2019-08-14:
@0xc0170 Hi,
I have not persued this approach further but rather used the nrf5 SDK directly.

0xc0170, 2019-08-23:
I was able to find similar issue reported already. This is known issue for mbed-cli.

I'll close this as duplicate, https://github.com/ARMmbed/mbed-os/issues/9640

closed 2019-08-23 by 0xc0170