ARMmbed/mbed-os #10664: Better MemoryPool Implementation

maxgerhardt, 2019-05-24:
### Description

* target: `NUCLEO_L476RG`
* toolchain: GCC_ARM 8
* mbed-os: newest 5.12

When using the `MemoryPool` object to have a statically allocated pool of memory of objects of type `T`, the user has to make sure to initialize the objects acquired via a `alloc()` call. 

However, in the use case of dynamically allocating a `Queue` object, this isn't even possible because the initialization is done in the constructor code and is not accessible from the outside. The `MemoryPool` only allocates memory and casts it to the specific type, but it doesn't call the default constructor on the `T` object.

https://github.com/ARMmbed/mbed-os/blob/73f096399b4cda1f780b140c87afad9446047432/rtos/Queue.h#L66-L75

This means that a code like this would create an uninitialized queue for which sending and receiving fails. 

```cpp
#include <mbed.h>
#include <rtos.h>
#include <Queue.h>

Serial pc(USBTX, USBRX, "debug", 921600);

MemoryPool<Queue<bool, 1>, 3> memPool;
Thread ackThread;

static bool goodVal = true;
static bool badVal = false;

void ack_thread(Queue<bool, 1>* ackQueue) {
    printf("Ack thread\n");
    wait_ms(3000);
    printf("sending ACK\n");
    ackQueue->put(&goodVal);
}

int main() {
    printf("Firmware start\n");
    auto queue = memPool.alloc();
    ackThread.start(callback(ack_thread, queue));
    printf("Waiting for queue element\n");
    auto ret = queue->get(10*1000UL);
    if (ret.status == osEventMessage && ret.value.p != nullptr){
        bool r = *(bool*)ret.value.p;
        printf("Received value: %d\n", (int) r);
    } else {
        printf("Timeout or error while waiting for ACK. %08x\n",
                (int) ret.status);
    }
    memPool.free(queue);
}
```

Fails with 

```
Firmware start
Waiting for queue element

++ MbedOS Error Info ++
Error Status: 0x80010138 Code: 312 Module: 1
Error Message: MessageQueue: 0x20001754, Parameter error
Location: 0x8002553
Error Value: 0x20001754
Current Thread: main  Id: 0x20000E44 Entry: 0x800234F StackSize: 0x1000 StackMem: 0x20001980 SP: 0x10007F38 
For more info, visit: https://mbed.com/s/error?error=0x80010138&tgt=NUCLEO_L476RG
-- MbedOS Error Info --
```

When replacing the `memPool.alloc()` call with `new Queue<bool, 1>()` and `memPool.free(queue);` with `delete queue;`, the firmware works as expected.

```
Firmware start
Waiting for queue element
Ack thread
sending ACK
Received value: 1
```

This introduces an arbitrary restriction on the `MemoryPool`. Can't we not call the default, parameter-less `T` constructor on the object? Or use `new` and `delete` in the allocation and free routines to do it for us, but still enjoy being within a bounded memory region? 

### Issue request type
    [ ] Question
    [X] Enhancement
    [ ] Bug

kjbracey-arm, 2019-05-29:
This was discussed in #9435 - I suggest you read the discussion there, although it did get a bit deep

We could still revisit this, especially as we're activating C++14, but the point remains that it would need to be an extension, rather than a change to existing behaviour.

With the current API, you can use placement `new` in the blocks returned by the allocator.

maxgerhardt, 2019-05-29:
Closing since it's a duplicate then.

closed 2019-05-29 by maxgerhardt