ARMmbed/mbed-os #10667: Maxim 32630 Feather - MTU Cannot Be Changed

nick-atmosphereiot, 2019-05-24:
### Description
Using the Max32630 Feather board, the MTU is always set to 23 with a maximum payload size of 20. 

I am unable to negotiate the MTU with the device, as it seems that the code just isn't there to handle it.

Can anyone provide an example of changing the MTU with exactLE? There are some defines and functions related to MTU, but none of them do anything.

-Nick

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ X] Bug

0xc0170, 2019-05-27:
cc @ARMmbed/team-maximintegrated

0xc0170, 2019-06-04:
cc @bentcooke

nick-atmosphereiot, 2019-09-03:
Any update on this?

0xc0170, 2019-09-04:
@bentcooke Is there anyone to help here? or @ARMmbed/mbed-os-pan ?

This is related to maxim implementation and their provided library so might be the best if @ARMmbed/team-maximintegrated answers.

pan-, 2019-09-04:
@nick-atmosphereiot As far as I know, the CC2564 on this chip only supports 23 bytes MTU.

nick-atmosphereiot, 2019-09-04:
@pan- Yeah that's what I'm seeing here as well:
http://e2e.ti.com/support/wireless-connectivity/bluetooth/f/538/p/820736/3041343#3041343?jktype=e2e

I guess we will have to implement a chunking mechanism on our end or something like that then. Thanks.

Nick

0xc0170, 2019-09-05:
@nick-atmosphereiot Question answered, can be closed?

adbridge, 2020-10-15:
Assuming this has now been addressed and thus closing.

closed 2020-10-15 by adbridge