ARMmbed/mbed-os #10669: BLE power consumption using CORDIO stack too high

sagrebin, 2019-05-24:
### Description
I've cloned the [mbed-os-example-ble](https://github.com/ARMmbed/mbed-os-example-ble) repo and removed the LED blinking code and the print statements. When I run the application on my dev board and use nordics power profiler I get an average of 1mA when advertising (advertising interval: 2 sec).
I did the same with the with the deprecated sample [BLE_BatteryLevel](https://github.com/ARMmbed/mbed-os-example-ble/tree/master/deprecated/BLE_BatteryLevel) which uses nordics soft device and I get around 20uA when advertising.
I use a custom dev board with the ublox NINA-B112 module, there are no other components on the board which could consume power.

- Target: NRF52_DK
- Toolchan: GCC_ARM
- Tools: mbed-cli 1.10.0
- sha: a4738fa9a8

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

40Grit, 2019-05-25:
If debugging with an svg loaded,

are you able to confirm which peripherals are enabled and running between the two examples?

Also would be interesting to see the clock configurations between the two.

40Grit, 2019-05-25:
Mistyped .svd above.

yogeshk19, 2019-06-02:
@40Grit  We are seeing similar power consumption and at times even higher than 1ma during advertisements. One of the other issues, had listed out disabling the cryptocell and here is what the mbed_app.json looks like. Ours is a custom board with NRF52840 BLE module. We are using Cordio API without the Nordic Softdevice.

`
{
    "target_overrides": {
        "NRF52840_DK": {
            "target.features_add": ["BLE"],
            "target.extra_labels_add": ["CORDIO", "CORDIO_LL", "SOFTDEVICE_NONE", "NORDIC_CORDIO"],
            "target.features_remove": ["CRYPTOCELL310"],
            "target.macros_remove": ["MBEDTLS_CONFIG_HW_SUPPORT"],
            "target.extra_labels_remove": ["SOFTDEVICE_COMMON", "SOFTDEVICE_S140_FULL", "NORDIC_SOFTDEVICE"],
            "platform.stdio-baud-rate": 115200,
            "platform.default-serial-baud-rate": 115200,
            "target.console-uart-flow-control": null
        }
    }
}
`
Thanks,
Yogesh

40Grit, 2019-06-03:
My questions above are wondering what soft device does with the clocks/peripherals/radio settings in order have significantly lower power consumption than cordio stack.

0xc0170, 2019-06-03:
> My questions above are wondering what soft device does with the clocks/peripherals/radio settings in order have significantly lower power consumption than cordio stack.

cc @ARMmbed/mbed-os-pan

yogeshk19, 2019-06-09:
@40Grit Sorry I haven't had a chance to dump the peripheral registers. I am planning on re-running my power draw tests again after the next mbed-os release which is supposed to have updates to hci driver i believe. I am not planning on using the softdevice at this point, only the Cordio API as is.

sagrebin, 2019-07-17:
@40Grit I was hoping that with the arrival of nordics sdk 15 it would be fixed by unfortunately its not. I've dumped the registers:
cordio: [https://pastebin.com/i1q4T08m](https://pastebin.com/i1q4T08m)
nordic SD: [https://pastebin.com/47XEkrBh](https://pastebin.com/47XEkrBh)

pan-, 2019-07-17:
From the diff it looks like the LFCLK is synthesized from the HFCLK when Cordio is used. That might be one of the cause of the high power consumption.

sagrebin, 2019-07-17:
Indeed, for some reason the NRF_LF_CLK_SRC is set to CLOCK_LFCLKSRC_SRC_Synth in NRFCordioHCIDriver.cpp
`#if MBED_CONF_NORDIC_NRF_LF_CLOCK_SRC == NRF_LF_SRC_SYNTH
    #define NRF_LF_CLK_SRC CLOCK_LFCLKSRC_SRC_Synth
`
But what I don't get is why MBED_CONF_NORDIC_NRF_LF_CLOCK_SRC is set to NRF_LF_SRC_SYNTH. Every definition I can find is set to NRF_LF_SRC_XTAL

40Grit, 2019-07-18:
@aglass0fmilk

jrobeson, 2019-10-08:
@sagrebin : so you see `NRF_LF_CLK_SRC` set to `NRF_LF_SRC_XTAL` in `mbed_config.h` ?

kleunen, 2019-10-15:
I have seen the same issue. I am not sure if it is the crystal configuration, I tried changing back the crystal to XTAL, in the initComplete of the BLE stack. But this does not seem to help.

bsturgess, 2019-10-26:
Same issue here too. watching...

40Grit, 2019-10-26:
The best way to get a picture of what os going on is to dump the peripheral and cpu registers between the two examples you are comparing power for.

Then do a diff of the dumps.

It needs to be considered as well that if a gpio is left in an output state it may result in leaks to other unpowered components on a board.

Additionally the pull resistors internal to the gpios for this part are pretty strong. Deffinitely confirm the state of your gpios.

bsturgess, 2019-10-26:
I've checked the consumption before the ble.init and its 3ua, so roughly what it should be.

I can also see that the device just doesn't deep sleep after ble init, see logs below, 0 time in deepsleep :( ...

Results from logger
Uptime: 2001 Sleep time: 1930 Deep Sleep: 0 Uptime: 4001 Sleep time: 3864 Deep Sleep: 0 Uptime: 6001 Sleep time: 5798 Deep Sleep: 0 Uptime: 8001 Sleep time: 7733 Deep Sleep: 0 Uptime: 10001 Sleep time: 9667 Deep Sleep: 0 Uptime: 12001 Sleep time: 11601 Deep Sleep: 0 Uptime: 14001 Sleep time: 13535 Deep Sleep: 0

bsturgess, 2019-10-26:
What is the code for dumping the registers? Thanks

40Grit, 2019-10-26:
Should be able to do it through gdb command line or pyocd. J-link commander maybe if using j-link

40Grit, 2019-10-26:
you will need to load the .svd file for your target mcu

bsturgess, 2019-10-27:
Is this being looked at internally by the ARM team? If so i'll just keep on using SoftDevice until resolved

40Grit, 2019-10-27:
Latest news is that the BLE PAN team is busy.
Also the Cordio team was let go in April and became PacketCraft Inc.

Embedded Planet, my employer, will probably be looking into this over the next couple months but no promises.

Not sure how Cordio governance is going to go moving forward. PacketCraft is probably the new maintainer.

40Grit, 2019-10-27:
Also softdevice has its own RTOS internally. I have heard and experienced that it doesnt play very nice with external RTOS's.

jrobeson, 2019-10-27:
and there's no history at their github repo https://github.com/packetcraft-inc/cordio.

pentabarf, 2019-11-27:
> I've checked the consumption before the ble.init and its 3ua, so roughly what it should be.
>
> I can also see that the device just doesn't deep sleep after ble init, see logs below, 0 time in [...]

I noticed that there is a deep sleep lock in the Cordio idle_hook. (I am using nRF52)
And it gets called when there is nothing to do i guess. Possibly this has an effect on deep sleep
See in mbed-os/targets/TARGET_NORDIC/TARGET_NORDIC_CORDIO/TARGET_NRF5x/NRFCordioHCIDriver.cpp

`static void idle_hook(void)`
`{`
`[...]`
`    // critical section to complete sleep with locked deepsleep`
`    core_util_critical_section_enter();`
`    sleep_manager_lock_deep_sleep();`
`    sleep();`
`    sleep_manager_unlock_deep_sleep();`
`    core_util_critical_section_exit();`
`}`

kleunen, 2020-01-12:
@pentabarf  You are right

It actually says in the documentation:
https://github.com/ARMmbed/mbed-os/blob/master/features/FEATURE_BLE/targets/TARGET_NORDIC/TARGET_NORDIC_CORDIO/TARGET_NRF5x/README.md

Resources used:
HF clock is enabled

That is just complete BS, this really removes LE from BLE.

yogeshk19, 2020-01-16:
@40Grit, @pan-  I was wondering based on @pentabarf  and @kleunen  comments, is there any progress on helping improve the power consumption? Currently our application we have built which advertises 90% of the time and 10% time stays connected, the avg. power consumption is 200 uA when advertising and 300 uA when connected using the Cordio API. When using the softdevice the power draw is around 20 uA when advertising and 9-12 uA when connected. This clearly indicates the Cordio platform is consuming more power than when just using Nordic's softdevice to build a similar application. 

As mentioned by you earlier that Cordio is no longer the priority for ARM and a different company has taken over the development effort, but I am wondering when companies like mine migrated to Cordio platform assuming that this was the better platform to be on since ARM supported this new platform for BLE and that something so important as BLE Power consumption needs to be handled, what is the escalation mechanism to get  either ARM or the new company to address this issue? This issue has been lingering from 9 months now, with no clear priority or resolution in sight.

Sorry just venting out!

Thanks,
Yogesh

0xc0170, 2020-01-16:
Thanks @yogeshk19 for the feedback. I'll share this with relevant people.

40Grit, 2020-01-16:
@yogeshk19 Contact Packetcraft.
I recently talked with them about the Mbed Cordio stack integration at AIoT but forgot to mention this issue. I think they could be helpful too.

info@packercraft.com
https://www.packetcraft.com/

40Grit, 2020-01-16:
@AGlass0fMilk I know the magic stick has a monstrous battery but if you get a moment maybe take a look further into this?

AGlass0fMilk, 2020-01-16:
@yogeshk19 Have you verified the connection and advertising parameters are the same in both applications? How are you measuring this current draw? The power profiler kit from Nordic would be ideal in this case because it can show the large spikes in current during transmission and also the low quiescent current draw.

I'll hopefully be able to look into this when I'm working more with BLE in the next few weeks.

yogeshk19, 2020-01-18:
@AGlass0fMilk  Yes the device primarily connects to an IPhone or Android phone and we use the appropriate connection parameters. We didn't use the Nordic power profiler kit, we used other means to determine the average power draw.  I have to say however the initial power draw was way higher with Cordio, but with the latest MBED OS and other software and hardware improvements on our side has reduced the power draw from 1.5 mA to 200  uA while advertising and connected. But as suggested by @pentabarf  if the MBED OS is not entering deep sleep as expected, I am sure when the system is idle it will continue to draw power. Is this something that can be investigated and a fix for ensuring that the device enters sleep mode when idle. This seems to work effectively when using Nordics Softdevice and hence the much lower power draw?

Thanks,
Yogesh

40Grit, 2020-01-18:
We have struggled with NRF and deep sleep as well even without BLE being used.

Since the Port of the nrf sdk is only maintained by ARM, partners and community, there are some bugs we are all still working out. As we use different peripherals and power schemes they pop up.

If not entering deep sleep then a peripheral driver is probably setting the sleep lock.

we've found some little things like setting NFC pins to gpio save a couple uA. Might be circuit dependant though.

pentabarf, 2020-01-22:
In the meantime... is there a simple way to switch back to softdevice? just changing the flags in mbed_lib.json doesn't seem to do the trick.
Is the ble codebase still supporting softdevice?

kleunen, 2020-01-22:
I failed to do so, i had to go a couple of version back to get a working softdevice BLE again.

kleunen, 2020-02-19:
I switched back to mbed os version 5.11.5, it has the nordic softdevice. The power consumption for this version is still ok. It is rather annoying, because sometimes the mbed version gets updated to the latest version somehow, resulting in much higher power consumption.

bsturgess, 2020-02-19:
Totally agree, 5.11 is the only version of mbed that can be used if you
want any sensible power consumption.

Very frustrating

CodingGhost, 2020-02-27:
any progress here?
Is systemOFF mode affected by this issue?
I just ported our Software to cordio and I am seeing 37uA in systemOFF mode, it was at about 0.8 before.
can cordio be the cause of this?

AGlass0fMilk, 2020-02-27:
> any progress here?
> Is systemOFF mode affected by this issue?
> I just ported our Software to cordio and I am seeing 37uA in systemOFF mode, it was at about 0.8 before.
> can cordio be the cause of this?

I would check the peripherals that are still active in system off (RTC possibly) and see if they’re still enabled. Also check RAM retention settings. Lastly, make sure there weren’t any changes in how your GPIO are set during system off.

It’s possible Cordio could be changing one of these settings whereas the softdevice didn’t.

CodingGhost, 2020-03-02:
I tried to disable everything I know 
`    NRF_RTC0->TASKS_STOP = 1;
    NRF_RTC1->TASKS_STOP = 1;
    NRF_CRYPTOCELL->ENABLE = 0;
    NRF_RADIO->TASKS_STOP = 1; //!< Peripheral power control. 
    NRF_RADIO->POWER = 0; //!< Peripheral power control. 
    NRF_UART0->ENABLE = 0; //!< Peripheral power control. 
    NRF_SPI0->ENABLE = 0; //!< Peripheral power control. 
    NRF_SPIS1->ENABLE = 0; //!< Peripheral power control. 
    NRF_TIMER0->TASKS_STOP = 1; //!< Peripheral power control. 
    NRF_TIMER1->TASKS_STOP = 1; //!< Peripheral power control.  
    NRF_RNG->TASKS_STOP = 1; //!< Peripheral power control. /
    NRF_LPCOMP->TASKS_STOP = 1; //!< Peripheral power control. 
    NRF_CLOCK->TASKS_HFCLKSTART = 0;
    NRF_CLOCK->TASKS_HFCLKSTOP = 1;`

but this still gives me 20uA instead of 1.

anything I missed?

CodingGhost, 2020-03-02:
so, after some testing, I could reduce the problem to following lines:

`    BLE &ble = BLE::Instance(BLE::DEFAULT_INSTANCE);

    volatile ble_error_t initerr = ble.init(bleInitComplete);
    thread_sleep_for(2000);
    ble.shutdown();
    
    NRF_POWER->SYSTEMOFF = 1;`

What can I do about this now?
what peripheral could still be active and drawing about 20uA?

Also, ofcourse systemON current is waay too high! I was at about 200uA AVG with SD, now I am at over 900uA AVG!
@0xc0170

pentabarf, 2020-03-02:
@CodingGhost 
I think you need to ENABLE RTC for Low Power Mode.

Also try to activate DC-DC when assembled:
`NRF_POWER->DCDCEN = 0x00000001;`

And Disable NFC functionallity:
`NRF_UICR->NFCPINS = 0x00000000;`

And disable UART in your mbed_app.json in `"target_overrides":{"*":{"target.console-uart": false}}`

I don't know if it is still necessary but add this to "target.macros_add":
```
                "DEVICE_SLEEP",
                "MBED_TICKLESS",
                "DEVICE_LPTICKER",
                "MBED_CONF_TARGET_TICKLESS_FROM_US_TICKER=0",
                "MBED_SLEEP_TRACING_ENABLED"
```

Then try this (without warranty for stability): comment following lines in 
_mbed-os/features/FEATURE_BLE/targets/TARGET_NORDIC/TARGET_NORDIC_CORDIO/TARGET_NRF5x/NRFCordioHCIDriver.cpp_

in function "static void idle_hook(void)"

```
core_util_critical_section_enter();
sleep_manager_lock_deep_sleep();
sleep_manager_unlock_deep_sleep();
core_util_critical_section_exit();
```

CodingGhost, 2020-03-02:
@pentabarf 
You are the man!
I am at 0.8uA of current again.
I tried to uncomment the section again, and my current went up to 3mA?!
Can I safely comment this section, for what is it good?
and is this a bug?

thanks a lot!

0xc0170, 2020-03-02:
@pentabarf  💯 thanks for sharing this!

yogeshk19, 2020-03-03:
@CodingGhost you have listed out quite a number of different things you tried out to lower the current draw. My question is which change seemed to help the most to drop the current draw. Was it just commenting out the code that @pentabarf  suggested? Also is the average current draw significantly lowered when advertising and connected with the changes you have tried out so far? 

@pentabarf, @0xc0170   The code that was commented out, isn't it necessary for the board to enter deep sleep, when idle? 

Thanks,
Yogesh

pentabarf, 2020-03-03:
@yogeshk19 no the function `static void idle_hook(void)` replaces the default idle function of mbed and as the names of the called functions say (`sleep_manager_lock_deep_sleep()` and `sleep_manager_unlock_deep_sleep()` encapsulate the `sleep()` function), deep sleep gets prevented.
I guess this is for a stable work of BLE because wakeup from deep sleep needs some time but i have no problems with it right now. But this may depend on ble workload. Maybe someone of the developers at packetcraft can tell us more?

you can check this out with the MBED_SLEEP_TRACING_ENABLED macro. When uncommenting the lines deep sleep never gets called.

CodingGhost, 2020-03-05:
@yogeshk19 
I tried many different things now,
all the registers I have set to disable peripheral hardware dont seem to make a difference.
disabling UART makes the biggest difference, then, adding the macros saves another ~2uA.
commenting the section, allows me to enter deepsleep at all after adding in the macros,
I cannot imagine this is wanted behaviour, is there any way of entering systemOFF without having to modify MBED code? 
@0xc0170 do you possibly know which developer of packetcraft? is responsible for this? could you mention him here?
for now, I will use this workaround and hope it works out.

I have to add, that I am not using the sleep() or deepsleep() function, but NRF_POWER->SYSTEMOFF = 1. I think its a bit strange that the MBED code prevents the chip from entering systemOFF, when I am setting it directly in the registers. I do not want to "sleep" I want the chip to turn off and then wake by GPIO, which works fine, but this does not have to be locked by MBED... It doesnt make sense to me.

yogeshk19, 2020-03-05:
@CodingGhost, @pentabarf  None of them worked for me, infact the power draw went up by 100uA's. Although I am not testing the system power off, I am testing just plain advertisement. Currently without any of the changes, the current draw is at 250 uA's and once I add the changes you tried out it goes up to 350 uA's.  At this point I am not sure what is going to get the cordio platform to operate at the softdevice levels of power draw?

@CodingGhost  did you make the following change, to reduce it to 500 uAs? Also what version of mbed_os are you on? I am still on mbed-os-5.13.4.

```
__WFE();
// critical section to complete sleep with locked deepsleep
/core_util_critical_section_enter();
//sleep_manager_lock_deep_sleep();
sleep();
//sleep_manager_unlock_deep_sleep();
//core_util_critical_section_exit();
```
Thanks,
Yogesh

CodingGhost, 2020-03-06:
@yogeshk19 my main concerns were regarding the systemOFF state, which i am happy now with.
250uA while advertising is still better than I am doing with my 500uA.
regarding the code change, I am sorry I confused you, but I misread something. I had commented everything, Includeing the sleep(); and replaced it with __WFE();, but upon further investigation, I dont think it has any benefits.

I am on MBEDOS 5.14.1

I really hope some dev at packetcraft will read through this, as I think this is a really critical issue.

yogeshk19, 2020-03-07:
I just posted a message on Packetcrafts contact us page and hopefully they would take the time to read it and do something about it and I am not sure why the MBED team which has been receiving all these messages regarding this github issue, hasn't raised it as a priority issue with the Packetcrafts team and get them to address it ASAP. At least by now we could have closed this issue and focus on developing our respective products.

Thanks,
Yogesh

ladislas, 2020-03-08:
I agree with @yogeshk19 and everything that has been said in this conversation.  I'm cc'ing @jyi-packetcraft from Packetcraft

I'm a bit worried with Packetcrafts taking over the Cordio development. They seem to be very competent (they all have strong backgrounds and most worked for arm) but they are a startup, so what happens if they run out of business or need to focus more on they own products than on Cordio?

Cordio is supposed to be open source, and in a way it is (https://github.com/packetcraft-inc/cordio) but there is no way to collaborate: we can't open issues and PR are not considered at all (except for the "next version" whatever that means).

A clear roadmap should be available, people should be able to contribute, ask questions, raise issues to make Cordio better. Cordio should be like mbed, with an incredible team doing the foundation work, and a lot of external contributors giving them a hand.

With the current situation, as a business it's hard to make the decision to use Cordio in a real product as I don't have any way of knowing that Cordio will be maintained in the future.

bsturgess, 2020-03-08:
Agree, we are having to move away from mbed for exactly this reason :(

5.11 is the last usable version of mbed for anything Bluetooth or low power

CodingGhost, 2020-03-08:
yep, we are also thinking about switching to zephyr, but we are very deep into mbed, so its a difficult decision.. Currently we hope things get better with mbed

ladislas, 2020-03-08:
I truly love mbed, its feature richness, the way the code is written and its API. I've tried zephyr, FreeRTOS, ChibiOS (liked this one), but none of them feel as natural to me as mbed. But it's only a matter a personal taste.

Regarding BLE, we have also been studying the idea of using the BLE modules with their native software/RTOS and have them communicate with our app mcu through SPI, UART, etc., with a simple protocol we've defined.

That way you keep the best of both worlds. But it means flashing both mcu independently.

bsturgess, 2020-03-08:
Problem with that is you just lose all the benefit of the NRF52 :(

ladislas, 2020-03-08:
True, but in our case, the main mcu (STM32F769) is running the firmware and needs to drive a LCD screen.

AGlass0fMilk, 2020-03-08:
It would probably take less time to debug and diagnose this issue than to migrate one's entire codebase and workflow to another RTOS/platform.

Thankfully the Cordio BLE stack is entirely open source and so that is possible. I'm sure everyone is aware, but the Softdevice has its own detrimental effects on development: complicates debugging considerably, it's entirely closed-source with mysterious faults, claims control over many system resources, eats a lot of flash and complicates in-field updates, the list goes on.

ARM should definitely be supporting this effort, however the larger Mbed community should take on some of the burden as this _is_ an open-source project.

ladislas, 2020-03-09:
> ARM should definitely be supporting this effort, however the larger Mbed community should take on some of the burden as this is an open-source project.

@AGlass0fMilk agreed! but as I said in a previous post, it's only open source as "the source are available, do what you want, but don't bother send patches, we don't work that way".

The Packetcraft website references this as a service:

[image]

Which is clearly not compatible with a community effort to improve Cordio as any new version might not be compatible with the improvements.

Or we decide as a community to fork Cordio and improve it there without having to rely on Packetcraft. And as they release new versions, we can incorporate the meaningful changes they make to our fork.

What's the real story behind Cordio? Where did it start? When did they decide to let Packetcraft handle the thing?

40Grit, 2020-03-09:
ARM bought a company that wrote Cordio. ARM open sourced cordio. Around this time last year ARM decided they didnt want to maintain cordio (thus the open sourcing) they also let go of the team that wrote it.

Packetcraft reformed from those that were let go.

Has anybody reached out to packetcraft directly on this?
If you have not then I reccomend you do so. This industry is driven by communication between engineers and partnerships between companies.

40Grit, 2020-03-09:
@jyi-packetcraft

AGlass0fMilk, 2020-03-09:
I find it’s easier to force ARM to figure this process out with a PR than with a feature request.

ladislas, 2020-03-09:
> Has anybody reached out to packetcraft directly on this?
> If you have not then I reccomend you do so. This industry is driven by communication between engineers and partnerships between companies.

@40Grit just did, I'll let you know what they say.

[140 more replies]