ARMmbed/mbed-os #10671: SSEL not working on STM32F NUCLEO_F767ZI

sjalloq, 2019-05-26:
### Description

Target:  NUCLEO_F767ZI
Toolchain: GCC_ARM
Tools:  mbed-cli 1.8.3
mbed-os: 73f096399b

I'm bringing up a new board and a simple testcase seems to find an issue with the SPI master.  I'm not seeing the SSEL line be asserted/deasserted.  I've found other STM32F HAL bugs around SPI but this seems like something basic that would have been caught.  Am I doing something wrong?

Shareef.

`#include "mbed.h"

DigitalOut led(LED1);
Ticker tick;
SPI spi(PA_7, PA_6, PA_5, PA_4); // MOSI, MISO, SCK, SSEL
Serial uart(USBTX, USBRX);

void toggle(void) {
	led = !led;
}

int main() {

	led = 0;
	tick.attach(&toggle, 2.0);

	spi.format(8,3);
	spi.frequency(1000000);
	
	//spi.select();

	while(1) {
		//spin
		spi.select();
		spi.write(0xAA);
		spi.deselect();
	}
}`

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

LMESTM, 2019-05-27:
@sjalloq is SSEL continuously high or low ?

0xc0170, 2019-05-27:
@sjalloq if you have logic analyzer , also useful to paste screenshot with signals - what is happening on the bus.

sjalloq, 2019-05-27:
I've just captured two views on my scope that show MOSI on the first shot toggling and then SSEL stuck low.  According to your pin list on the F767 page, SSEL is on pin PA_4 which is one up from the bottom left of CN7.

The code is slightly different now and I'm using the spi.transfer() function to send a buffer.

Cheers, Shareef.

[image]
[image]

VVESTM, 2019-05-29:
Hi,

I have submitted a new PR to fix this issue : #10708.
A setting was missing in our implementation.

Then, as I mentioned in the PR, be careful, this feature is not compatible with all SPI modes. It will work only in modes 0 and 2. See "NSS pulse mode" chapter in our reference manual. ([link ](https://www.st.com/content/ccc/resource/technical/document/reference_manual/02/35/09/0c/4f/f7/40/03/DM00083560.pdf/files/DM00083560.pdf/jcr:content/translations/en.DM00083560.pdf)for L4 as example)

What I have now : 
[image]

LMESTM, 2019-05-29:
@sjalloq it means you have to change your code to 
`spi.format(8,0);`
or 
`spi.format(8,2);`

sjalloq, 2019-05-29:
Thanks.  I'll give it a go when I get a chance.

linlingao, 2019-05-31:
PR submitted.

jeromecoutant, 2019-06-03:
@linlingao 
Why did you close the issue?
Issue is used to be closed when the patch is merged, which is not the case for #10708 

@0xc0170

PR #10708 (2019-06-09): TARGET_STM: SPI: add pulse on master transmissions
### Description

Add a pulse when using hardware chip select for SPI transmissions.
CS is at low level when a transmission is on-going.

Be careful, this is not compatible with all modes. It will work only
if PHA is 0, ie spi mode is 0 or 2. See stm32xx reference manual,
chapter "NSS pulse mode" for more details.

Fix #10671

### Pull request type

    [x ] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change
commit 16475829 (2019-06-09): TARGET_STM: SPI: add pulse on master transmissions

Add a pulse when using hardware chip select for SPI transmissions.
CS is at low level when a transmission is on-going.

Be careful, this is not compatible with all modes. It will work only
if PHA is 0, ie spi mode is 0 or 2. See stm32xx reference manual,
chapter "NSS pulse mode" for more details.

Fix #10671
closed 2019-06-09 by 0xc0170