ARMmbed/mbed-os #10673: usb serial device does not work on STM32L4R5ZI

shlomow, 2019-05-26:
### Description

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

Hi!

I imported today the blinky example and tryed to work with the usb device library as serial. I exported the code to cmake and compiled with gcc. I copied the example of the serial device here: https://os.mbed.com/docs/mbed-os/v5.12/apis/usbserial.html .
I added also `#define DEVICE_USBDEVICE 1` in my `mbed_config.h` file. I ran the example on NUCLEO_L4R5ZI device which seems to be ported in `USBPhy_STM32.cpp` file. The device was not even enumerated by the operating system (win10). Does anyone know about this? Am I doing something wrong? Thanks!

0xc0170, 2019-05-27:
@ARMmbed/team-st-mcd anything to be aware of when adding new targets?

0xc0170, 2019-06-04:
> @ARMmbed/team-st-mcd anything to be aware of when adding new targets?

Bump

MarceloSalazar, 2019-06-06:
Note this is a bug.

jeromecoutant, 2019-06-06:
Hi
USB tests are on going.
You can set devices:st label

0xc0170, 2019-08-13:
> USB tests are on going.

@ARMmbed/team-st-mcd Is there any update for this target? Is this still valid issue?