ARMmbed/mbed-os #10678: Disabling mbed-os features in .mbedignore broken in version >=5.12

kluszkiewicz, 2019-05-27:
### Description

Target: DISCO_L072CZ_LRWAN1
Toolchain: GCC_ARM (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
Tools: mbed-cli 1.8.3
mbed-os: 73f096399b4c (mbed-os-5.12.4)
program: mbed-os-example-blinky-no-rtos (#e98302381ff3)

When I'm building https://github.com/janjongboom/mbed-os-example-blinky-no-rtos in orignal state (with mbed-os 5.11.2) with some mbed-os features disabled in .mbedignore (no changes to repo state) everything works fine - features mentioned in .mbedignore are disabled and project is properly compiled and linked. It is fine also for mbed-os 5.11.5.

But if I'm trying to build the project with mbed-os 5.12.* build fails because of header dependencies to disabled features (nanostack, netsocket,...). Example:
`
Compile [  1.3%]: rf_configuration.c
[Fatal Error] rf_configuration.c@16,44: nanostack/platform/arm_hal_phy.h: No such file or directory
[ERROR] .\mbed-os\components\802.15.4_RF\stm-s2lp-rf-driver\source\rf_configuration.c:16:44: fatal error: nanostack/platform/arm_hal_phy.h: No such file or directory
 #include "nanostack/platform/arm_hal_phy.h"
`

Same issue occures for https://github.com/ARMmbed/mbed-os-example-lorawan (d1ccef0a039cc2ac76c15d36571d248ee88e2519).

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

So it seems that in mbed-os version >=5.12 there is dependency issue which makes disabling features in .mbedignore impossible.

Krystian

AnttiKauppila, 2019-05-27:
@kluszkiewicz This is a valid issue and is going to be fixed. @artokin will take care of this, but in the meanwile you should ignore \mbed-os\components\802.15.4_RF folder (you don't need 802.15.4 drivers for LoRaWAN or blinky examples)

artokin, 2019-05-27:
PR https://github.com/ARMmbed/mbed-os/pull/10680 created to fix this issue.

0xc0170, 2019-05-27:
Quick fix !

kluszkiewicz, 2019-05-28:
> 
> 
> @kluszkiewicz This is a valid issue and is going to be fixed. @artokin will take care of this, but in the meanwile you should ignore \mbed-os\components\802.15.4_RF folder (you don't need 802.15.4 drivers for LoRaWAN or blinky examples)

true, but for  https://github.com/ARMmbed/mbed-os-example-lorawan  i need this driver, but have same issue.

AnttiKauppila, 2019-05-29:
What exactly are you trying to do?

kluszkiewicz, 2019-05-29:
@AnttiKauppila 
I'm trying to disable some mbed-os features in mbed-os-example-lorawan application to reduce code and program size. I'm tring to do this following:
https://github.com/ARMmbed/mbed-os-example-lorawan/issues/122
https://os.mbed.com/questions/82060/Does-mbedignore-still-work/
https://os.mbed.com/blog/entry/Reducing-memory-usage-by-tuning-RTOS-con/

artokin, 2019-05-31:
@kluszkiewicz , PR https://github.com/ARMmbed/mbed-os/pull/10680 is now merged in to Mbed OS master and it will fix the issue with compilation of `rf_configuration.c`. 

For me the application builds OK if I add `mbed-os/usb/*
mbed-os/features/frameworks/TARGET_PSA/*` to the `.mbedignore`.

Would you please verify the fix?

kluszkiewicz, 2019-05-31:
@artokin 
I've checked out mbed-os at master and updated .mbed ignore and can confirm that application build OK now.

AnttiKauppila, 2019-06-04:
@kluszkiewicz Can you close this now?

closed 2019-06-04 by 0xc0170