ARMmbed/mbed-os #10679: GPRS and EthernetInterface Fails if tried to use simultaneously.

ThePhilomathsApprentice, 2019-05-27:
### Description
Hello Guys,
I'm trying to take some data over Ethernet Modbus Protocol on the Ethernet Network and Send the Data over to the cloud using MQTT protocol via External Cellular SIMCOM 900A.
SIMCOM Module is connected Serially to my K64F Board on Pins PTC17(Tx), and PTC16(Rx).
I'm facing trouble regarding the connection of both these Interfaces.
I'm Losing GPRS connection over SIMCOM module as soon as I'm initialising my Ethernet Interface.
However Both Connections/Interfaces work absolutely fine when used exclusively( 1 interface out of these both is initialised and the other is not).

As Ethernet Interface is Using Mbed OS Classes(EthernetInterface Class) and the SIMCOM module is connected as a purely Serial Device,I believe they are absolutely independent of each other and are not connected at any point in the code.
I'm unable to find what exactly is interfering in each others connectivity.

For Ethernet's Initialization, I'm Using eth.connect, where eth is an object of EthernetInterface( EthernetInterface eth; )

For Cellular Modules Initialization, I'm using AT Commands as per the SIMCOM module's  AT Command Guide. 

Can Someone Please tell me if I'm doing something wrong or any workaround to solve the problem ?
Problem is persistent when compiled in either mbed-cli or MCUXpresso IDE(with SDK for K64F & Processor: MK64FN1MOXXX12(Package: QFP100)).

### Issue request type
- What target does this relate to? --> K64F+ SIMCOM SIM900A Cellular Module
- What toolchain (name + version) are you using? --> GCC_ARM(v8.2.1)
- What tools (name + version - is it mbed-cli, online compiler or IDE) are you using? --> mbed-cli and MCUXpresso IDE 

    [x] Question
    [ ] Enhancement
    [ ] Bug

SeppoTakalo, 2019-06-03:
Hi @ThePhilomathsApprentice 

Please check the K64F datasheet: https://www.nxp.com/docs/en/data-sheet/K64P144M120SF5.pdf

PTC16 and PTC17 pins are shared with Ethernet. Therefore UART3 can only be used from these pins, if Ethernet is not active.

[image]

On the FRDM-K64F development boards, these pins conflict. Ethernet driver seems to initialise these exact pins https://github.com/ARMmbed/mbed-os/blob/master/features/netsocket/emac-drivers/TARGET_Freescale_EMAC/TARGET_K64F/hardware_init_MK64F12.c#L74..L81

ThePhilomathsApprentice, 2019-06-04:
Hey @SeppoTakalo, Thank you so much.
I am surprised how this fact went unnoticed while choosing the pins for my UART.
Thank Again. ;)

ThePhilomathsApprentice, 2019-06-04:
I think this issue is solved to my satisfaction I believe I can close this issue for now.

closed 2019-06-04 by ThePhilomathsApprentice