ARMmbed/mbed-os #10682: Adaptation of Mbed generated Keil projects

Kabron287, 2019-05-28:
Unfortunately, Mbed generate Keil projects do not works out of the box, generating a lot of "No such file" errors.
The reason in wrong Include Paths, produced by Mbed.

 Fast temporary solution is:
1. Copy the whole **Include Paths** content from the **C/C++** tab to clipboard(RightMouseClick, Select All, Copy)
2. Paste the string into any editor
3. In Editor **Ctrl+H** search for **;/** replace to **;.\\** Copy modified string back to clipboard
4. Paste clipboard content back into **Include Paths** on the **C/C++** tab
5. Enjoy

0xc0170, 2019-05-28:
`mbed export -m K64F -i uvision` for mbed-os-example-blinky: `".\BUILD\mbed-os-example-blinky.axf" - 0 Error(s), 58 Warning(s).`

I can't reproduce "No such file" locally. How to reproduce?

Kabron287, 2019-05-28:
Take HSP_RPC_GUI for MAX32620 project

0xc0170, 2019-05-28:
https://os.mbed.com/teams/MaximIntegrated/code/HSP_RPC_GUI/export - export to uvision - I see the problem. 

Locally using mbed-cli all fine. the include paths are similar. The one problem I see there is missing one include path .; at least ,otherwise they look identical.

cc @ARMmbed/mbed-os-tools

Kabron287, 2019-05-28:
I did not used mbed-cli. My case concerned online mbed ide export.

adbridge, 2020-10-15:
This applies to the old Mbed 5 tools which are no longer supported. I have passed this on to the tools team to see if there is anything that needed to be added to the new Mbed 6 tools to address this. For now I will close this here though.

closed 2020-10-15 by adbridge