ARMmbed/mbed-os #10685: SerialBase: abort_read() blocks deep sleep

teijokinnunen, 2019-05-28:
### Description

Target board: DISCO_L072CZ_LRWAN1
MBED version: 5.12.4 (SHA 73f096399b4cda1f780b140c87afad9446047432)
Toolchain: GCC ARM (gcc version 7.3.1); using Eclipse IDE, built using "release" config

Initiating an async UART read() and then aborting it with abort_read() will not release the deep sleep lock on MBED 5.12.4 (assuming no TX is going on).

A quick fix might be:
```
diff --git a/drivers/SerialBase.cpp b/drivers/SerialBase.cpp
index 6557a93e65..0d7eca93bb 100644
--- a/drivers/SerialBase.cpp
+++ b/drivers/SerialBase.cpp
@@ -228,7 +228,7 @@ void SerialBase::start_write(const void *buffer, int buffer_size, char buffer_wi
 void SerialBase::abort_write(void)
 {
     // rx might still be active
-    if (_rx_callback) {
+    if (!_rx_callback) {
         sleep_manager_unlock_deep_sleep();
     }
     _tx_callback = NULL;
@@ -238,7 +238,7 @@ void SerialBase::abort_write(void)
 void SerialBase::abort_read(void)
 {
     // tx might still be active
-    if (_tx_callback) {
+    if (!_tx_callback) {
         sleep_manager_unlock_deep_sleep();
     }
     _rx_callback = NULL;
```
The problem seems to have been introduced in commit f6c34a2303. In master branch there seems to be already related commits 6d648092e1 and cbb84d8ad35 that may fix this.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ X ] Bug

0xc0170, 2019-05-29:
> master branch there seems to be already related commits 6d64809 and cbb84d8 that may fix this.

Hi @teijokinnunen , can you test master branch ? As the next release  is 5.13 (no more 5.12.x release), it would be good to know this is not an issue there.

teijokinnunen, 2019-05-29:
Hi, I tested this with master branch and I can confirm that it works there: the system is able to go to deep sleep after abort_read().

linlingao, 2019-05-31:
Will be fixed in the next release.

closed 2019-05-31 by linlingao