ARMmbed/mbed-os #10690: USB MSD test fails on unmount

AGlass0fMilk, 2019-05-28:
### Description

While running USB Device MSD tests a failure occurs in test case 2 (mount/unmount test) with the following error:

`Error unmounting block device 8:16: GDBus.Error:org.freedesktop.UDisks2.Error.Failed: Error unmounting /dev/sdb: not mounted`

My understanding is that at this point in the test, the DUT tells the host test to unmount the MSD from the host side (rather than from the DUT side, which is what was tested up until that point).

I tried introducing a 1sec delay and it still fails. I'm not sure if it's a permissions issue or maybe just needs a longer delay?

My hardware setup:
- nRF52840_DK
- Direct USB connection of both programmer and DUT USB port
- Host is running Ubuntu 18.04, `uname -a` output:

`Linux magic-man 4.15.0-43-generic #46-Ubuntu SMP Thu Dec 6 14:45:28 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux`

I modified the host test script to simply skip calling unmount (and return success immediately) and the rest of the tests completed fine.

Log of test run attached:
[msd-test.log](https://github.com/ARMmbed/mbed-os/files/3228480/msd-test.log)

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

AGlass0fMilk, 2019-05-28:
@fkjagodzinski

maciejbocianski, 2019-05-29:
Looks like the disk was unmounted before the test got the chance to do it.
Could you show the log to see where the test failed on DUT side

0xc0170, 2019-05-31:
@adbridge Please check why bot is not running to clone this to our internal tracker

AGlass0fMilk, 2019-05-31:
Okay, so it was a problem with my system:

The test disk was being mounted as `/media/username/EC21-00002` and there were stale, out of date folders in my media directory. 

These were named `/media/username/EC21-0000` and `EC21-00001`. So I'm assuming, during development of the USB port for the nRF52840_DK, I tried to use the MSD driver with a buggy USB Phy driver and my OS never cleaned up what got left behind. I deleted the corrupted `EC21-0000*` folders and the drive started to mount as `EC21-0000` again. The unmount problem didn't reoccur.

Closing! :+1:

closed 2019-05-31 by AGlass0fMilk