ARMmbed/mbed-os #10713: Code compiled with IAR compiler 8.2 version crashes while printing NULL

sivasanath, 2019-05-30:
### Description

We tried compiling mbed-os-example-blinky usinf IAR compile(8.2 version), the only addition to the example is a new thread creation using cmsis_os.h,  with this we see a crash in function:report_thread_stats of stats_report.h file.

Where the crash seems to be due to "thread_stats[i].name" which is giving "null" for the new thread created, and printf is not able to print the same and crashing. if i comment the printf , app works fine.

But same thing when we did with GCC_ARM compiler, it printed "null" as the thread name.

i moved to cmsis_os2.h and  "thread_stats[i].name"  got the correct thread name.

i have two questions here
1) Why with cmsis_os.h, thread name was obtained as "null"?
2) Even though (1), why IAR compiler failed in printing the "null" string?

### Issue request type
    [ ] Question
    [ ] Enhancement
    [x ] Bug

0xc0170, 2019-05-31:
> We tried compiling mbed-os-example-blinky usinf IAR compile(8.2 version), the only addition to the example is a new thread creation using cmsis_os.h, with this we see a crash in function:report_thread_stats of stats_report.h file.

Can you provide code snippet? Why are you using `cmsis_os.h` ? It's RTX4 file - legacy - should not be used. Doesn't Mbed OS provide what is needed (via `mbed.h`) ?

sivasanath, 2019-06-04:
we moved to cmsis_os2.h api's , and by changing OS_DYNAMIC_MEM_SIZE macro from "0" to "25000" in mbed_rtx_conf.h, we see that every thing seems to be fine without crashes

But two questions 
1) with GCC_ARM compilation we didn't update OS_DYNAMIC_MEM_SIZE macro , but still things used to work fine, why we are seeing this issue only with IAR.
2) If the OS_DYNAMIC_MEM_SIZE  change is required , then what is the best way to update , it should be platform based right?

toyowata, 2019-06-04:
Probably related to: https://github.com/ARMmbed/simple-mbed-cloud-client/issues/96

teetak01, 2019-06-04:
Similar to https://github.com/ARMmbed/mbed-os-example-cellular/issues/125 which was fixed in https://github.com/ARMmbed/mbed-os/pull/10172 for that specific case.

kjbracey-arm, 2019-06-04:
So, the report_thread_stats function in blinky does have a bug, in that it will pass NULL to `printf("%s")` if any thread in the systemthread is totally unnamed.

Normally it gets away with it because all system threads (main, timer, idle) have names, and even if you make an unnamed thread, GCC and ARM libraries let you get away with printing NULL.

I don't know why you'd see any difference between CMSIS-RTOS 1 and 2 APIs - if you're creating a thread with NULL name, it's the same both ways as far as I can see.

RTX has been configured so that its dynamic allocation pools are zero-sized, because all Mbed OS objects use built-in storage, allowing them to be statically allocated. It is preferred that you use those objects in Mbed OS code. Use a `rtos::Thread`, rather than `osThread` functions.

If you do want to write CMSIS-RTOS code, then you can do what `rtos::Thread` etc do, which is pass in your own storage, rather than having it dynamically allocate.

If you really need the dynamic allocation, maybe because you have a body of portable CMSIS-RTOS code relying on it, then there's currently no mechanism to change the configuration, other than manually editing mbed_rtx_conf.h.

If a `#ifndef` was added there, then it would be possible to add a define to your application's mbed_app.json, or we could add a configuration option like the existing `rtos.idle-thread-stack-size`.

sivasanath, 2019-06-10:
thanks @kjbracey-arm for the info

0xc0170, 2019-08-13:
@sivasanath Did it answer your questions? Shall we close this issue?

sivasanath, 2019-08-13:
Yes , thanks

closed 2019-08-23 by 0xc0170