ARMmbed/mbed-os #10714: Driver/HAL for hardware timers capture and compare functionality

0x6d61726b, 2019-05-30:
### Description
In order to write portable code that is not specific to a single target, a driver/HAL seems to be a good idea for the capture and compare functionality of the microcontroller hardware timers. At the moment there seems to be only PwmOut functionality available which uses hardware timers for outputs.

What is the correct way for such an enhancement request?

### Issue request type

    [x] Question
    [x] Enhancement
    [ ] Bug

40Grit, 2019-05-30:
@farrenv @aglass0fmilk

0x6d61726b, 2019-08-12:
May I ask, if there is an updated information available?

40Grit, 2019-08-12:
@0xc0170 who is in charge of API's

0xc0170, 2019-08-12:
This comes to @ARMmbed/mbed-os-hal 

@0x6d61726b Do you have any references how this would look like? 

> I have seen user libraries out there, but those are very specific to a single target or even more restrictive to a single timer of a dedicated target.

Can you share urls to get an idea what users use and not in drivers/HAL yet.

closed 2019-10-31 by ciarmcom