ARMmbed/mbed-os #10715: exporters: kw24d armc6 fails in nightly

0xc0170, 2019-05-30:
### Description

Blinky fails for kw24d target, uvision6.

```
# Failed build combinations
################################################################################
# mbed-os-example-blinky KW24D uvision6

Error: L6366E: .\build\except.o attributes are not compatible with the provided attributes .
Object .\build\except.o contains Build Attributes that are incompatible with the provided attributes.
    Tag_FP_arch = VFPv4 instructions were permitted, but only citing registers D0-D15 (=6)
    Tag_ABI_HardFP_use = This code should execute on the single-precision variant derived from Tag_FP_arch (=1)
    Tag_FP_HP_extension = Use of the optional half-precision extension to VFPv3/Advanced SIMDv1 was permitted (=1)
C:\Users\ADMINI~1\AppData\Local\Temp\pb44-3: Error: L6366E: .\build\irq_cm4f.o attributes are not compatible with the provided attributes .
Object .\build\irq_cm4f.o contains Build Attributes that are incompatible with the provided attributes.
    Tag_FP_arch = VFPv4 instructions were permitted, but only citing registers D0-D15 (=6)
    Tag_ABI_HardFP_use = This code should execute on the single-precision variant derived from Tag_FP_arch (=1)
    Tag_FP_HP_extension = Use of the optional half-precision extension to VFPv3/Advanced SIMDv1 was permitted (=1)
```

The logs are here http://mbed-os-ci.s3-website-eu-west-1.amazonaws.com/?prefix=jenkins-ci/ARMmbed/mbed-os/mbed-os-ci-nightly/artifacts/master/223/exporter/FAIL/uvision6/KW24D/

cc @ARMmbed/team-nxp

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-05-31:
@kjbracey-arm Have you experienced this error ? It's new in the exporters - I haven't yet identified when it was introduced.

kjbracey-arm, 2019-06-06:
This is something to do with the KW24D being an M4 without FP (that's rare).  The handling was tweaked a bit for that recently.

I can see irq_cm4f.S having some `#ifdef __FPU_PRESENT` which should be `#if __FPU_PRESENT`, so that might be putting in some FP instructions (which would be avoided at runtime). I don't see any problem with except.S though.

The FPU selection is common for compiler and assembler, so any mismatch shouldn't be at our end. It comes out as `<Cpu>CPUTYPE("Cortex-M4") </Cpu>`, and that seems fine.

kjbracey-arm, 2019-06-06:
Actually, the `#ifdef __FPU_PRESENT` test isn't it a problem - even though it's defined as `0` in the device C headers, those aren't included by the assembler - instead the tools python puts a `-D` option in for FP devices, or not.

0xc0170, 2020-05-08:
Not valid anymore, will close.

closed 2020-05-08 by 0xc0170