ARMmbed/mbed-os #10717: [Warning] stm32f4xx_ll_adc.h@1844,3: 'register' storage class specifier is deprecated and incompatible with C++17 [-Wdeprecated-register]

teetak01, 2019-05-31:
### Description

ARMC6 compiler warnings with  stm32f4xx_ll_adc.h with Mbed OS 5.13.0-RC1

Tested with ublox_evk_odin_w2 target using mbed-cloud-client-example.

```
[Warning] stm32f4xx_ll_adc.h@4242,3: 'register' storage class specifier is deprecated and incompatible with C++17 [-Wdeprecated-register]
[Warning] stm32f4xx_ll_rtc.h@1213,3: 'register' storage class specifier is deprecated and incompatible with C++17 [-Wdeprecated-register]
[Warning] stm32f4xx_ll_usart.h@1062,3: 'register' storage class specifier is deprecated and incompatible with C++17 [-Wdeprecated-register]
```
### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-05-31:
@ARMmbed/team-st-mcd You already reported this in the PR updating the compilers. This is known warning for newer C++1x standards. 

@kjbracey-arm Understanding this should be fixed but if the latest C++1x bump for 5.13 caused so many warnings, shouldn't we mask this register at least for now and create an issue for each component causing the warnings to clean them up? My suggestion would be for 5.13 - for newer C++1x standard define register as empty for only components that need it (so we won't pollute other spaces). 
Different suggestions ?

chrissnow, 2019-05-31:
It actually slows the compiler down a fair bit too as it has to print all the warnings, nearly 8000 of them on the F429!

kjbracey-arm, 2019-06-03:
We can add the warning suppression to the build profiles: `-Wno-deprecated-register`

Seems like a reasonable thing to do for now, if there's a lot of them in the target code.

Not sure of the best way to track the clean-up; one big issue or multiple sub-issues or both. That's a low-priority task though, I guess. Not anticipating a C++17 upgrade soon.

kjbracey-arm, 2019-06-03:
Made a fix to shut the compiler up for now: #10740

jeromecoutant, 2019-06-03:
ST_INTERNAL_REF 67609

0xc0170, 2019-06-03:
>ST_INTERNAL_REF 67609

@jeromecoutant We are reviewing 10740. Once all green, will go into the release which would mean this could be closed. Do you want to keep this tracked (opened) to fix register usage in ST driver code or rather create new issue for the clean-up

0xc0170, 2019-06-03:
OK to be closed once 10740 is in

closed 2019-06-04 by 0xc0170

JojoS62, 2020-01-09:
This issue has been fixed for ARMC6 by using  "-Wno-deprecated-register". The counterpart for gcc is
  "-Wno-register".
Or will this be changed in STM HAL?
CMSIS has also been updated to not use the 'register' storage class in to 5.6, Mbed is still using CMSIS 5.3.

PR #10740 (2019-06-04): ARMC6: Suppress "register deprecated" warning
### Description

Lots of target code, STM in particular, uses the `register` keyword, so
it'll take a little while to clean up. In the interim, some builds are
producing a lot of warnings. Suppress the warning for now, as `register`
remains legal C++14 and C11, despite C++14 deprecating it.

C++17 removes `register`, so code will need to be cleaned before any
further C++ version update.

Fixes #10717

### Pull request type

    [X] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [  [... cut]