ARMmbed/mbed-os #10718: Ublox odin w2 driver does not compile with ARMC6 + release-profile (5.13.0-RC1)

teetak01, 2019-05-31:
### Description

Application: https://github.com/armmbed/mbed-cloud-client-example
Compiler: ARMC6
Mbed OS: 5.13.0-RC1
Profile: release

Fails to compile:

```
In file included from ./mbed-os/targets/TARGET_STM/TARGET_STM32F4/device.h:40:
./mbed-os/targets/TARGET_STM/TARGET_STM32F4/device/stm32f4xx_ll_usart.h:1062:3: warning: 'register' storage class specifier is deprecated and incompatible with C++17 [-Wdeprecated-register]
  register uint32_t usartdiv = 0x0U;
  ^~~~~~~~~
./mbed-os/targets/TARGET_STM/TARGET_STM32F4/device/stm32f4xx_ll_usart.h:1063:3: warning: 'register' storage class specifier is deprecated and incompatible with C++17 [-Wdeprecated-register]
  register uint32_t brrresult = 0x0U;
  ^~~~~~~~~
./mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_MODULE_UBLOX_ODIN_W2/sdk/ublox-odin-w2-drivers/OdinWiFiInterface.cpp:1744:64: warning: cast to 'void *' from smaller integer type 'std::uint16_t' (aka 'unsigned short') [-Wint-to-void-pointer-cast]
    status = cbWLAN_ioctl(cbWLAN_IOCTL_SET_AP_BEACON_INTERVAL, (void*)beacon_interval);
                                                               ^
./mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_MODULE_UBLOX_ODIN_W2/sdk/ublox-odin-w2-drivers/OdinWiFiInterface.cpp:2082:1: error: control may reach end of non-void function [-Werror,-Wreturn-type]
}

```

Compiles with debug & develop profiles (with warnings).

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-05-31:
cc @ARMmbed/team-ublox @ARMmbed/mbed-os-pan

0xc0170, 2019-05-31:
> ./mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_MODULE_UBLOX_ODIN_W2/sdk/ublox-odin-w2-drivers/OdinWiFiInterface.cpp:2082:1: error: control may reach end of non-void function [-Werror,-Wreturn-type]

This should be easy to fix. However, the fix might require rebuild of the driver? @ARMmbed/team-ublox Please advise. 

This would be good to have in 5.13.0-rc2 - fix asap.

aqib-ublox, 2019-05-31:
@0xc0170 we are working on ARMC6 support for driver but that is currently leading to hard-fault 

we require support from u there is currently no label to mbed export -i eclipse_armc6
secondly i will need support regarding how u setup openocd debug agent in eclipse after mbed export - eclipse_armc5 using stlink?

0xc0170, 2019-05-31:
> @0xc0170 we are working on ARMC6 support for driver but that is currently leading to hard-fault

Awesome, so should also fix this error about no return value.

> we require support from u there is currently no label to mbed export -i eclipse_armc6
secondly i will need support regarding how u setup openocd debug agent in eclipse after mbed export - eclipse_armc5 using stlink?

In the meantime, use uvision6 for debugging with armcc6. Or `make armc6` and add this to your eclipse project.

I havent used openocd for a while, pyOCD should work out of the box without any config.

0xc0170, 2019-06-03:
Note, to have this fix in rc2, it should be sent for review latest tomorrow

jeromecoutant, 2019-06-03:
Note that warning seen in the description should be solved by #10740

0xc0170, 2019-08-13:
@aqib-ublox Is there any update for this issue?

MarceloSalazar, 2019-08-13:
I confirm this is still an issue on Mbed OS 5.13.2
@aqib-ublox can you please investigate?

aqib-ublox, 2019-08-29:
@0xc0170 @MarceloSalazar https://github.com/ARMmbed/mbed-os/pull/11370 generated please verify?

0xc0170, 2019-09-09:
Tested locally, test compilation for ARM with release profile OK. I'll close this issue as resolved

closed 2019-09-09 by 0xc0170