ARMmbed/mbed-os #10719: Ublox odin w2 linker warnings with ARMC6

teetak01, 2019-05-31:
### Description

Application: https://github.com/armmbed/mbed-cloud-client-example
Compiler: ARMC6
Mbed OS: 5.13.0-RC1
Profile: debug & develop (release does not compile #10718 )

```
05:20:41 Link: app
05:20:41 [Warning] @0,0: L3912W: Option 'legacyalign' is deprecated.
05:20:41 [Warning] @0,0: L6869W: ./mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_MODULE_UBLOX_ODIN_W2/sdk/TOOLCHAIN_ARM/libublox-odin-w2-driver.ar(cb_main.o) requested the Rogue Wave Standard C++ Library which has been removed from the product. This might cause link errors because libc++ is used instead.
05:20:41 [Warning] @0,0: L6869W: ./mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_MODULE_UBLOX_ODIN_W2/sdk/TOOLCHAIN_ARM/libublox-odin-w2-driver.ar(hci_vendor_cmd.o) requested the Rogue Wave Standard C++ Library which has been removed from the product. This might cause link errors because libc++ is used instead.
05:20:41 [Warning] @0,0: L6869W: ./mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_MODULE_UBLOX_ODIN_W2/sdk/TOOLCHAIN_ARM/libublox-odin-w2-driver.ar(cb_ticker_wrapper.o) requested the Rogue Wave Standard C++ Library which has been removed from the product. This might cause link errors because libc++ is used instead.
05:20:41 [Warning] @0,0: L6439W: Multiply defined Global Symbol ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor>::GenericSecurityManager__sub_object(ble::pal::SecurityManager<ble::pal::vendor::cordio::CordioSecurityManager<ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor>>, ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor>>&, ble::pal::ConnectionEventMonitor<ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor>>&, ble::vendor::cordio::SigningEventMonitor<ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor>>&) defined in .text._ZN3ble7generic22GenericSecurityManagerINS_3pal6vendor6cordio21CordioSecurityManagerENS_6vendor6cordio19SigningEventMonitorEEC2ERNS2_15SecurityManagerINS5_IS9_EES9_EERNS2_22ConnectionEventMonitorIS9_EERNS8_IS9_EE(BUILD/UBLOX_EVK_ODIN_W2/ARMC6/mbed-os/features/FEATURE_BLE/targets/TARGET_CORDIO/source/CordioImplementations.o) rejected in favor of Symbol defined in .text._ZN3ble7generic22GenericSecurityManagerINS_3pal6vendor6cordio21CordioSecurityManagerENS_6vendor6cordio19SigningEventMonitorEEC2ERNS2_15SecurityManagerINS5_IS9_EES9_EERNS2_22ConnectionEventMonitorIS9_EERNS8_IS9_EE(BUILD/UBLOX_EVK_ODIN_W2/ARMC6/mbed-os/features/FEATURE_BLE/targets/TARGET_CORDIO/source/CordioBLE.o).
05:20:41 [Warning] @0,0: L6439W: Multiply defined Global Symbol ble::interface::SecurityManager<ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor>>::EventHandler::~EventHandler__sub_object() defined in .text._ZN3ble9interface15SecurityManagerINS_7generic22GenericSecurityManagerINS_3pal6vendor6cordio21CordioSecurityManagerENS_6vendor6cordio19SigningEventMonitorEEEE12EventHandlerD2Ev(BUILD/UBLOX_EVK_ODIN_W2/ARMC6/mbed-os/features/FEATURE_BLE/targets/TARGET_CORDIO/source/CordioImplementations.o) rejected in favor of Symbol defined in .text._ZN3ble9interface15SecurityManagerINS_7generic22GenericSecurityManagerINS_3pal6vendor6cordio21CordioSecurityManagerENS_6vendor6cordio19SigningEventMonitorEEEE12EventHandlerD2Ev(BUILD/UBLOX_EVK_ODIN_W2/ARMC6/mbed-os/features/FEATURE_BLE/targets/TARGET_CORDIO/source/CordioBLE.o).
05:20:41 [Warning] @0,0: L6439W: Multiply defined Global Symbol ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor>::~GenericSecurityManager__sub_object() defined in .text._ZN3ble7generic22GenericSecurityManagerINS_3pal6vendor6cordio21CordioSecurityManagerENS_6vendor6cordio19SigningEventMonitorEED2Ev(BUILD/UBLOX_EVK_ODIN_W2/ARMC6/mbed-os/features/FEATURE_BLE/targets/TARGET_CORDIO/source/CordioImplementations.o) rejected in favor of Symbol defined in .text._ZN3ble7generic22GenericSecurityManagerINS_3pal6vendor6cordio21CordioSecurityManagerENS_6vendor6cordio19SigningEventMonitorEED2Ev(BUILD/UBLOX_EVK_ODIN_W2/ARMC6/mbed-os/features/FEATURE_BLE/targets/TARGET_CORDIO/source/CordioBLE.o).
05:20:41 [Warning] @0,0: L6439W: Multiply defined Global Symbol ble::interface::GattClient<ble::generic::GenericGattClient<ble::pal::vendor::cordio::CordioGattClient, ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor>>>::~GattClient__sub_object() defined in .text._ZN3ble9interface10GattClientINS_7generic17GenericGattClientINS_3pal6vendor6cordio16CordioGattClientENS2_22GenericSecurityManagerINS6_21CordioSecurityManagerENS_6vendor6cordio19SigningEventMonitorEEEEEED2Ev(BUILD/UBLOX_EVK_ODIN_W2/ARMC6/mbed-os/features/FEATURE_BLE/targets/TARGET_CORDIO/source/CordioImplementations.o) rejected in favor of Symbol defined in .text._ZN3ble9interface10GattClientINS_7generic17GenericGattClientINS_3pal6vendor6cordio16CordioGattClientENS2_22GenericSecurityManagerINS6_21CordioSecurityManagerENS_6vendor6cordio19SigningEventMonitorEEEEEED2Ev(BUILD/UBLOX_EVK_ODIN_W2/ARMC6/mbed-os/features/FEATURE_BLE/targets/TARGET_CORDIO/source/CordioBLE.o).
05:20:41 [Warning] @0,0: L6439W: Multiply defined Global Symbol rtos::Thread::Thread__sub_object(osPriority_t, unsigned, unsigned char*, const char*) defined in i._ZN4rtos6ThreadC1E12osPriority_tjPhPKc(./mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F439xI/TARGET_MODULE_UBLOX_ODIN_W2/sdk/TOOLCHAIN_ARM/libublox-odin-w2-driver.ar(cb_ticker_wrapper.o)) rejected in favor of Symbol defined in .text._ZN4rtos6ThreadC2E12osPriority_tjPhPKc(BUILD/UBLOX_EVK_ODIN_W2/ARMC6/mbed-os/components/802.15.4_RF/atmel-rf-driver/source/NanostackRfPhyAtmel.o).
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-05-31:
cc @ARMmbed/team-ublox @ARMmbed/mbed-os-pan

pan-, 2019-05-31:
Looks like `libublox-odin-w2-driver.ar` hasn't been compiled with ARMC6.

0xc0170, 2019-06-03:
Possibly fix for https://github.com/ARMmbed/mbed-os/issues/10718 will address  this as well

0xc0170, 2019-10-25:
#11370 should have fixed this problem. I'll close this as resolved

closed 2019-10-25 by 0xc0170