ARMmbed/mbed-os #10721: Out of range needs to be checked before casting to enum

Arjun765, 2019-05-31:
### Description

In "attestation_core.c:719" integer is casted to enum and then checked to see if it is out of range which is unspecified behaviour. 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

Arjun765, 2019-05-31:
Adding "-Werror=type-limits" flag will help us to find these at compile time.

0xc0170, 2019-05-31:
cc @ARMmbed/mbed-os-psa

jainvikas8, 2019-11-18:
The error reproducible only via GCC, unable to do it with ARM6

`mbed test --compile -n *attestation* -m K64F -t GCC_ARM --profile debug --clean
debug profile modified with "-Werror=type-limits"`

This error only relates to the secure side, so K64F emulates that.

Also when building V8-M targets like **ARM_MUSCA_A1_S** and **LPC55S69_S** (need to hack the target.json as GCC_ARM is not yet supported for this), we see this issue.

`python3 tools/psa/release.py -m ARM_MUSCA_A1_S -t GCC_ARM -d`

Seeing these could go away with the new version of TF-M, but there are errors when the compiler option is used.

The issue has been raised - **https://developer.trustedfirmware.org/T586

closed 2019-11-18 by ciarmcom