ARMmbed/mbed-os #10723: Add new network interface

shlomow, 2019-05-31:
### Description

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

Hi! I want to work with the SLIP framing protocol as a network interface. I know I can use PPP instead of that, but I already have a system which relies on that protocol. How can I add this functionality? I want to use the on board lwip stack of course. 

Thanks!

AnttiKauppila, 2019-06-04:
@shlomow Can you use features/lwipstack/lwip/src/include/netif/slipif.h ?

40Grit, 2019-06-04:
@loverdeg-ep

loverdeg-ep, 2019-06-04:
@shlomow The Nanostack Border Router example may provide some insight too.

[image]

https://github.com/ARMmbed/nanostack-border-router

0xc0170, 2019-10-25:
@shlomow If you have any further questions, please follow up on forum.mbed.com

closed 2019-10-25 by 0xc0170