ARMmbed/mbed-os #10725: UBLOX_C030_U201: tests-mbed_drivers-lp_timer failing nightly

0xc0170, 2019-05-31:
### Description

IAR and GCC ARM affected at least. At least two latest nightly run 

```
[1559265072.39][CONN][RXD] >>> Running case #2: 'Test: LowPowerTimer - measure time accumulation.'...
[1559265072.39][CONN][INF] found KV pair in stream: {{__testcase_start;Test: LowPowerTimer - measure time accumulation.}}, queued...
mbedgt: :123::FAIL: Expected 0.010000 Was 0.011125
[1559265072.49][CONN][RXD] :123::FAIL: Expected 0.010000 Was 0.011125
[1559265072.60][CONN][INF] found KV pair in stream: {{__testcase_finish;Test: LowPowerTimer - measure time accumulation.;0;1}}, queued...
[1559265072.69][CONN][RXD] >>> 'Test: LowPowerTimer - measure time accumulation.': 0 passed, 1 failed with reason 'Assertion Failed'
[1559265072.69][CONN][RXD]
[1559265072.79][CONN][RXD] >>> Test cases: 1 passed, 1 failed with reason 'Assertion Failed'
[1559265072.79][CONN][RXD] >>> TESTS FAILED!
```

The full report:
```
| UBLOX_C030_U201-IAR | UBLOX_C030_U201 | tests-mbed_drivers-lp_timer                                                  | Test: LowPowerTimer - float operator.                                                             | 0      | 0      | SKIPPED | 0.0                |
| UBLOX_C030_U201-IAR | UBLOX_C030_U201 | tests-mbed_drivers-lp_timer                                                  | Test: LowPowerTimer - measure time accumulation.                                                  | 0      | 1      | FAIL    | 0.1                |
| UBLOX_C030_U201-IAR | UBLOX_C030_U201 | tests-mbed_drivers-lp_timer                                                  | Test: LowPowerTimer - reset.                                                                      | 0      | 0      | SKIPPED | 0.0                |
| UBLOX_C030_U201-IAR | UBLOX_C030_U201 | tests-mbed_drivers-lp_timer                                                  | Test: LowPowerTimer - start started timer.                                                        | 0      | 0      | SKIPPED | 0.0                |
| UBLOX_C030_U201-IAR | UBLOX_C030_U201 | tests-mbed_drivers-lp_timer                                                  | Test: LowPowerTimer - stopped after creation.                                                     | 1      | 0      | OK      | 0.09               |
| UBLOX_C030_U201-IAR | UBLOX_C030_U201 | tests-mbed_drivers-lp_timer                                                  | Test: LowPowerTimer - time measurement 1 ms.                                                      | 0      | 0      | SKIPPED | 0.0                |
| UBLOX_C030_U201-IAR | UBLOX_C030_U201 | tests-mbed_drivers-lp_timer                                                  | Test: LowPowerTimer - time measurement 1 s.                                                       | 0      | 0      | SKIPPED | 0.0                |
| UBLOX_C030_U201-IAR | UBLOX_C030_U201 | tests-mbed_drivers-lp_timer                                                  | Test: LowPowerTimer - time measurement 10 ms.                                                     | 0      | 0      | SKIPPED | 0.0                |
| UBLOX_C030_U201-IAR | UBLOX_C030_U201 | tests-mbed_drivers-lp_timer                                                  | Test: LowPowerTimer - time measurement 100 ms.                                                    | 0      | 0      | SKIPPED | 0.0                |
```

Full report can be found http://mbed-os-ci.s3-website-eu-west-1.amazonaws.com/?prefix=jenkins-ci/ARMmbed/mbed-os/mbed-os-ci-nightly/artifacts/master/224/greentea-test-HAL-Core/78/FAIL/UBLOX_C030_U201/

cc @ARMmbed/team-ublox @MarceloSalazar 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

fahim-ublox, 2019-05-31:
I haven't seen this problem with C030-U201 in our local automation system. But we will check your logs and update you. Internal JIRA ticket for reference PGM_NO_MBD-238

mudassar-ublox, 2019-06-13:
Issue didn't reproduced with us. Attaching logs.

[C030_U201_Test_Logs.txt](https://github.com/ARMmbed/mbed-os/files/3284674/C030_U201_Test_Logs.txt)

mprse, 2019-12-13:
`UBLOX_C030_U201` has `STM32F437VG76` MCU. This MCU does not have `LPTIM` peripheral and low power timer is driven by `RTC (LSI)`:

https://github.com/ARMmbed/mbed-os/blob/8fb43a8db5d74d2e8c49e10ad0388d7719648aad/targets/TARGET_STM/lp_ticker.c#L512-L519

https://github.com/ARMmbed/mbed-os/blob/8fb43a8db5d74d2e8c49e10ad0388d7719648aad/targets/TARGET_STM/rtc_api_hal.h#L41-L45

According to the documentation of `STM32F437VG76` MCU:

The LSI RC acts as an low-power clock source that can be kept running in Stop and
Standby mode for the independent watchdog (IWDG) and Auto-wakeup unit (AWU). The
clock frequency is **around 32 kHz**. For more details, refer to the electrical characteristics
section of the datasheets.

[image]

It seems that typical LSI frequency is `32 kHz`, but it may vary from `17` to `47 kHz`! This makes a huge difference and means that we can't rely on configured LSI frequency:

https://github.com/ARMmbed/mbed-os/blob/8fb43a8db5d74d2e8c49e10ad0388d7719648aad/targets/TARGET_STM/TARGET_STM32F4/device/stm32f4xx_hal_conf.h#L125-L127

I have measured the real lp-ticker frequency on my local target (where the test passes) and on the nightly/RAAS target (where the test fails). The table below shows the number of lp-ticker ticks during 1 sec (lp-ticker frequency). The 1 sec time period was defined using the us-ticker.

Test | RAAS | LOCAL
-- | -- | --
1 | 8835 | 7951
2 | 8853 | 7953
3 | 8866 | 7971
4 | 8873 | 7973
5 | 8873 | 7966
6 | 8866 | 7965
7 | 8853 | 7970
8 | 8847 | 7966
9 | 8855 | 7971
10 | 8867 | 7974
11 | 8857 | 7971
12 | 8854 | 7976
13 | 8846 | 7972
14 | 8860 | 7974
15 | 8871 | 7973
16 | 8870 | 7976
17 | 8871 | 7980
18 | 8873 | 7978
19 | 8872 | 7977
20 | 8891 | 7974

  | RAAS | LOCAL
-- | -- | --
Average | 8863 | 7971

We can see that the frequency is not stable and we have different average lp-ticker frequencies on RAAS target (`8.8 kHz`) and my local target (`7.9 kHz`). This gives the following real LSI frequency: 
- `RAAS: 8863  Hz * 4 = 35 452 Hz`
- `LOCAL: 7971 Hz * 4 = 31 884‬ Hz`

This explains why the test fails on RAAS target (configured LSI frequency is `32 000 Hz`).

Got the following propositions:

1. Disable lp-ticker for STM targets that use LSI. According to the requirements, lp-ticker accuracy should be +/- 10%. Since the test is failing this requirement is not met.

2. Extend the ticker information structure with a field like `calibration_required`:
https://github.com/ARMmbed/mbed-os/blob/8fb43a8db5d74d2e8c49e10ad0388d7719648aad/hal/ticker_api.h#L53-L56

a) if `calibration_required` field is set to `true` then lp-ticker tests should be skipped - bad solution since in such case lp-ticker will not be tested.
b) provide some calibration mechanism that should be used before the test (if `calibration_required` field is set to `true`) - this might be tricky but makes sense to me.

Other ideas?
cc @jamesbeyond @0xc0170 @fahim-ublox @LMESTM @jeromecoutant

MarceloSalazar, 2020-03-19:
LPtimer for F4x9 is disabled and not supported.

closed 2020-03-19 by MarceloSalazar