ARMmbed/mbed-os #10728: K82F: qspi test failing in nightly

0xc0170, 2019-05-31:
### Description

All toolchains, should be possible to test on the latest master

```
mbedgt: test suite 'tests-mbed_hal-qspi' ............................................................. FAIL in 36.44 sec
	test case: 'qspi frequency setting test' ..................................................... OK in 0.13 sec
	test case: 'qspi init/free test' ............................................................. OK in 0.09 sec
	test case: 'qspi memory id test' ............................................................. OK in 0.13 sec
	test case: 'qspi write(1-1-1)/x1  read(1-1-1)/x1  repeat/x1  test' ........................... OK in 0.12 sec
	test case: 'qspi write(1-1-1)/x1  read(1-1-1)/x1  repeat/x4  test' ........................... OK in 0.20 sec
	test case: 'qspi write(1-1-1)/x1  read(1-1-1)/x4  repeat/x1  test' ........................... OK in 0.07 sec
	test case: 'qspi write(1-1-1)/x1  read(1-1-2)/x1  repeat/x1  test' ........................... OK in 0.15 sec
	test case: 'qspi write(1-1-1)/x1  read(1-1-2)/x1  repeat/x4  test' ........................... OK in 0.30 sec
	test case: 'qspi write(1-1-1)/x1  read(1-1-2)/x4  repeat/x1  test' ........................... OK in 0.13 sec
	test case: 'qspi write(1-1-1)/x1  read(1-1-4)/x1  repeat/x1  test' ........................... FAIL in 1.81 sec
	test case: 'qspi write(1-1-1)/x1  read(1-1-4)/x1  repeat/x4  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-1-1)/x1  read(1-1-4)/x4  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-1-1)/x1  read(1-2-2)/x1  repeat/x1  test' ........................... OK in 0.10 sec
	test case: 'qspi write(1-1-1)/x1  read(1-2-2)/x1  repeat/x4  test' ........................... OK in 0.21 sec
	test case: 'qspi write(1-1-1)/x1  read(1-2-2)/x4  repeat/x1  test' ........................... OK in 0.10 sec
	test case: 'qspi write(1-1-1)/x1  read(1-4-4)/x1  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-1-1)/x1  read(1-4-4)/x1  repeat/x4  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-1-1)/x1  read(1-4-4)/x4  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-1-1)/x4  read(1-1-1)/x1  repeat/x1  test' ........................... OK in 0.10 sec
	test case: 'qspi write(1-1-1)/x4  read(1-1-2)/x1  repeat/x1  test' ........................... OK in 0.15 sec
	test case: 'qspi write(1-1-1)/x4  read(1-1-4)/x1  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-1-1)/x4  read(1-2-2)/x1  repeat/x1  test' ........................... OK in 0.20 sec
	test case: 'qspi write(1-1-1)/x4  read(1-4-4)/x1  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x1  read(1-1-1)/x1  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x1  read(1-1-1)/x1  repeat/x4  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x1  read(1-1-1)/x4  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x1  read(1-1-2)/x1  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x1  read(1-1-2)/x1  repeat/x4  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x1  read(1-1-2)/x4  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x1  read(1-1-4)/x1  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x1  read(1-1-4)/x1  repeat/x4  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x1  read(1-1-4)/x4  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x1  read(1-2-2)/x1  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x1  read(1-2-2)/x1  repeat/x4  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x1  read(1-2-2)/x4  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x1  read(1-4-4)/x1  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x1  read(1-4-4)/x1  repeat/x4  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x1  read(1-4-4)/x4  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x4  read(1-1-1)/x1  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x4  read(1-1-2)/x1  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x4  read(1-1-4)/x1  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x4  read(1-2-2)/x1  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x4  read(1-4-4)/x1  repeat/x1  test' ........................... SKIPPED in 0.00 sec
```

Latest report can be found : http://mbed-os-ci.s3-website-eu-west-1.amazonaws.com/?prefix=jenkins-ci/ARMmbed/mbed-os/mbed-os-ci-nightly/artifacts/master/223/greentea-test-HAL-Core/77/FAIL/K82F/

cc @ARMmbed/team-nxp @ARMmbed/mbed-os-hal 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

maciejbocianski, 2019-06-04:
@0xc0170 
This bug was introduced by https://github.com/ARMmbed/mbed-os/pull/10435

0xc0170, 2019-06-04:
@maciejbocianski would you be able to find the offender in that PR - what is causing this? it should be reverted or fixed.

maciejbocianski, 2019-06-04:
fix #10755

0xc0170, 2019-06-05:
Fix integreated, should be resolved now

closed 2019-06-05 by 0xc0170