ARMmbed/mbed-os #10729: GR_LYCHEE: timeout in various tests in nightly

0xc0170, 2019-05-31:
### Description

I can't find the common error. It seems that tests just timeout , not failure. It is only for ARM.

The report:

```
mbedgt: test suite report:
| target          | platform_name | test suite                                                                   | result  | elapsed_time (sec) | copy_method |
|-----------------|---------------|------------------------------------------------------------------------------|---------|--------------------|-------------|
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | features-frameworks-utest-tests-unit_tests-basic_test                        | OK      | 130.17             | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | features-frameworks-utest-tests-unit_tests-basic_test_default                | OK      | 167.99             | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | features-frameworks-utest-tests-unit_tests-case_async_validate               | OK      | 56.26              | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | features-frameworks-utest-tests-unit_tests-case_control_async                | OK      | 67.09              | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | features-frameworks-utest-tests-unit_tests-case_control_repeat               | OK      | 244.32             | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | features-frameworks-utest-tests-unit_tests-case_selection                    | OK      | 146.38             | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | features-frameworks-utest-tests-unit_tests-case_setup_failure                | OK      | 155.66             | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | features-frameworks-utest-tests-unit_tests-case_teardown_failure             | OK      | 249.02             | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | features-frameworks-utest-tests-unit_tests-control_type                      | OK      | 85.05              | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | features-frameworks-utest-tests-unit_tests-minimal_async_scheduler           | OK      | 59.45              | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | features-frameworks-utest-tests-unit_tests-minimal_scheduler                 | OK      | 159.4              | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | features-frameworks-utest-tests-unit_tests-test_assertion_failure_test_setup | OK      | 166.18             | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | features-frameworks-utest-tests-unit_tests-test_setup_case_selection_failure | OK      | 138.45             | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | features-frameworks-utest-tests-unit_tests-test_setup_failure                | OK      | 208.52             | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | features-frameworks-utest-tests-unit_tests-test_skip                         | OK      | 184.67             | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-events-queue                                                           | OK      | 35.56              | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-events-timing                                                          | OK      | 180.69             | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_drivers-c_strings                                                 | OK      | 77.81              | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_drivers-crc                                                       | TIMEOUT | 176.65             | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_drivers-dev_null                                                  | OK      | 110.19             | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_drivers-echo                                                      | OK      | 245.71             | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_drivers-flashiap                                                  | TIMEOUT | 198.8              | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_drivers-generic_tests                                             | OK      | 112.91             | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_drivers-lp_timer                                                  | OK      | 58.44              | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_drivers-race_test                                                 | OK      | 74.69              | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_drivers-rtc                                                       | OK      | 121.62             | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_drivers-sleep_lock                                                | OK      | 108.75             | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_drivers-stl_features                                              | OK      | 69.27              | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_drivers-timer                                                     | OK      | 116.26             | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_drivers-timerevent                                                | OK      | 212.02             | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_functional-callback                                               | OK      | 62.04              | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_functional-callback_big                                           | OK      | 67.11              | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_functional-callback_small                                         | OK      | 60.54              | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_functional-functionpointer                                        | OK      | 102.18             | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_hal-common_tickers                                                | OK      | 33.36              | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_hal-flash                                                         | TIMEOUT | 47.71              | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_hal-gpio                                                          | OK      | 26.46              | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_hal-lp_ticker                                                     | OK      | 27.93              | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_hal-minimum_requirements                                          | OK      | 26.01              | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_hal-pinmap                                                        | OK      | 28.17              | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_hal-rtc                                                           | OK      | 70.95              | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_hal-rtc_reset                                                     | OK      | 32.47              | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_hal-rtc_time                                                      | OK      | 31.82              | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_hal-rtc_time_conv                                                 | OK      | 85.75              | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_hal-sleep                                                         | OK      | 29.28              | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_hal-sleep_manager                                                 | OK      | 28.46              | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_hal-sleep_manager_racecondition                                   | OK      | 40.48              | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_hal-ticker                                                        | OK      | 39.69              | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_hal-trng                                                          | OK      | 34.73              | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbed_hal-us_ticker                                                     | OK      | 27.23              | default     |
[... cut ...]
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbedmicro-rtos-mbed-signals                                            | OK      | 35.45              | default     |
| GR_LYCHEE-ARMC6 | GR_LYCHEE     | tests-mbedmicro-rtos-mbed-threads                                            | TIMEOUT | 47.96              | default     |
```

Full logs can be found: http://mbed-os-ci.s3-website-eu-west-1.amazonaws.com/?prefix=jenkins-ci/ARMmbed/mbed-os/mbed-os-ci-nightly/artifacts/master/224/greentea-test-HAL-Core/77/FAIL/GR_LYCHEE/

@ARMmbed/team-renesas 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

d-kato, 2019-06-06:
I'm checking now.

d-kato, 2019-06-06:
Revision : fbc489e6a51a5afefe669fe32481fc564ea6ed44

I will only pull requests for the following fixes. Is this all right?  
tests-mbed_drivers-crc  
tests-mbed_drivers-flashiap  
tests-mbed_hal-flash  

#### The following two tests could not be reproduced.  
- **tests-mbed_platform-stats_cpu**
A test was not created because the following conditions were not met.  
mbed-os\TESTS\mbed_platform\stats_cpu\main.c  
```
#if !defined(MBED_CPU_STATS_ENABLED) || !DEVICE_LPTICKER || !DEVICE_SLEEP
#error [NOT_SUPPORTED] test not supported
#endif
```
- **tests-mbedmicro-rtos-mbed-threads**
I tried several times but did not fail. Fixed in https://github.com/ARMmbed/mbed-os/pull/10596 ?  (The stack usage may have decreased.)

#### The following needs to be fixed.  
- **tests-mbed_drivers-crc**
The task stack is insufficient. It succeeded when it increased from 320 bytes to 380 bytes.  
**Change file** : mbed-os\TESTS\mbed_drivers\crc\main.c (147)  
`` Thread t1(osPriorityNormal1, 320); -> Thread t1(osPriorityNormal1, 380); ``

- **tests-mbed_drivers-flashiap** and **tests-mbed_hal-flash**
The flash driver process needs to be located in RAM, but the function "clear_spimd_reg()" has been replaced with "memset()" located in ROM due to the effect of optimization.  
The function "clear_spimd_reg ()" has been changed to not be affected by optimization.  
**Change file** : mbed-os\targets\TARGET_RENESAS\TARGET_RZ_A1XX\flash.c  
```
#if defined (__ARMCC_VERSION) && (__ARMCC_VERSION >= 6010050)
#pragma clang optimize off
#endif
static void clear_spimd_reg(st_spibsc_spimd_reg_t * regset)
{
=== omit ===
}
#if defined (__ARMCC_VERSION) && (__ARMCC_VERSION >= 6010050)
#pragma clang optimize on
#endif
```

toyowata, 2019-06-06:
@d-kato 
For the `clear_spimd_reg()` function optimization, you can use `volatile` keyword rather than #ifdef and #pragma.

```cpp
static RAM_CODE_SEC void clear_spimd_reg(volatile st_spibsc_spimd_reg_t * regset);

// snip

static void clear_spimd_reg(volatile st_spibsc_spimd_reg_t * regset)
{
```
[Test result]
```
mbedgt: test suite report:
| target       | platform_name | test suite                  | result | elapsed_time (sec) | copy_method |
|--------------|---------------|-----------------------------|--------|--------------------|-------------|
| RZ_A1H-ARMC6 | RZ_A1H        | tests-mbed_drivers-flashiap | OK     | 21.45              | default     |
mbedgt: test suite results: 1 OK
mbedgt: test case report:
| target       | platform_name | test suite                  | test case                         | passed | failed | result | elapsed_time (sec) |
|--------------|---------------|-----------------------------|-----------------------------------|--------|--------|--------|--------------------|
| RZ_A1H-ARMC6 | RZ_A1H        | tests-mbed_drivers-flashiap | FlashIAP - init                   | 1      | 0      | OK     | 0.1                |
| RZ_A1H-ARMC6 | RZ_A1H        | tests-mbed_drivers-flashiap | FlashIAP - program                | 1      | 0      | OK     | 0.51               |
| RZ_A1H-ARMC6 | RZ_A1H        | tests-mbed_drivers-flashiap | FlashIAP - program across sectors | 1      | 0      | OK     | 0.16               |
| RZ_A1H-ARMC6 | RZ_A1H        | tests-mbed_drivers-flashiap | FlashIAP - program errors         | 1      | 0      | OK     | 0.11               |
| RZ_A1H-ARMC6 | RZ_A1H        | tests-mbed_drivers-flashiap | FlashIAP - timing                 | 1      | 0      | OK     | 1.01               |
mbedgt: test case results: 5 OK
mbedgt: completed in 22.51 sec
```

d-kato, 2019-06-06:
@toyowata , Thank you. I will use "volatile".

toyowata, 2019-08-01:
Fixed in 5.13.1 onwards.

closed 2019-08-01 by toyowata