ARMmbed/mbed-os #10730: DISCO_L475VG_IOT01A: qspi failing in nightly

0xc0170, 2019-05-31:
### Description

All toolchains affected

```
mbedgt: test suite 'tests-mbed_hal-qspi' ............................................................. FAIL in 33.39 sec
	test case: 'qspi frequency setting test' ..................................................... OK in 0.71 sec
	test case: 'qspi init/free test' ............................................................. OK in 0.10 sec
	test case: 'qspi memory id test' ............................................................. OK in 0.10 sec
	test case: 'qspi write(1-1-1)/x1  read(1-1-1)/x1  repeat/x1  test' ........................... OK in 0.10 sec
	test case: 'qspi write(1-1-1)/x1  read(1-1-1)/x1  repeat/x4  test' ........................... OK in 0.20 sec
	test case: 'qspi write(1-1-1)/x1  read(1-1-1)/x4  repeat/x1  test' ........................... OK in 0.11 sec
	test case: 'qspi write(1-1-1)/x1  read(1-1-2)/x1  repeat/x1  test' ........................... OK in 0.10 sec
	test case: 'qspi write(1-1-1)/x1  read(1-1-2)/x1  repeat/x4  test' ........................... OK in 0.30 sec
	test case: 'qspi write(1-1-1)/x1  read(1-1-2)/x4  repeat/x1  test' ........................... OK in 0.21 sec
	test case: 'qspi write(1-1-1)/x1  read(1-1-4)/x1  repeat/x1  test' ........................... FAIL in 1.90 sec
	test case: 'qspi write(1-1-1)/x1  read(1-1-4)/x1  repeat/x4  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-1-1)/x1  read(1-1-4)/x4  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-1-1)/x1  read(1-2-2)/x1  repeat/x1  test' ........................... OK in 0.10 sec
	test case: 'qspi write(1-1-1)/x1  read(1-2-2)/x1  repeat/x4  test' ........................... OK in 0.31 sec
	test case: 'qspi write(1-1-1)/x1  read(1-2-2)/x4  repeat/x1  test' ........................... OK in 0.19 sec
	test case: 'qspi write(1-1-1)/x1  read(1-4-4)/x1  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-1-1)/x1  read(1-4-4)/x1  repeat/x4  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-1-1)/x1  read(1-4-4)/x4  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-1-1)/x4  read(1-1-1)/x1  repeat/x1  test' ........................... OK in 0.21 sec
	test case: 'qspi write(1-1-1)/x4  read(1-1-2)/x1  repeat/x1  test' ........................... OK in 0.10 sec
	test case: 'qspi write(1-1-1)/x4  read(1-1-4)/x1  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-1-1)/x4  read(1-2-2)/x1  repeat/x1  test' ........................... OK in 0.25 sec
	test case: 'qspi write(1-1-1)/x4  read(1-4-4)/x1  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x1  read(1-1-1)/x1  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x1  read(1-1-1)/x1  repeat/x4  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x1  read(1-1-1)/x4  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x1  read(1-1-2)/x1  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x1  read(1-1-2)/x1  repeat/x4  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x1  read(1-1-2)/x4  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x1  read(1-1-4)/x1  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x1  read(1-1-4)/x1  repeat/x4  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x1  read(1-1-4)/x4  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x1  read(1-2-2)/x1  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x1  read(1-2-2)/x1  repeat/x4  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x1  read(1-2-2)/x4  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x1  read(1-4-4)/x1  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x1  read(1-4-4)/x1  repeat/x4  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x1  read(1-4-4)/x4  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x4  read(1-1-1)/x1  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x4  read(1-1-2)/x1  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x4  read(1-1-4)/x1  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x4  read(1-2-2)/x1  repeat/x1  test' ........................... SKIPPED in 0.00 sec
	test case: 'qspi write(1-4-4)/x4  read(1-4-4)/x1  repeat/x1  test' ........................... SKIPPED in 0.00 sec
```

Full logs can be found http://mbed-os-ci.s3-website-eu-west-1.amazonaws.com/?prefix=jenkins-ci/ARMmbed/mbed-os/mbed-os-ci-nightly/artifacts/master/224/greentea-test-HAL-Core/78/FAIL/DISCO_L475VG_IOT01A/

cc @ARMmbed/team-st-mcd @MarceloSalazar 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

jeromecoutant, 2019-06-03:
Here is the status I have:

Target | Toolchain | TestSuite | TestCase | mbed-os-5.12.4 | mbed-os-5.12.3 | mbed-os-5.12.2 | mbed-os-5.12.1
-- | -- | -- | -- | -- | -- | -- | --
DISCO_L475VG_IOT01A | GCC_ARM | tests-mbed_hal-qspi | qspi write(1-1-1)/x1    read(1-1-4)/x1  repeat/x1  test | FAIL | FAIL | OK | OK

jainvikas8, 2019-06-03:
I see the same failure with **ARMC6** toolchain on mbed-os-5.12.4.

trowbridgec, 2019-06-03:
I am seeing similar results with the `NRF52840_DK` and `GCC_ARM`.

jeromecoutant, 2019-06-04:
Faulty PR is #10435 
@morser499

jeromecoutant, 2019-06-04:
> I am seeing similar results with the NRF52840_DK

NRF52840_DK and DISCO_L475 are using same flash test config file: MX25RXX35F_config.h

0xc0170, 2019-06-04:
Thanks for all feedback.

> Faulty PR is #10435

@ARMmbed/team-cypress Please review. I'll comment on the PR itself

0xc0170, 2019-06-04:
@jeromecoutant good catch, looks like that PR edited common test code that might be faulty. We will need to find the fix or revert the offender.

@ARMmbed/mbed-os-hal If you can help here with qspi test case

maciejbocianski, 2019-06-04:
fix #10755

0xc0170, 2019-06-05:
Fix integreated, should be resolved now

closed 2019-06-05 by 0xc0170