ARMmbed/mbed-os #10731: CY8CPROTO_062_4343W: out of memory in nightly tests

0xc0170, 2019-05-31:
### Description

```
[1559264085.60][HTST][INF] CALLBACKs updated
[1559264085.60][HTST][INF] host test detected: default_auto
[1559264085.77][CONN][RXD]
[1559264085.77][CONN][RXD] >>> Running case #1: 'function init race'...
[1559264085.77][CONN][INF] found KV pair in stream: {{__testcase_name;class init race}}, queued...
[1559264085.80][CONN][INF] found KV pair in stream: {{__testcase_start;function init race}}, queued...
[1559264086.02][CONN][RXD]
[1559264086.02][CONN][RXD]
[1559264086.02][CONN][RXD] ++ MbedOS Error Info ++
[1559264086.02][CONN][RXD] Error Status: 0x8001011F Code: 287 Module: 1
[1559264086.02][CONN][RXD] Error Message: Operator new[] out of memory
[1559264086.02][CONN][RXD]
[1559264086.02][CONN][RXD] Location: 0x10009927
[1559264086.02][CONN][RXD] Error Value: 0x200
[1559264086.12][CONN][RXD] Current Thread: main  Id: 0x8004514 Entry: 0x1000B6C3 StackSize: 0x1000 StackMem: 0x8002E10 SP: 0x8003B2C
[1559264086.39][CONN][RXD] For more info, visit: https://mbed.com/s/error?error=0x8001011F&osver=999999&core=0x410FC241&comp=1&ver=6110004&tgt=CY8CPROTO_...
[1559264086.39][CONN][RXD]
[1559264086.39][CONN][RXD] -- MbedOS Error Info --
[1559264106.41][HTST][INF] test suite run finished after 20.82 sec...
[1559264106.42][CONN][INF] received special event '__host_test_finished' value='True', finishing
[1559264106.42][urllib3.connectionpool]Starting new HTTPS connection (1): mervi.mbedcloudtesting.com:443
[1559264106.60][urllib3.connectionpool]https://mervi.mbedcloudtesting.com:443 "PUT /resource/190100008b131b16028b131b00000000000000002e127069/disconnect HTTP/1.1" 200 87
[1559264106.60][urllib3.connectionpool]Starting new HTTPS connection (1): mervi.mbedcloudtesting.com:443
[1559264111.89][urllib3.connectionpool]https://mervi.mbedcloudtesting.com:443 "PUT /resource/190100008b131b16028b131b00000000000000002e127069/release HTTP/1.1" 200 87
[1559264111.89][HTST][INF] CONN exited with code: 0
[1559264111.89][HTST][INF] No events in queue
[1559264111.89][HTST][INF] stopped consuming events
[1559264111.89][HTST][INF] host test result(): None
[1559264111.89][HTST][WRN] missing __exit event from DUT
[1559264111.89][HTST][WRN] missing __exit_event_queue event from host test
[1559264111.89][HTST][ERR] missing __exit_event_queue event from host test and no result from host test, timeout...
[1559264111.89][HTST][INF] calling blocking teardown()
[1559264111.89][HTST][INF] teardown() finished
[1559264111.89][HTST][INF] {{result;timeout}}
```

Tests like tests-mbedmicro-rtos-mbed-semaphore, tests-mbed_drivers-race_test  and others hardfault for ARMC6

Full logs available http://mbed-os-ci.s3-website-eu-west-1.amazonaws.com/?prefix=jenkins-ci/ARMmbed/mbed-os/mbed-os-ci-nightly/artifacts/master/224/greentea-test-HAL-Core/78/FAIL/CY8CKIT_062_WIFI_BT/

@ARMmbed/team-cypress Have you experienced any of these? What is the target configuration that is causing these hardfaults?

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

linlingao, 2019-05-31:
@0xc0170 This is a known issue for CY8CPROTO_062_4343W. I suspect it's the same issue on WIFI_BT. The heap for ARM is too small (1K). I've told Cypress via email how to fix it.

hennadiykytsun, 2019-05-31:
Fix is the following?
In the assembly file:
1. remove the implementation of the legacy __user_initial_stackheap function
2. Add IMPORT __use_two_region_memory

linlingao, 2019-05-31:
1. Remove the Heap_Size definition and __user_initial_stackheap
2. Set ARM_LIB_HEAP in scatter file, you can refer here
http://www.keil.com/support/man/docs/ARMLINK/ARMLINK_pge1362065977713.htm

hennadiykytsun, 2019-05-31:
Thanks @linlingao 
This is the same issue as #10613 
ARM_LIB_HEAP / ARM_LIB_STACK are implemented correctly in the scatter file.
Just need to update the assembly file to remove __user_initial_stackheap function and add IMPORT __use_two_region_memory
We will make the update, run the tests and submit the PR.

0xc0170, 2019-06-03:
Thanks for the quick support @hennadiykytsun @linlingao. 

I'll close this as duplicate, lets focus on #10613

closed 2019-06-03 by 0xc0170