ARMmbed/mbed-os #10732: STM: ETH\LWIP breaks FATFS

chrissnow, 2019-05-31:
I'm having some odd issues with LWIP and FATFS not getting along.

I'm using the following:
STM NUCLEO_F429ZI but also tested a LPC1768 which works fine.
armc6
mbed-os at master SHA-1: 9cc1caa0312c3332afed06f7d9b34b5753c95988

The FS works fine until eth.connect is called, then the FS acts oddly and throws exceptions\odd return codes.

You will need to modify slightly to suit hardware, example supports SD and dataflash(tries to mount SD first)
Logs from code below.
[LPC1768.txt](https://github.com/ARMmbed/mbed-os/files/3241351/LPC1768.txt)
[STMF429.txt](https://github.com/ARMmbed/mbed-os/files/3241352/STMF429.txt)

```cpp
#include <stdio.h>
#include <errno.h>
#include <fstream>
#include <string>

#include "mbed.h"
#include "rtos.h"
#include "mbed.h"

#include "DataFlashBlockDevice.h"
#include "FATFileSystem.h"
#include "SDBlockDevice.h"
#include "EthernetInterface.h"

#include "string.h"

#if defined(TARGET_LPC1768) || defined(TARGET_LPC1769)

DigitalOut EECS(p15, 1);
DigitalOut EEHOLD(p17, 1);
DigitalOut EEWP(p16, 1);

DigitalOut SDCS(p8, 1);

Serial pc(USBTX, USBRX);

DataFlashBlockDevice flash(p5, p6, p7,p15,8000000,p16);
SDBlockDevice sd(p5, p6, p7, p8);

#endif

#if defined(TARGET_STM32F429xI) ||defined(TARGET_STM32F439xI)

DigitalOut EECS(D2, 1);
DigitalOut EEHOLD(D9, 1);
DigitalOut EEWP(D3, 1);

DigitalOut SDCS(SPI_CS, 1);

Serial pc(SERIAL_TX, SERIAL_RX);

DataFlashBlockDevice flash(SPI_MOSI, SPI_MISO, SPI_SCK,D2,8000000,D3);
SDBlockDevice sd(SPI_MOSI, SPI_MISO, SPI_SCK, SPI_CS);

#endif

FATFileSystem sdfs("sd");
FATFileSystem flashfs("flash");
EthernetInterface eth;
/*
DigitalOut EECS(D2, 1);
DigitalOut EEHOLD(D9, 1);
DigitalOut EEWP(D3, 1);

DigitalOut SDCS(SPI_CS, 1);

Serial pc(SERIAL_TX, SERIAL_RX);

FATFileSystem sdfs("sd");
FATFileSystem flashfs("flash");

DataFlashBlockDevice flash(SPI_MOSI, SPI_MISO, SPI_SCK, D2, 8000000, D3);
SDBlockDevice sd(SPI_MOSI, SPI_MISO, SPI_SCK, SPI_CS);

EthernetInterface eth;
*/
void return_error(int ret_val) {
	if (ret_val)
		printf("Failure. %d\r\n", ret_val);
	else
		printf("done.\r\n");
}

void errno_error(void* ret_val) {
	if (ret_val == NULL)
		printf(" Failure. %d \r\n", errno);
	else
		printf(" done.\r\n");
}

bool FSTest() {
	int error = 0;
	printf("Opening root directory.");
	DIR* dir = opendir("/flash/");

	struct dirent* de;
	printf("Printing all filenames:\r\n");
	while ((de = readdir(dir)) != NULL) {
		printf("  %s\r\n", &(de->d_name)[0]);
	}

	printf("Closing root directory. ");
	error = closedir(dir);
	return_error(error);
	printf("Filesystem Demo complete.\r\n");

	printf("Opening a new file, numbers.txt.");
	FILE* fd = fopen("/flash/numbers.txt", "w+");
	errno_error(fd);

	for (int i = 0; i < 20; i++) {
		printf("Writing decimal numbers to a file (%d/20)\r\n", i);
		fprintf(fd, "%d\r\n", i);
	}
	printf("Writing decimal numbers to a file (20/20) done.\r\n");

	printf("Closing file.");
	fclose(fd);
	printf(" done.\r\n");

	printf("Re-opening file read-only.");
	fd = fopen("/flash/numbers.txt", "r");
	errno_error(fd);

	printf("Dumping file to screen.\r\n");
	char buff[16] = { 0 };
	while (!feof(fd)) {
		int size = fread(&buff[0], 1, 15, fd);
		fwrite(&buff[0], 1, size, stdout);
	}
	printf("EOF.\r\n");

	printf("Closing file.");
	fclose(fd);
	printf(" done.\r\n");

	printf("Opening root directory.");
	dir = opendir("/flash/");
	errno_error(fd);

	printf("Printing all filenames:\r\n");
	while ((de = readdir(dir)) != NULL) {
		printf("  %s\r\n", &(de->d_name)[0]);
	}

	printf("Closing root directory. ");
	error = closedir(dir);
	return_error(error);
	printf("Filesystem Demo complete.\r\n");
	return (0);
}

void TestTask() {

	while (true) {

		wait(5);
		FSTest();

	}
}

bool mountFS(bool format = false) {
	int error = 0;

	if (format == true) {
		printf("Format filesystem.\r\n");

		error = FATFileSystem::format(&flash);
		return_error(error);
		error = flashfs.mount(&flash);
		return_error(error);
	} else {

		printf("Mounting DF on \"/flash\". \r\n");
		error = flashfs.mount(&flash);
		return_error(error);
		if (error != FR_OK) {
			printf("Format filesystem as error exists.\r\n");

			error = FATFileSystem::format(&flash);
			return_error(error);
			error = flashfs.mount(&flash);
			return_error(error);
		} else {
			DIR* dir = opendir("/flash/");
			if (dir == 0) //check we can open the root
					{
				//apparantly not
				printf(
						"Format filesystem as root folder couldnt be opened.\r\n");

				flashfs.unmount();
				error = FATFileSystem::format(&flash);
				return_error(error);
				error = flashfs.mount(&flash);
				return_error(error);
			} else {
				closedir(dir);
			}
		}

	}
	return 0;
}
int mountSDFS(bool format = false) {
	int error = 0;
	if (format == true) {
		printf("Format filesystem.\r\n");

		error = FATFileSystem::format(&sd);
		return_error(error);
		error = flashfs.mount(&sd);
		return_error(error);
	} else {
		printf("Mounting SD on \"/flash\". \r\n");
		error = flashfs.mount(&sd);
		return_error(error);
		if (error == FR_NO_FILESYSTEM) {
			printf("Format filesystem as none exists.\r\n");

			error = FATFileSystem::format(&sd);
			return_error(error);
			error = flashfs.mount(&sd);
			return_error(error);
		}
	}
	return (error);
}

int main() {
	int error = 0;

	pc.baud(115200);

	if (mountSDFS() != 0) {
		flashfs.unmount();
		mountFS();
	}
	printf("fstest pre eth\r\n");

	FSTest();
	printf("\r\nStarting Ethernet\r\n");

	eth.set_dhcp(true);

	eth.connect();
	printf("fstest post eth connect\r\n");

	FSTest();
	const char *rip = eth.get_ip_address();
	const char *rmac = eth.get_mac_address();
	printf("IP address is: %s\r\n", rip ? rip : "No IP");
	printf("MAC address is: %s\r\n", rmac ? rmac : "No MAC");

	printf("fstest post mac\r\n");

	FSTest();

	while (1) {
		wait(0.5);
	}
}

```

```json
{
 
    "target_overrides": {
        "*": {
            "target.components_add": ["SD","DATAFLASH"],
            "target.features_add": ["STORAGE"],
			"dataflash.SPI_FREQ": "40000000",
			"dataflash.binary-size": "1",
			"dataflash.dataflash-size": "0"
        },
		"LPC1768": {
			"dataflash.SPI_MOSI": "p5",
			"dataflash.SPI_MISO": "p6",
			"dataflash.SPI_CLK": "p7",
			"dataflash.SPI_CS": "p15",
			"sd.SPI_MOSI": "p5",
			"sd.SPI_MISO": "p6",
			"sd.SPI_CLK": "p7",
			"sd.SPI_CS": "p8"
        },
		"NUCLEO_F429ZI": {
           
			
			"dataflash.SPI_CS": "D2",
			"sd.SPI_CS": "SPI_CS"
        }
    }
   
}
```
 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

jeromecoutant, 2019-06-03:
Hi
Please, have a look on this page:
https://os.mbed.com/teams/ST/wiki/Nucleo-144pins-ethernet-spi-conflict

chrissnow, 2019-06-04:
@jeromecoutant that sorts it thanks.

However is there a way we can throw and exception if the user tries to use SPI1 and ETH without that defined?

It's rather hard to debug and certainly wasted a lot of my time and no doubt others in future.

jeromecoutant, 2019-06-04:
Hi
I don't know how to add some code to check this ... :-(

I have updated the board page: https://os.mbed.com/platforms/ST-Nucleo-F429ZI/#board-pinout
To be more explicit

chrissnow, 2019-06-04:
@0xc0170 any thoughts on this?

0xc0170, 2019-06-05:
> @0xc0170 any thoughts on this?

Is it about pin ownership? Because above link about spi conflicts points to board rather than software - modify target for some specific pin usage (ethernet/SPI).

chrissnow, 2019-06-05:
It's because a pin is shared, if we can throw an exception when a pin conflict happens it would save a lot of debugging to figure out why, currently things just break at runtime. SPI1 and ETH will never work on this board without a hardware mod, and I think it's a pretty common thing to happen!.

0xc0170, 2020-05-08:
> It's because a pin is shared, if we can throw an exception when a pin conflict happens it would save a lot of debugging to figure out why,

That would make sense. @jeromecoutant have you thought about this making it internal ?  To see if the pin is used (was initialized) already and requested again to just error?

adbridge, 2020-10-15:
@jeromecoutant @chrissnow can we close this ?

chrissnow, 2020-10-15:
I haven't tested it in a while but I assume the problem is still there.