ARMmbed/mbed-os #10734: Fix PWM glitch on write(), TARGET_STM/pwmout_api.c,

GrahamSayers, 2019-05-31:
### Description

It looks like for all STM32 devices, in pwmout_api.c, when you attempt to pwmout_write() a new duty cycle it updates the timer in a way that can cause the pwm signal to glitch.  It is most noticeable on faster pwm signals, say 100kHz to 200kHz.  I posted a screenshot of a glitch here:

[https://os.mbed.com/questions/86046/Issue-with-STM32F769-PWM-output-when-upd/]

For glitch-free pwm the pwmout_write() function, when loading new duty, should simply update the timer CCR register.  Instead it ends up reinitializing the timer (I believe even turning it off briefly) which can cause output to glitch if things align the right way.

Present in all versions of mbed-os 5.
All toolchains.  Have observed in hardware on GCC and mbed-os online compiler.

Hardware supports glitch-free PWM it seems like the software should as well.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-06-03:
cc @ARMmbed/team-st-mcd

jeromecoutant, 2019-06-13:
@GrahamSayers please check #10824

jeromecoutant, 2019-06-13:
@0xc0170 you can set devices st label

PR #10824 (2019-06-28): STM32: pwmout_write: configure channel only when not already enabled
### Description
STM32: pwmout_write: configure channel only when not already enabled
to fix PWM glitch on write(), TARGET_STM/pwmout_api.c.

Fixes #10734
### Pull request type

    [x ] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

### Reviewers

### Release Notes
closed 2019-06-28 by 0xc0170