ARMmbed/mbed-os #10735: Stack address on LPC55S69 is incorrect

mmahadevan108, 2019-05-31:
### Description
The stack on LPC55S69 is designed to be located in the address range 0x20043C00 - 0x20044000. The stack address is correct upto the call to osKernelStart. 
After calling this function the stack address is incorrect and not in the above range (e,g 0x200241C8).

This is seen with ARMC6 compiler using the uvision IDE

### Issue request type
    [X] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-06-03:
Is it being overwritten during Kernel startup ?

mmahadevan108, 2019-06-03:
Seems like it, I am not clear who is changing it.

mmahadevan108, 2019-06-03:
On reading further, looks like there are 2 stack pointers 1. Main stack pointer 2. Process Stack Pointer.

I would like to increase the size of the Process Stack, can you let me know how to do this.

0xc0170, 2019-06-04:
> I would like to increase the size of the Process Stack, can you let me know how to do this.

What is happening with PSP?  Not enough thread stack size?

mmahadevan108, 2019-06-04:
Stack overflow exception when running an application.

0xc0170, 2019-06-04:
rtos contains `mbed_lib.json` - configuration to tweak this - as this is application specific, please look at it.

0xc0170, 2019-08-13:
@mmahadevan108 Is there any update , was an application config changed to increase the stack ?

mmahadevan108, 2019-08-13:
This is no longer a issue, I was able to configure it

closed 2019-08-13 by 0xc0170