ARMmbed/mbed-os #10736: Debugging Psoc6 in vscode

farrenv, 2019-05-31:
### Description

Target: Psoc6, prototype dev kit, CY8CPROTO-062-4343W

Is anyone able to debug the PSoC 6 Wi-Fi BT Prototyping Kit in vscode?  With the use of pyocd 0.16.1 and cppdebug I am able to get to line 293 of startup_psoc6_02_cm4.S:

```
Cy_OnResetUser:
    bx lr
    .size   Cy_OnResetUser, . - Cy_OnResetUser
    .endfunc
```

pyocd gdbserver initializes two connections: one on port 3334 and one on port 3333.  I am assuming there is a port per core.  I am trying to debug specifying port 3334 assuming that is for the m4 core used for application.

Has anyone had success with debugging this platform in vscode?

### Issue request type

    [ x] Question
    [ ] Enhancement
    [ ] Bug

40Grit, 2019-05-31:
@flit @maclobdell

loverdeg-ep, 2019-06-03:
@farrenv Last I remember there was a resolution to this?

maclobdell, 2019-08-15:
@farrenv - is this resolved now?  sorry there was no response on this ticket.  We are definitely able to debug with pyocd gdb-server directly or with Eclipse and pyocd gdb-server.  I believe Cypress is also testing VSCode successfully now.  Please test out the latest version of pyOCD as I know there have been fixes.  Please close this ticket if you consider it resolved.