ARMmbed/mbed-os #10737: Cellular PSM - Wake Module Provisions?

AGlass0fMilk, 2019-05-31:
### Description

@AriParkkila @mirelachirica 

My team is working on using the PSM functionality of the ME910 cellular module (see driver PR here: https://github.com/ARMmbed/mbed-os/pull/10484). We have been attempting to use it with the mbed cloud client. 

We are experiencing issues where Mbed's network stack loses/drops the PPP connection with the cellular module. We found an app note from Telit:
[Telit_ME910C1_PSM_App_Note_r0.pdf](https://github.com/ARMmbed/mbed-os/files/3242363/Telit_ME910C1_PSM_App_Note_r0.pdf)

that explains the cell module essentially goes into deep sleep when it enters PSM. We think this is likely the cause of the dropped PPP connection. To wake the module, you have to pull a power on pin low for 5 seconds and probably reinitiate the PPP connection.

My question is this:
Does Mbed's cellular framework have provisions in place to handle this kind of behavior? If not, it seems like it would be a pretty common requirement of PSM-enabled designs (otherwise it would defeat the purpose of PSM). Therefore, the framework should support reinitializing the cellular module and PPP connection when a network request is made during PSM.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ ] Bug

trowbridgec, 2019-05-31:
@maclobdell

lauri-piikivi, 2019-06-04:
hi, so far the PSM has been application controlled. App has the context that it can deep sleep the modem. App will also pause pelion cloud client. then the client must wake up the modem or modem wakes up on its own timer. In the case where this was done the modem did keep all state including the ppp. Wake up was automatic with low power uart. 

this seems an area where there is some variation between modem implementations — lpuart or specific pins and differences in what state is kept over deep sleep. It would be good to understand if you run into ppp timeouts or is there some  higher level protocol timeout that then fails as modem is in deep sleep. As stated we did not have this problem so it might be some higher level protocol periodic keepalive. Or did is a case where we need new ppp negotiation. 

Ppp has some timeout definitions in https://github.com/ARMmbed/lwip/blob/master/src/include/netif/ppp/ppp_opts.h but to my understanding those have not been adjusted

trowbridgec, 2019-06-04:
@lauri-piikivi I found section in the Pelion client documentation which follows what you're saying and looks promising: https://www.pelion.com/docs/device-management/current/connecting/device-guidelines.html#client-pause-and-resume

I'll report back how it works out for us.

Thanks for the tip!

AriParkkila, 2019-06-05:
@trowbridgec That Pelion doc looks good. An application needs to disconnect before sleep and connect after wake-up:
7. Shut down the network stack => NetworkInterface::disconnect(), also disconnects PPP
9. Set up a network stack => NetworkInterface::connect()

Currently, it's up to the application to wake-up the modem before `connect()` and also if the modem tries to enter deep-sleep due to a longer period of network inactivity.

AGlass0fMilk, 2019-06-11:
I think this has been addressed. Thank you!

closed 2019-06-11 by AGlass0fMilk