ARMmbed/mbed-os #10743: UBLOX_EVK_ODIN_W2: Fatal run-time error in multihoming test

kimlep01, 2019-06-03:
### Description
UBLOX_EVK_ODIN_W2 with network-multihoming.tests-network-multihoming test case ends up in a fatal run-time error. Happens with all three compilers.
`
[1559524666.25][HTST][INF] CALLBACKs updated
[1559524666.25][HTST][INF] host test detected: default_auto
[1559524672.47][CONN][RXD] 
[1559524672.47][CONN][RXD] 
[1559524672.48][CONN][RXD] ++ MbedOS Error Info ++
[1559524672.57][CONN][RXD] Error Status: 0x80FF0100 Code: 256 Module: 255
[1559524672.57][CONN][RXD] Error Message: Fatal Run-time error
[1559524672.67][CONN][RXD] Location: 0x802C81D
[1559524672.67][CONN][RXD] Error Value: 0x0
[1559524672.78][CONN][RXD] Current Thread: <unnamed>  Id: 0x2000C414 Entry: 0x8029E09 StackSize: 0x200 StackMem: 0x2000C4A0 SP: 0x2000C560 
[1559524672.88][CONN][RXD] For more info, visit: https://mbed.com/s/error?error=0x80FF0100&osver=999999&core=0x410FC241&comp=1&ver=6110004&tgt=UBLOX_EVK_...
[1559524672.88][CONN][RXD] 
[1559524672.88][CONN][RXD] -- MbedOS Error Info --
[1559524672.98][CONN][RXD] RTC initialization failed
`
Link to full report: http://mbed-os-ci.s3-website-eu-west-1.amazonaws.com/?prefix=jenkins-ci/ARMmbed/mbed-os/mbed-os-ci-nightly/artifacts/master/224/greentea-test-IPCore/69/FAIL/UBLOX_EVK_ODIN_W2/

cc @ARMmbed/team-ublox 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-06-07:
@ARMmbed/team-ublox Please let us know if you are able to reproduce this

aqib-ublox, 2019-06-11:
@kimlep01 let me try

aqib-ublox, 2019-06-11:
@kimlep01 this test is new one and i think it never succeeded before am i right?

aqib-ublox, 2019-06-11:
@kimlep01 does this test require bridge support?

kimlep01, 2019-06-11:
For the test to run successfully you need to have both Ethernet and Wifi for the board. Jumper J21 should be shorted. Edit file under mbed-os/tools/test_configs/WiFiInterface.json and set wifi-secure-ssid and wifi-password in the config option.

Compile:
mbed test --compile -t ARM -m UBLOX_EVK_ODIN_W2 -n tests-network-multihoming --app-config=tools/test_configs/WiFiInterface.json

Run:
mbedgt -m UBLOX_EVK_ODIN_W2 -n tests-network-multihoming -V

I noticed we had an issue in our CI and wrong board without Ethernet was allocated for testing. Still the test should not crash like this. I have fixed this issue and will see the results tomorrow.

The problem is that I cannot reproduce the issue on my desk when I disconnect Ethernet. The test keeps crashing every night with all compilers in CI. Please see if you can reproduce this.

aqib-ublox, 2019-06-12:
@kimlep01 i have also tried without ethernet connection and not getting crash like above although it is waiting for connection to bring up and as result timeout.

Above scenerio is not reproducible.

kimlep01, 2019-06-12:
I will close this as it cannot be reproduced. I created a second issue which I can reproduce in our lab: https://github.com/ARMmbed/mbed-os/issues/10815

closed 2019-06-12 by kimlep01

AmmadRehmat, 2019-06-21:
@kimlep01  did the tests pass for you?

kimlep01, 2019-06-24:
Yes the tests are passing for me when Ethernet is connected and Wifi settings are configured. I was able to reproduce this failure in the end by first using the following macros while compiling:
```
-DMBED_HEAP_STATS_ENABLED=1 -DMBED_STACK_STATS_ENABLED=1 -DMBED_TRAP_ERRORS_ENABLED=1 -DMBED_ALL_STATS_ENABLED -DSKIP_TIME_DRIFT_TESTS=1
```

Then it turned out we had used a bad jumper in J21. After replacing the jumper we had no more issues.