ARMmbed/mbed-os #10745: Is the USB Host supported?

desowin, 2019-06-03:
### Description

I would like to connect the USB flash memory to the STM32F746G-DISCO board. The device would save the logs to the USB memory stick. Making the USB type-A port available on target device enclosure is the most convenient option in our project. I have noticed that there is unsupported USB Host which could be used. I have done the minimal changes required to get it working on the USB FS port of the discovery board and submitted the changes [1] [2] [3]. 

The USB HS port on STM32F746G-DISCO does not work reliably with mbed unsupported USB Host stack as the driver does not properly support the NYET PING protocol required for high speed devices (it could be solved either by implementing the NYET PING protocol in software or implementing the DMA, in which case the hardware takes care of pinging the device).

In a comment under [2], @0xc0170 wrote "Note, we have new USB stack that should be used. This PR touches the old unsupported one."

While there is a supported USB device stack, I didn't find any other USB host stack than the unsupported one. Is it just that @0xc0170 simply confused the host and device parts or is the new USB host stack hidden well enough that I cannot find it?

[1] https://github.com/ARMmbed/mbed-os/pull/10651
[2] https://github.com/ARMmbed/mbed-os/pull/10653
[3] https://github.com/ARMmbed/mbed-os/pull/10654

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

40Grit, 2019-06-03:
@AGlass0fMilk

maciejbocianski, 2019-06-05:
The new one mentioned by @0xc0170 is here 
https://github.com/ARMmbed/mbed-os/tree/master/usb/device
But for now it supports slave only and I don't know what are the plans for the USB host support

0xc0170, 2019-10-25:
Thanks @maciejbocianski , I'll close this question.

closed 2019-10-25 by 0xc0170