ARMmbed/mbed-os #10748: Python Dependencies issues - Mbed O.S tool

jainvikas8, 2019-06-03:
mbedls - Version: 1.7.7
mbed - Version: 1.8.3
Platform - Windows 10
mbed-os (#73f096399b4c, tags: latest, mbed-os-5.12.4)

When using mbed cli to run a test:
**mbed test -t ARMC6 -m `DISCO_L475VG_IOT01A` -n tests-mbedmicro-rtos-mbed-basic --run**

End up getting an error:
```
Build` successes:
  * DISCO_L475VG_IOT01A::ARMC6::MBED-BUILD
  * DISCO_L475VG_IOT01A::ARMC6::TESTS-MBEDMICRO-RTOS-MBED-BASIC
Traceback (most recent call last):
  File "C:\Python27\Scripts\mbedgt-script.py", line 6, in <module>
    from pkg_resources import load_entry_point
  File "C:\Python27\lib\site-packages\pkg_resources\__init__.py", line 3017, in <module>
    @_call_aside
  File "C:\Python27\lib\site-packages\pkg_resources\__init__.py", line 3003, in _call_aside
    f(*args, **kwargs)
  File "C:\Python27\lib\site-packages\pkg_resources\__init__.py", line 3030, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
  File "C:\Python27\lib\site-packages\pkg_resources\__init__.py", line 661, in _build_master
    return cls._build_from_requirements(__requires__)
  File "C:\Python27\lib\site-packages\pkg_resources\__init__.py", line 674, in _build_from_requirements
    dists = ws.resolve(reqs, Environment())
  File "C:\Python27\lib\site-packages\pkg_resources\__init__.py", line 858, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (mbed-os-tools 0.0.8 (c:\python27\lib\site-packages), Requirement.parse('mbed-os-tools==0.0.1'), set(['mbed-host-tests']))
[mbed] ERROR: "mbedgt" returned error.
       Code: 1
       Path: "C:\Work\Workspace\mbed-os"
       Command: "mbedgt --test-spec .\BUILD\tests\DISCO_L475VG_IOT01A\ARMC6\test_spec.json -n tests-mbedmicro-rtos-mbed-basic"
       Tip: You could retry the last command with "-v" flag for verbose output
```

MarceloSalazar, 2019-06-03:
I'm seeing the same issue all the time with installing a new version of Mbed OS :(
It seems the packages in Mbed OS require different versions of the tools from mbed-os-tools.
As a workaround, I need to manually install specific versions as shown here: https://github.com/ARMmbed/mbed-os-tools/releases/tag/v0.0.8

Please fix the python dependencies, as this issue is very annoying ;-)

loverdeg-ep, 2019-06-03:
https://github.com/ARMmbed/cmsis-pack-manager/issues/110#issuecomment-497025151

@maclobdell Would love to see the community and partners approached in order to solve Mbed OS tooling issues for the long term.

loverdeg-ep, 2019-06-03:
https://github.com/ARMmbed/mbed-os/issues/9479#issuecomment-480836993
https://github.com/ARMmbed/mbed-os/issues/7887#issuecomment-498304269

loverdeg-ep, 2019-06-03:
Just in.

[image]

@trowbridgec

loverdeg-ep, 2019-06-03:
@theotherjimmy

0xc0170, 2019-06-04:
This was fixed on master, and will be in 5.13. For older versions, the best is to fix tools dependency tree. 

@ARMmbed/mbed-os-tools already are aware of this

linlingao, 2019-06-06:
@0xc0170 Can you link the PR to this issue so that we can close it?

linlingao, 2019-06-11:
@jainvikas8 Can you verify the fix on master?

jainvikas8, 2019-06-12:
@linlingao  What versions of the tools do I need to update with?
Followed - https://github.com/ARMmbed/mbed-os-tools/releases/tag/v0.0.8 to fix my tooling issue.

linlingao, 2019-06-12:
It appears the following commit fixed the issue:
https://github.com/ARMmbed/mbed-os/commit/9bc7e7c60be806b7dc22eadcb09df80c501609dc

If you update requirements.txt under mbed-os, the dependency issue should be fixed.

jainvikas8, 2019-06-17:
@linlingao, Tested on [commit](https://github.com/ARMmbed/mbed-os/commit/cd8e315ba5875cf5af9f0489eb95b67f1b554fce).
`mbed test -t ARMC6 -m DISCO_L475VG_IOT01A -n tests-mbedmicro-rtos-mbed-basic --run`

linlingao, 2019-06-17:
Please close it if it worked.

jainvikas8, 2019-06-17:
Closing issue.

MarceloSalazar, 2019-07-03:
This continues to be a problem with Mbed OS 5.13.0

Reproduce:
```
mbed new test
cd test
mbed test -n *echo*

Traceback (most recent call last):
  File "/usr/local/bin/mbedgt", line 10, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/site-packages/mbed_greentea/mbed_greentea_cli.py", line 332, in main
    gt_logger.gt_log(get_hello_string())
  File "/usr/local/lib/python2.7/site-packages/mbed_greentea/mbed_greentea_cli.py", line 100, in get_hello_string
    version = get_greentea_version()
  File "/usr/local/lib/python2.7/site-packages/mbed_greentea/mbed_greentea_cli.py", line 89, in get_greentea_version
    version = pkg_resources.require("mbed-greentea")[0].version
  File "/usr/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 900, in require
    needed = self.resolve(parse_requirements(requirements))
  File "/usr/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 791, in resolve
    raise VersionConflict(dist, req).with_context(dependent_req)
pkg_resources.ContextualVersionConflict: (mbed-os-tools 0.0.8 (/usr/local/lib/python2.7/site-packages), Requirement.parse('mbed-os-tools==0.0.6'), set(['mbed-greentea']))

```

0xc0170, 2019-08-13:
Is this still an issue? It's been some time to propagate proper versions.

MarceloSalazar, 2019-08-13:
I don't see this issue anymore when using 5.13.2+

closed 2019-08-13 by MarceloSalazar