ARMmbed/mbed-os #10754: Assertion failed: pcache->block == 0xffffffff

martinichka, 2019-06-04:
### Description

We get an assert when using LittleFileSystem with SPIFBlockDevice. We do not have that problem when using SDBlockDevice and HeapBlockDevice.

[crashdump.txt](https://github.com/ARMmbed/mbed-os/files/3250883/crashdump.txt)

Fails with the following assert:
```
Error Message: Assertion failed: pcache->block == 0xffffffff
File: ./mbed-os/features/storage/filesystem/littlefs/littlefs/lfs.c+183
```
Suspicious log lines before that: `lfs debug:1411: Bad block at 4` (followed by 6,7,7)

Processor: STM32L496VE
Flash memory: Spansion fs512s (512MBit)
Compiler: gcc-arm-none-eabi-6-2017-q2-update

Tested on MBED-os 5.11, 5.12 and the master:
a4738fa9a8 (HEAD -> master, origin/master, origin/HEAD) Merge pull request #10540 from c1728p9/fpga_ci_test_shield
73f096399b (HEAD -> mbed-os-5.12, tag: mbed-os-5.12.4, tag: latest, origin/mbed-os-5.12) Merge pull request #10610 from ARMmbed/release-candidate
51d55508e8 (HEAD -> mbed-os-5.11, tag: mbed-os-5.11.5, origin/mbed-os-5.11) Merge pull request #9805 from ARMmbed/release-candidate

#### Steps to reproduce
```C++
bd.init(); // succeeds

fs.reformat(&bd); // format before first time usage

fs.mount(&bd); // succeeds
while (X) // loop multiple times 
{
    FILE *f =fopen(SOME_FILE, "a"); // succeeds
    fwrite("data", 4, 1, f); // succeeds after first erase
    fclose(f); // succeeds 
}
/*
 * power cycle
 */
bd.init(); // succeeds
fs.mount(&bd); // succeeds
FILE *f =fopen(SOME_FILE, "a"); // succeeds
fwrite("data", 4, 1, f); // fails with assert 
```
### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

martinichka, 2019-06-04:
Could reproduce the issue with mbed-os-example-filesystem by adding a few lines of code. I pushed the added code here: https://github.com/ARMmbed/mbed-os-example-filesystem/compare/master...martinichka:debug/assert_lfs_bad_block

Looping to write some data:
```C++
    const char data[] = "some data";
    for (int i = 0; i < 1000; i++)
    {
	    writeData(data, sizeof(data));
    }
```
by opening each time the file before write:
```C++
void writeData(const char *data, uint16_t len)
{
    FILE *f = fopen("/fs/somefile.txt", "a");
    tr_info("%s\n", (!f ? "Fail :(" : "OK"));
    if (!f) {
        tr_error("error while reading file, does it exist?: %s (%d)\n", strerror(errno), -errno);
    }

    tr_info("Writing data to flash");
    fwrite((const char*)data, len, 1, f);

    int err = fclose(f);
    tr_info("%s\n", (err < 0 ? "Fail :(" : "OK"));
    if (err < 0) {
        tr_error("error: %s (%d)\n", strerror(errno), -errno);
    }
}
```

martinichka, 2019-06-04:
I have to note that the vanilla example does not crash with an assert and the numbers are as expected, but it does report `bad block` errors. 

First run
```
--- Mbed OS filesystem example ---
Mounting the filesystem... OK
Opening "/fs/numbers.txt"... Fail :(
No file found, creating a new file... OK
Writing numbers (10/10)... OK
Seeking file... OK
Incrementing numbers (3/10)... lfs debug:1411: Bad block at 8
lfs debug:1411: Bad block at 9
lfs debug:1411: Bad block at 10
lfs debug:1411: Bad block at 11
lfs debug:1411: Bad block at 12
lfs debug:1411: Bad block at 13
lfs debug:1411: Bad block at 14
lfs debug:1411: Bad block at 15
lfs debug:1411: Bad block at 16
lfs debug:1411: Bad block at 17
lfs debug:1411: Bad block at 18
lfs debug:1411: Bad block at 19
lfs debug:1411: Bad block at 20
lfs debug:1411: Bad block at 21
lfs debug:1411: Bad block at 22
lfs debug:1411: Bad block at 23
lfs debug:1411: Bad block at 24
lfs debug:1411: Bad block at 25
lfs debug:1411: Bad block at 26
lfs debug:1411: Bad block at 27
Incrementing numbers (10/10)... OK
Closing "/fs/numbers.txt"... OK
Opening the root directory... OK
root directory:
    .
    ..
    numbers.txt
Closing the root directory... OK
Opening "/fs/numbers.txt"... OK
numbers:
    1
    2
    3
    4
    5
    6
    7
    8
    9
    10
Closing "/fs/numbers.txt"... OK
Unmounting... OK
Mbed OS filesystem example done!
```
Second run
```
--- Mbed OS filesystem example ---
Mounting the filesystem... OK
Opening "/fs/numbers.txt"... OK
Incrementing numbers (4/10)... lfs debug:1411: Bad block at 8
lfs debug:1411: Bad block at 9
lfs debug:1411: Bad block at 10
lfs debug:1411: Bad block at 11
lfs debug:1411: Bad block at 12
lfs debug:1411: Bad block at 13
lfs debug:1411: Bad block at 14
lfs debug:1411: Bad block at 15
lfs debug:1411: Bad block at 16
lfs debug:1411: Bad block at 17
lfs debug:1411: Bad block at 18
lfs debug:1411: Bad block at 19
lfs debug:1411: Bad block at 20
lfs debug:1411: Bad block at 21
lfs debug:1411: Bad block at 22
lfs debug:1411: Bad block at 23
lfs debug:1411: Bad block at 24
lfs debug:1411: Bad block at 25
lfs debug:1411: Bad block at 26
lfs debug:1411: Bad block at 27
lfs debug:1411: Bad block at 29
lfs debug:1411: Bad block at 30
lfs debug:1411: Bad block at 31
lfs debug:1411: Bad block at 32
lfs debug:1411: Bad block at 33
lfs debug:1411: Bad block at 34
lfs debug:1411: Bad block at 35
Incrementing numbers (10/10)... OK
Closing "/fs/numbers.txt"... OK
Opening the root directory... OK
root directory:
    .
    ..
    numbers.txt
Closing the root directory... OK
Opening "/fs/numbers.txt"... OK
numbers:
    2
    3
    4
    5
    6
    7
    8
    9
    10
    11
Closing "/fs/numbers.txt"... OK
Unmounting... OK
Mbed OS filesystem example done!
```

martinichka, 2019-06-04:
I tried with another manufacter SPIF chip: MT25QU01GBBB (1Gbit) which does not seem to have the problem. So it seems it related to the **Spansion fs512s (512MBit)** only.

martinichka, 2019-06-04:
Can someone verify if other Spansion chips have similar problems?

adbridge, 2020-10-15:
@martinichka is this still an issue with Mbed 6 ? (Contains the upgraded LittleFS)

martinichka, 2020-10-30:
Hi @adbridge, sorry for the late response. 

Why the @ciarmcom bot is requesting more information? All information on which hardware, which MBedOS, tools and steps to reproduce are in the original bug report.

This is an issue of more than one year ago. We solved it by using another flash chip. Current we don't have the PCB available with the flash chip causing the issue, so I am not able to check if the issue is resolved in MbedOS 6. Therefore I will close this issue.

Best regards,
Marten

closed 2020-10-30 by martinichka

adbridge, 2020-11-02:
@martinichka thanks for the update and closing this issue.