ARMmbed/mbed-os #10760: Repeated macro in FLEXSPI_SetFlashConfig()

sjwalke, 2019-06-05:
### Description

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X ] Bug

sjwalke, 2019-06-05:
Developing for iMXRT 1064 EVK, statis analysis tools have found this issue, have reported on NxP community as well.

 In function FLEXSPI_SetFlashConfig()

configValue &= ~(FLEXSPI_FLSHCR2_AWRWAITUNIT_MASK | FLEXSPI_FLSHCR2_AWRWAIT_MASK | FLEXSPI_FLSHCR2_AWRSEQNUM_MASK |
 FLEXSPI_FLSHCR2_AWRSEQID_MASK | FLEXSPI_FLSHCR2_ARDSEQNUM_MASK | FLEXSPI_FLSHCR2_AWRSEQID_MASK);

The macro FLEXSPI_FLSHCR2_AWRSEQID_MASK appears twice in the expression (looks like a copy 'n paste error at the end of the line).

Should the second occurrence be using the sequence ID read mask instead of the write mask, i.e. using FLEXSPI_FLSHCR2_**ARD**SEQID_MASK.

0xc0170, 2019-06-05:
I believe the community report should be sufficient ot have this fixed in the driver itself. We would benefit from it when the driver is updated.

cc @mmahadevan108

sjwalke, 2019-06-07:
cc @MaureenHelm in case this is relevant to Zephyr

MaureenHelm, 2019-06-07:
> cc @MaureenHelm in case this is relevant to Zephyr

@sjwalke Thanks for the heads up. Will you please share a link to the community report?
@AntoineZen This may affect zephyrproject-rtos/zephyr#15364

sjwalke, 2019-06-07:
@MaureenHelm - here you go
https://community.nxp.com/message/1160708

MaureenHelm, 2019-06-07:
> @MaureenHelm - here you go
> [community.nxp.com/message/1160708](https://community.nxp.com/message/1160708)

Thank you!

0xc0170, 2019-10-25:
@mmahadevan108 has this been fixed in this code base (driver updated) ?

mmahadevan108, 2019-10-25:
Yes, this has been fixed. I am working on a PR to update the MIMXRT SDK drivers, it should be coming shortly

maclobdell, 2020-03-27:
Fixed by https://github.com/ARMmbed/mbed-os/pull/12095

please re-open if still an issue

closed 2020-03-27 by maclobdell