ARMmbed/mbed-os #10765: TRNG Greentea Test requires manual RESET for target EP_AGORA

trowbridgec, 2019-06-05:
### Description

While running the Mbed OS Greentea tests to validate adding the `EP_AGORA` target, I found that I had to manually press the RESET button during the `tests-mbed_hal-trng` test in order for it to pass (see PR #10629).

The 'Agora' (`EP_AGORA`) board does not have a DAPLink circuit on board; we debug and program it via an nRF52840 DK running a modified version of DAPLink that forces the interface chip to use the external debug port. I'm not positive, but that could be related?

Below I've attached some verbose logs of the test failing on `GCC_ARM` and `IAR`. It seems like the host test hangs waiting after sending the 2nd `sync`. If I manually RESET the target right after the `sync` is sent, the test continues and passes without issue.
* `IAR`: [EP_AGORA_IAR_tests-mbed_hal-trng-failure.txt](https://github.com/ARMmbed/mbed-os/files/3257275/EP_AGORA_IAR_tests-mbed_hal-trng-failure.txt)
* `GCC_ARM`:  [EP_AGORA_GCC_ARM_tests-mbed_hal-trng-failure.txt](https://github.com/ARMmbed/mbed-os/files/3257285/EP_AGORA_GCC_ARM_tests-mbed_hal-trng-failure.txt)

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-06-05:
This looks like the interface/host test problem , not related  to trng - however do not understand how this is not seen in other tests (the start sequence is different, should not be prior sync packet?).

linlingao, 2019-06-06:
The TRNG test initiates a hardware reset in the middle of the test to verify the randomly generated data is indeed random. By default, it requires DAPLink to intercept break to initiate the reset. If there's no DAPLink, it will fail. If it hangs after the 2nd sync, it typically means the reset did not go through. There is another alternative to initiates reset using pyOCD but that requires some hack.

loverdeg-ep, 2019-06-06:
@flit any insight?

flit, 2019-06-06:
I think @linlingao explained it. If the serial port is not going through DAPLink, then the serial break sent by the test host will not be intercepted to reset the target.

loverdeg-ep, 2019-06-06:
Sounds like the hack method might be less desirable than getting hard reset working with the current or other adapter.

linlingao, 2019-06-06:
Yeah, I would recommend getting DAPLink to work since it's an integral part of Mbed.

trowbridgec, 2019-06-07:
That makes sense because we are using a separate UART for communicating with the DUT (utilizing the mbedls mocking/override functionality).

linlingao, 2019-06-07:
@trowbridgec Do you mind closing this?

loverdeg-ep, 2019-06-07:
@linlingao @trowbridgec is PTO today.

Go ahead and close. Since we have debugger on a baseboard board, it is resolvable without source changes.

closed 2019-06-07 by linlingao