ARMmbed/mbed-os #10766: Disabled unstable time drifting tests cases

mprse, 2019-06-05:
### Description

In the following tests, time-drifting test cases will be disabled for now since they are unstable on CI (host machines is in AWS in Ireland and devices in Oulu and there is internet between those which quality changing all the time):
- `tests-mbed_drivers-lp_timeout` (PR https://github.com/ARMmbed/mbed-os/pull/10762)
- `tests-mbed_drivers-ticker` (PR https://github.com/ARMmbed/mbed-os/pull/10762)
- `tests-mbed_drivers-timeout` (PR https://github.com/ARMmbed/mbed-os/pull/10762)
- `tests-mbedmicro-rtos-mbed-basic` (https://github.com/ARMmbed/mbed-os-ci/blob/e9318eb6cbe3d70fcc4a9274f2052ffa46999fdf/vars/greentea.groovy#L34)
- `tests-mbed_hal-common_tickers_freq` (https://github.com/ARMmbed/mbed-os-ci/blob/e9318eb6cbe3d70fcc4a9274f2052ffa46999fdf/vars/greentea.groovy#L34)

The idea for the future is to use FPGA test shield for timing tests instead host scripts which are unstable (communication latency).

### Issue request type

    [ ] Question
    [X] Enhancement
    [ ] Bug

0xc0170, 2019-08-14:
@mprse is there any other action for this issue after #10762 was merged?

mprse, 2019-08-14:
This issue is solved and can be closed.

closed 2019-08-14 by 0xc0170