ARMmbed/mbed-os #10768: I2C recover()

pilotak, 2019-06-05:
### Description
I would like to reset I2C from SCL being low after an error. I found that **I2C** has `recover()` option but it is defined as private. Fine but because it's defined as `NonCopyable` i can't reinitialize it either.
Could please guide me how to reset I2C?

```cpp
#include <mbed.h>

I2C i2c(I2C_SDA_pin, I2C_SCL_pin);

int main() {
    ...
    someLib.init(&i2c);
    ...

    i2c.recover(); // fails because this it's private
    i2c = I2C(I2C_SDA_pin, I2C_SCL_pin); // fails because it's NonCopyable

    someLib.init(&i2c);

    while (1) {
        ...
    }
}
```

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-06-06:
I don't recall why recover was made private..

@ARMmbed/mbed-os-hal 

Workaround: Wouldn't invoking explicitly constructor do the job, however initializing everything again (frequency would need to be set up again).

pilotak, 2019-06-06:
@0xc0170 did you mean it like that?
```cpp
#include <mbed.h>

I2C i2c(I2C_SDA_pin, I2C_SCL_pin);

int main() {
    ...
    someLib.init(&i2c);
    ...

    I2C(I2C_SDA_pin, I2C_SCL_pin);

    i2c.frequency(400000);

    someLib.init(&i2c);  // mandatory

    while (1) {
        ...
    }
}
```

pilotak, 2019-06-10:
@0xc0170 when i try to reset I2C with above code I get an error:
```
++ MbedOS Error Info ++
Error Status: 0x80010101 Code: 257 Module: 1
Error Message: Tried to free invalid CThunkEntry
Location: 0x803DF85
File: CThunkBase.cpp+142
Error Value: 0x0
Current Thread: application_unnamed_thread  Id: 0x20006D88 Entry: 0x8041D99 StackSize: 0x1000 StackMem: 0x20009BF8 SP: 0x2000AA1C
Next:
application_unnamed_thread  State: 0x2 Entry: 0x08041D99 Stack Size: 0x00001000 Mem: 0x20009BF8 SP: 0x2000AA98
Ready:
application_unnamed_thread  State: 0x1 Entry: 0x08041D99 Stack Size: 0x00001000 Mem: 0x2000AC00 SP: 0x2000BB50
rtx_idle  State: 0x1 Entry: 0x0803FCB5 Stack Size: 0x00000200 Mem: 0x20005990 SP: 0x20005B38
Wait:
rtx_timer  State: 0x83 Entry: 0x08041C85 Stack Size: 0x00000300 Mem: 0x20005690 SP: 0x20005928
Delay:
main  State: 0x13 Entry: 0x0803FBC7 Stack Size: 0x00001000 Mem: 0x20005F68 SP: 0x20006B40
application_unnamed_thread  State: 0x1 Entry: 0x08041D99 Stack Size: 0x00001000 Mem: 0x2000AC00 SP: 0x2000BB50
rtx_idle  State: 0x1 Entry: 0x0803FCB5 Stack Size: 0x00000200 Mem: 0x20005990 SP: 0x20005B38
For more info, visit: https://mbed.com/s/error?error=0x80010101&tgt=NUCLEO_F412ZG
-- MbedOS Error Info --
```

maciejbocianski, 2019-06-10:
I think the code is wrong
If you want to re-init I2C  it should be allocated dynamically
Try this one:

```c++
#include <mbed.h>

I2C *i2c = NULL;

int main() {
    i2c = new I2C(I2C_SDA_pin, I2C_SCL_pin);
    ...
    someLib.init(i2c);
    ...

    delete i2c;
    i2c = new I2C(I2C_SDA_pin, I2C_SCL_pin);

    i2c.frequency(400000);

    someLib.init(i2c);  // mandatory

    while (1) {
        ...
    }
   delete i2c;
}
```

pilotak, 2019-06-10:
thanks a lot i will try this out

pilotak, 2019-06-11:
@maciejbocianski can it be used like that? so that size in RAM is known at compile time and i don't have to reinitialize libraries
```cpp
#include <mbed.h>

I2C *i2c = NULL;
uint32_t i2c_buffer[sizeof(I2C) / sizeof(uint32_t)];

int main() {
    i2c = new(i2c_buffer) I2C(I2C_SDA_pin, I2C_SCL_pin);
    ...
    someLib.init(i2c);
    ...

    i2c->~I2C();
    i2c = new(i2c_buffer) I2C(I2C_SDA_pin, I2C_SCL_pin);

    i2c->frequency(400000);

    someLib.init(i2c);  // optional as I2C remains at the same "place"

    while (1) {
        ...
    }

    delete i2c;
}
```

pilotak, 2019-06-24:
it looks like it works with above. Hopefully someone will find this useful in future

closed 2019-06-24 by pilotak