ARMmbed/mbed-os #10771: ARM_MUSCA_A1: psa-crypto_init test failed

ccli8, 2019-06-06:
### Description

#### Environment

**Target**
ARM_MUSCA_A1

**Toolchain**
Arm Compiler 6.10

**mbed-os**
fbc489e6a5 (HEAD -> master, origin/master, origin/HEAD) Merge pull request #10733 from pan-/fix-nrf52-memory-pools

#### Steps to reproduce

1.  Re-build secure code with:
    ```sh
    $ mbed compile -t ARMC6 -m ARM_MUSCA_A1_S --app-config mbed-os/tools/psa/tfm/mbed_app.json --profile release
    ```
1.  Test `mbed-os-tests-psa-crypto_init` and failed
    ```sh
    $ mbed test -m ARM_MUSCA_A1_NS -t ARMC6 -n mbed-os-tests-psa-crypto_init
    ```

The issue is first found on my NUMAKER_PFM_M2351 TFM port, and can reproduce on ARM_MUSCA_A1. Per my look, in **mbed-os/features/mbedtls/platform/inc/platform_mbed.h**:

```c++
/* Automatically enable the Mbed Crypto entropy injection API if
 * MBEDTLS_ENTROPY_NV_SEED is enabled. WARNING: the current implementation of
 * the Mbed Crypto entropy injection API is incompatible with other entropy
 * sources. When MBEDTLS_ENTROPY_NV_SEED is used on PSA target, the NV Seed is
 * the sole source of entropy and all other entropy sources are ignored. */
#define MBEDTLS_NO_DEFAULT_ENTROPY_SOURCES
#define MBEDTLS_PSA_INJECT_ENTROPY
```

All entropy sources except NV seed are ignored. But in **mbed-os/features/mbedtls/mbed-crypto\src/entropy.c** > **mbedtls_entropy_init**:

```c++
#if defined(MBEDTLS_ENTROPY_NV_SEED)
    mbedtls_entropy_add_source( ctx, mbedtls_nv_seed_poll, NULL,
                                MBEDTLS_ENTROPY_BLOCK_SIZE,
                                MBEDTLS_ENTROPY_SOURCE_STRONG );
    ctx->initial_entropy_run = 0;
#endif
#endif /* MBEDTLS_NO_DEFAULT_ENTROPY_SOURCES */
```

NV seed entropy source is also ignored due to within the ambit of `MBEDTLS_NO_DEFAULT_ENTROPY_SOURCES`. If I move the NV seed entropy source out of the `MBEDTLS_NO_DEFAULT_ENTROPY_SOURCES` ambit, the `mbed-os-tests-psa-crypto_init` passes. Besides the `mbed-os-tests-psa-crypto_init` test, many other PSA tests are also failed with it.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-06-06:
@ARMmbed/mbed-os-crypto Please review

0xc0170, 2019-06-06:
Is this related to this bugfix? #10770

ccli8, 2019-06-06:
It seems #10770 can fix the issue. I would try it.

ccli8, 2019-06-06:
Yes. #10770 can fix the issue.

0xc0170, 2019-06-06:
Great ! Thanks for the report and confirmation.

ccli8, 2019-06-10:
Closed via #10770

closed 2019-06-10 by ccli8