ARMmbed/mbed-os #10773: Serial IRQ handler called for disabled IRQs

fkjagodzinski, 2019-06-06:
### Description

When only the RX IRQ is enabled, the attached IRQ handler is called in both `RxIrq` (correct) and `TxIrq` (incorrect) contexts.  See code below:
```cpp
#include "mbed.h"
#include "serial_api.h"
 
typedef struct {
    serial_t *ser;
    uint32_t rx_cnt;
    uint32_t tx_cnt;
    bool dump_data;
} serial_test_data_t;

static void test_irq_handler(uint32_t id, SerialIrq event)
{
    serial_test_data_t *td = (serial_test_data_t *)id;
    if (event == RxIrq) {
        td->rx_cnt++;
        int c = serial_getc(td->ser);
        if (c == '\n') {
            td->dump_data = true;
        }
    }
    if (event == TxIrq) {
        td->tx_cnt++;
        // serial_putc(td->ser, '*');
    }
}

int main()
{
    puts("Please type a few chars (new line dumps IRQ counters).");
    puts("Note the TX IRQ is not enabled, expected tx_cnt is 0.");
    puts("");
    puts("");
    const int baudrate = 9600;
    const int data_bits = 8;
    const SerialParity parity = ParityNone;
    const int stop_bits = 1;
    const PinName tx = USBTX;
    const PinName rx = USBRX;
    const PinName cts = NC;
    const PinName rts = NC;

    bool use_flow_control = (cts != NC && rts != NC) ? true : false;

    serial_t serial;
    serial_init(&serial, tx, rx);
    serial_baud(&serial, baudrate);
    serial_format(&serial, data_bits, parity, stop_bits);
    if (use_flow_control) {
        serial_set_flow_control(&serial, FlowControlRTSCTS, rts, cts);
    } else {
        serial_set_flow_control(&serial, FlowControlNone, NC, NC);
    }

    volatile serial_test_data_t td = {
        &serial,
        0,
        0,
        false
    };

    serial_irq_handler(&serial, test_irq_handler, (uint32_t) &td);
    serial_irq_set(&serial, RxIrq, 1);

    uint32_t rx_cnt, tx_cnt;

    while (true) {
        rx_cnt = tx_cnt = 0;
        core_util_critical_section_enter();
        if (td.dump_data) {
            rx_cnt = td.rx_cnt;
            tx_cnt = td.tx_cnt;
            td.dump_data = false;
        }
        core_util_critical_section_exit();
        if (rx_cnt || tx_cnt) {
            printf("rx_cnt = %8lu\ttx_cnt = %8lu\r\n", rx_cnt, tx_cnt);
        }
    }

    serial_irq_set(&serial, RxIrq, 0);
    serial_free(&serial);
}
```

Runnig this code on the current `master` branch (https://github.com/ARMmbed/mbed-os/commit/2b354a40b068147f7ccddf15204d1e77a5b88ba5) with `mbed compile -t GCC_ARM -m K64F -f --sterm` produces the following output:
```
--- Terminal on /dev/ttyACM1 - 9600,8,N,1 ---
Please type a few chars (new line dumps IRQ counters).
Note the TX IRQ is not enabled, expected tx_cnt is 0.

foobar
rx_cnt =        8	tx_cnt =        8
another test
rx_cnt =       22	tx_cnt =       22
--- [QUIT] ---
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-06-06:
Good catch 👍

fkjagodzinski, 2019-06-06:
> Good catch +1

All thanks to FPGA shield. :sunglasses:

PR #10774 (2019-06-25): Fix serial IRQ handling 
### Description

Check that the RX or TX interrupt is enabled before calling a registered handler with `RxIrq` or `TxIrq` arg.

Fixes #10773

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

### Reviewers

### Release Notes
closed 2019-06-25 by 0xc0170