ARMmbed/mbed-os #10778: UBLOX cellular: missing a callback on socket closed

pilotak, 2019-06-06:
### Description
as per conversation #10687 

Once you are succesfully connected to server and server stops working you can see in debug:
```
[DBG ][CELL]: AT OoB readable 1, len 0
[INFO][CELL]: AT RX (14): <cr><ln>+UUSOCL: 0<cr><ln>
[DBG ][CELL]: AT OoB done
```
which is perfect but i have callback registered with `_socket.sigio(callback(this, &Test::socketCb));` and there is no cb called. If I try to listen for data `_socket.recv(buffer, size)` after i received above, no negative error is returned so it looks the socket is still open (from OS point of view).
The only way to check if the socket is still open is to write data to it _socket.send(data, size) - this will give you NSAPI_ERROR_DEVICE_ERROR (-3012)

The problem is that `UBLOX_AT_CellularStack` doesn't have a callback/not acting for this command received.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-06-07:
cc @ARMmbed/mbed-os-wan

0xc0170, 2019-06-07:
@ARMmbed/team-ublox

fahim-ublox, 2019-06-10:
Internal jira reference : PGM_NO_MBD-239

mudassar-ublox, 2019-06-12:
Updated callback for socket close. @pilotak can you please check it works fine now or anything else required.
https://github.com/ARMmbed/mbed-os/pull/10811

pilotak, 2019-06-12:
Thanks @mudassar-ublox it now works as expected

pilotak, 2019-06-12:
just a question though, i noticed that cb is only called on disconnect ie manual or now remote close. Is that the intended behaviour? or is there a chance it can be called on some other event? so i know i can treat it like that

mudassar-ublox, 2019-06-13:
Callback is also called on +UUSORD and +UUSORF.

pilotak, 2019-06-13:
ok and how would i get a reason why cb was called? `sigio` doesn't pass any argument

bqam-ublox, 2019-06-13:
To differentiate between what triggered the cb, you will need to keep a check on [rx_avail](https://github.com/ARMmbed/mbed-os/blob/master/features/cellular/framework/AT/AT_CellularStack.h#L120) and [closed](https://github.com/ARMmbed/mbed-os/blob/master/features/cellular/framework/AT/AT_CellularStack.h#L117) parameters from the CellularSocket class. @mudassar-ublox  will shortly generate another PR to update these parameters.

mudassar-ublox, 2019-06-17:
Updated socket flags.
[10840](https://github.com/ARMmbed/mbed-os/pull/10840)

pilotak, 2019-06-17:
thanks a lot ublox team, i hope that both PR will be merged shortly

closed 2019-06-17 by pilotak

pilotak, 2019-07-11:
@bqam-ublox @mudassar-ublox could i kindly ask you to show me an example how to retreive `rx_avail` and `closed`? 

```cpp
// TCPSocket _socket;

void Test::socketCb() {
    debug("socket change\n");
   // some magic here
}

void Test::connect(){
   ...
   _socket.sigio(callback(this, &Test::socketCb));
}
```

bqam-ublox, 2019-07-11:
@0xc0170 the AT_CellularStack class is using these parameters, have a look here: https://github.com/ARMmbed/mbed-os/blob/master/features/cellular/framework/AT/AT_CellularStack.cpp#L252
Not sure how you can use them in your application. Could you please create a new issue on mbed with more details. The ARM guys might help to figure out if there is already any example or test case present. Thanks

pilotak, 2019-07-11:
will do thanks