ARMmbed/mbed-os #10785: Bad printf formatting prevents from treating formatting warning as errors

LaurentLouf, 2019-06-07:
## Description

I created some custom build profiles where I added to treat all warnings related to formatting as errors, to avoid long debug sessions only due to a bad format specifier. Problem is, for STM targets, it fails due to at least the following formatting mistakes (still present in current `master` branch as of writing) : 

https://github.com/ARMmbed/mbed-os/blob/489bd10626352683a4568a7d8d1cfbf8c3134c72/targets/TARGET_STM/can_api.c#L746

```
[ERROR] ./mbed-os/targets/TARGET_STM/can_api.c: In function 'can_frequency':
./mbed-os/targets/TARGET_STM/can_api.c:257:38: error: format '%x' expects argument of type 'unsigned int', but argument 2 has type 'long unsigned int' [-Werror=format=]
                 error("can ESR  0x%04x.%04x + timeout status %d", (can->ESR & 0xFFFF0000) >> 16, (can->ESR & 0xFFFF), status);
                                   ~~~^                            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                   %04lx
./mbed-os/targets/TARGET_STM/can_api.c:257:43: error: format '%x' expects argument of type 'unsigned int', but argument 3 has type 'long unsigned int' [-Werror=format=]
                 error("can ESR  0x%04x.%04x + timeout status %d", (can->ESR & 0xFFFF0000) >> 16, (can->ESR & 0xFFFF), status);
                                        ~~~^                                                      ~~~~~~~~~~~~~~~~~~~
                                        %04lx
```

https://github.com/ARMmbed/mbed-os/blob/489bd10626352683a4568a7d8d1cfbf8c3134c72/targets/TARGET_STM/gpio_irq_api.c#L128

```
ERROR] ./mbed-os/targets/TARGET_STM/gpio_irq_api.c: In function 'handle_interrupt_in':
./mbed-os/targets/TARGET_STM/gpio_irq_api.c:125:50: error: format '%d' expects argument of type 'int', but argument 2 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
     error("Unexpected Spurious interrupt, index %d\r\n", irq_index);
                                                 ~^       ~~~~~~~~~
                                                 %ld
```

It would be nice to add this flag `Wformat=error` in the CI to make sure that users will be able to use this for their own code without having to first fix all formatting errors in mbed-os.

## Issue request type

[ ] Question
[ ] Enhancement
[X ] Bug

0xc0170, 2019-06-07:
> It would be nice to add this flag Wformat=error in the CI to make sure that users will be able to use this for their own code without having to first fix all formatting errors in mbed-os.

👍 
@ARMmbed/mbed-os-core Please review

To the errors:

@ARMmbed/team-st-mcd Can you review these

jeromecoutant, 2019-06-07:
Hi
About warning, I asked yesterday to @kjbracey-arm some advice to remove them !
It should come soon in a PR

LaurentLouf, 2019-06-10:
PR  #10793 should fix this issue, closing the issue.

jeromecoutant, 2019-06-10:
#10793 solves the compilation warnings,
but I thought that issue was raised in order to ask ARM to add Wformat=error flag in the CI?

LaurentLouf, 2019-06-10:
Indeed, I forgot about this part

closed 2020-07-16 by ciarmcom