ARMmbed/mbed-os #10799: USB MSD Filesystem Access Guidelines?

AGlass0fMilk, 2019-06-10:
### Description

@maciejbocianski 
What are the requirements of accessing a block device underlying a USB MSD instance that is connected to a host? How should one avoid conflicts in this situation?

I am using a FAT file system and a QSPI block device. I want to simply see when a file is placed on the MSD. I looked at the MSD tests for some inspiration but haven't been successful yet.

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

maciejbocianski, 2019-06-10:
The whole storage init is done in `storage_init` you have to only change the block device to `QSPIFBlockDevice`. This example shows how to use qspi block device https://github.com/ARMmbed/mbed-os-example-blockdevice-qspif
https://github.com/ARMmbed/mbed-os/blob/e7bc177b20427197763e1cd47261b04bf18db64e/TESTS/usb_device/msd/main.cpp#L251-L260

AGlass0fMilk, 2019-06-10:
Are there any limitations on concurrent access? How should that be handled?

AGlass0fMilk, 2019-06-10:
@maciejbocianski I think you misunderstood what I was asking. I have a USB MSD working with on-board QSPI using the QSPIFBlockDevice and FATFileSystem. I've been having trouble reading the filesystem (checking if files exist) while it's mounted on a host over USB. 

What are guidelines for concurrently accessing the same file system mounted on a host over USB? I noticed this:

https://github.com/ARMmbed/mbed-os/blob/e7bc177b20427197763e1cd47261b04bf18db64e/TESTS/usb_device/msd/main.cpp#L214-L225

Is that necessary to prevent conflicts with the USB MSD driver?

desowin, 2019-06-11:
I don't know about mbed implementation, but in general you want to avoid using the filesystem at the same time it is exposed to USB Host via USB MSD. Doing so is asking for trouble, especially that the USB Host (eg. Windows) can cache the sectors *and* there really isn't a proper way to invalidate such cache/prevent the host from doing so.

If you really want to access the files at the same time as mounted, then the MTP protocol could be used instead of the MTP. Using MTP would also make it possible to expose the LittleFS to the host.

maciejbocianski, 2019-06-11:
I agree with @desowin.
Actually the safest way is to unmount the MSD device then change its content and mount again

AGlass0fMilk, 2019-06-11:
@desowin I will check out MTP. Does the device still show up as a folder in a typical host system (windows, Mac, Linux, etc)?

desowin, 2019-06-11:
MTP is not implemented in mbed-os, so you'd need to implement that to be able to use it.

On Windows the MTP device shows normally as a drive under "This Computer" and you can access the directory structure. On Linux you can use the commandline tools to access the files or mount the filesystem or use tools like gMTP. On Mac, I don't know.

AGlass0fMilk, 2019-06-11:
I'm looking for functionality like DAPLink, where the device can detect when a file has been copied to it and then automatically disconnect itself. I believe DAPLink uses the MSD class so I will look into how they do it. Thanks for the insights!

closed 2019-06-11 by AGlass0fMilk