ARMmbed/mbed-os #10803: rtc_reset test error with Python 3.7

amak-cy, 2019-06-10:
### Description

rtc_reset throw python error and fails with strange error message.

Tried with Cypress targets CY8PROTO_062_4343W and CY8CKIT_062_WIFI_BT (ARM_GCC)
Used mbed-os 5.11.5, 5.12.0, 5.12.1, 5.12.2, 5.12.3 (Releases are downloaded from here https://os.mbed.com/mbed-os/releases/#)

 ### Steps to reproduce

1. Run rtc_reset greentea test with Python 3.7 on CY8PROTO_062_4343W or CY8CKIT_062_WIFI_BT
device. Info from logs:
[1559743506.20][HTST][INF] ==== Traceback start ====
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/mbed_os_tools/test/host_tests_runner/host_test_default.py", line 306, in run_test
    self.test_supervisor.setup()
  File "/Users/fbtest/mbed-os-ah-sec-release-2-RC4_p43/TESTS/host_tests/rtc_reset.py", line 43, in setup
    generator.next()
AttributeError: 'generator' object has no attribute 'next'
[1559743506.20][HTST][INF] ==== Traceback end ====

2. when manually try to run this test with debug session the mbedhtrun shows the following info:
[1560167836.58][HTST][INF] host test executor ver. 0.0.8
[1560167836.58][HTST][INF] copy image onto target... SKIPPED!
[1560167836.59][HTST][INF] starting host test process...
[1560167837.97][CONN][INF] starting connection process...
[1560167837.97][CONN][INF] notify event queue about extra 60 sec timeout for serial port pooling
[1560167837.97][CONN][INF] initializing serial port listener...
[1560167837.97][SERI][INF] serial(port=COM88, baudrate=9600, read_timeout=0.01, write_timeout=5)
[1560167837.97][HTST][INF] setting timeout to: 60 sec
[1560167837.97][SERI][TXD] mbedmbedmbedmbedmbedmbedmbedmbedmbedmbed
[1560167837.97][CONN][INF] sending up to 2 __sync packets (specified with --sync=2)
[1560167837.97][CONN][INF] sending preamble '3e3541f4-d86b-4d13-a2da-53d615ba8fa5'
[1560167837.97][SERI][TXD] {{__sync;3e3541f4-d86b-4d13-a2da-53d615ba8fa5}}
[1560167838.11][CONN][RXD] mbedmbedmbedmbedmbedmbedmbedmbed
[1560167838.15][CONN][INF] found SYNC in stream: {{__sync;3e3541f4-d86b-4d13-a2da-53d615ba8fa5}} it is #0 sent, qu
[1560167838.15][HTST][INF] sync KV found, uuid=3e3541f4-d86b-4d13-a2da-53d615ba8fa5, timestamp=1560167838.151812
[1560167838.17][CONN][INF] found KV pair in stream: {{__version;1.3.0}}, queued...
[1560167838.17][HTST][INF] DUT greentea-client version: 1.3.0
[1560167838.19][CONN][INF] found KV pair in stream: {{__timeout;100}}, queued...
[1560167838.19][HTST][INF] setting timeout to: 100 sec
[1560167838.23][CONN][INF] found KV pair in stream: {{__host_test_name;rtc_reset}}, queued...
[1560167838.23][HTST][ERR] host test not detected: rtc_reset
[1560167838.23][HTST][INF] __exit_event_queue received
[1560167838.23][HTST][INF] test suite run finished after 0.03 sec...
[1560167838.24][CONN][INF] found KV pair in stream: {{start;1}}, queued...
[1560167838.24][CONN][INF] received special event '__host_test_finished' value='True', finishing
[1560167838.30][HTST][INF] CONN exited with code: 0
[1560167838.30][HTST][INF] Some events in queue
**[1560167838.30][HTST][WRN] >>> orphan event: {{start;1}}, timestamp=1560167838.237564**
[1560167838.30][HTST][INF] stopped consuming events
[1560167838.30][HTST][INF] host test result() call skipped, received: error
[1560167838.30][HTST][WRN] missing __exit event from DUT
[1560167838.30][HTST][INF] calling blocking teardown()
[1560167838.30][HTST][INF] teardown() finished
[1560167838.30][HTST][INF] {{result;error}}

3. Observe unrelated to test result errors.

Is this expected? 

    [ X] Question
    [ ] Enhancement
    [ X] Bug

Thanks

0xc0170, 2019-06-11:
@amak-cy This should be fixed on master and will be in 5.13.0.

amak-cy, 2019-06-11:
Sounds good. 

Will try to pull from master and try on my side.

Thanks.

amak-cy, 2019-06-11:
Confirmed with the latest from master the rtc_reset test is passing with Python 3.7

BTW my college also fixed this test on his desk:
diff --git a/TESTS/host_tests/rtc_reset.py b/TESTS/host_tests/rtc_reset.py
index 1ea8b373c3..3412975a8e 100644
--- a/TESTS/host_tests/rtc_reset.py
+++ b/TESTS/host_tests/rtc_reset.py
@@ -40,7 +40,7 @@ class RtcResetTest(BaseHostTest):
         """Register callbacks required for the test"""
         self._error = False
         generator = self.rtc_reset_test()
-        generator.next()
+        next(generator)

         def run_gen(key, value, time):
             """Run the generator, and fail testing if the iterator stops"""

Thanks

closed 2019-06-11 by amak-cy