ARMmbed/mbed-os #10806: How to configure USB pins for custom target based on STM32F746?

desowin, 2019-06-11:
### Description

I want to create custom target based on STM32F746 discovery, however I want to be able to override clock configuration. Thus I no longer want to inherit the DISCO_F746NG to be able to provide my own pin names and clock configuration.

Initially I had
```
{
    "MY_TARGET": {
        "inherits": [
            "DISCO_F746NG"
        ],
        "OUTPUT_EXT": "hex"
    }
}
```
in my `custom_targets.json`. As the clock configuration cannot be altered via configuration options, I decided to modify the `custom_targets.json` to contain copy of the definitions from `targets.json`:
```
{
    "MY_TARGET": {
        "inherits": ["FAMILY_STM32"],
        "core": "Cortex-M7F",
        "extra_labels_add": [
            "STM32F7",
            "STM32F746",
            "STM32F746xG",
            "STM32F746NG",
            "STM_EMAC"
        ],
        "components_add": ["QSPIF", "FLASHIAP"],
        "supported_form_factors": ["ARDUINO"],
        "config": {
            "clock_source": {
                "help": "Mask value : USE_PLL_HSE_EXTC | USE_PLL_HSE_XTAL | USE_PLL_HSI",
                "value": "USE_PLL_HSE_XTAL|USE_PLL_HSI",
                "macro_name": "CLOCK_SOURCE"
            },
            "usb_speed": {
                "help": "Select the USB speed/connector (0=FullSpeed, 1=HighSpeed)",
                "value": "1"
            },
            "lpticker_lptim": {
                "help": "This target supports LPTIM. Set value 1 to use LPTIM for LPTICKER, or 0 to use RTC wakeup timer",
                "value": 1
            }
        },
        "detect_code": ["0815"],
        "macros_add": [
            "MBED_TICKLESS",
            "USB_STM_HAL",
            "USBHOST_OTHER"
        ],
        "device_has_add": [
            "ANALOGOUT",
            "CAN",
            "EMAC",
            "SERIAL_ASYNCH",
            "TRNG",
            "FLASH",
            "QSPI",
            "USBDEVICE",
            "MPU"
        ],
        "release_versions": ["2", "5"],
        "device_name": "STM32F746NG",
        "bootloader_supported": true,
        "overrides": {
            "lpticker_delay_ticks": 4,
            "network-default-interface-type": "ETHERNET"
        },
        "OUTPUT_EXT": "hex"
    }
}
```

When I tried to compile the project using such custom target, the compilation did fail at:
```
[Error] USBPhy_STM32.cpp@296,2: #error "USB pins are not configured !"
```

There doesn't seem to be any mechanism to configure the USB pins.

What is the correct way to handle the problem? I can think about placing modified USBPhy_STM32.cpp in my project and adding `mbed-os/usb/device/targets/TARGET_STM/USBPhy_STM32.cpp` to `.mbedignore` but it doesn't sound like a proper solution.

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

jeromecoutant, 2019-06-11:
Hi
If you don't need USB, remove USBDEVICE from your custom_targets.json ?

desowin, 2019-06-11:
> If you don't need USB, remove USBDEVICE from your custom_targets.json ?

What if the USB is needed?

jeromecoutant, 2019-06-11:
OK, if you need USB, you have to update USBPhy_STM32.cpp file,
and find the correct configuration

Maybe just add some
|| defined(TARGET_MY_TARGET)

after the  defined(TARGET_DISCO_F746NG_OTG_HS) line ?

desowin, 2019-06-11:
@jeromecoutant  What you write is pretty much what I described as "placing modified USBPhy_STM32.cpp in my project". Ideal solution would be to be able to do all this without modyfying mbed-os directory contents.

jeromecoutant, 2019-06-11:
> Ideal solution would be to be able to do all this without modyfying mbed-os directory contents.

I agree...

jeromecoutant, 2019-10-30:
#11675 should help
Regards,
Jerome

PR #11675 (2019-10-28): STM32 USB update step 1
### Description

Hi

Goal is to make USB Device integration easy for each STM32 targets.

Step 1 (this PR):
- USB pins are declared for all current supported ST boards
even if USB_DEVICE feature is not supported yet
even is the board doesn't have any connector
- So, no more need to manually hardcode pins value for each target 
This fixes #10806 
Only addition of USB_DEVICE in the local mbed_app.json file is necessary
- removes compilation warning
This fixes #10822
- this should work for every targets supporting OTG and with full speed connector
**So targets, which was s [... cut]