ARMmbed/mbed-os #10814: STM32F439Zi run into Hardfault if compiled with ARMC5 in debug mode

aqib-ublox, 2019-06-12:
### Description

A code compiled using GCC_ARM and IAR in debug profile runs fine without any issue however with ARMCC it ran into hardfault. 

[image]
[image]

### Targets tested:
NUCLEO_F439ZI
UBLOX_EVK_ODIN_W2

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

aqib-ublox, 2019-06-12:
@0xc0170 can u look into this?

evedon, 2019-06-19:
Thanks for reporting the issue. We will investigate today.

michalpasztamobica, 2019-06-19:
Hello @aqib-ublox , would you please let us know which code revision is required to reproduce the issue?
I tried with latest [mbed-os-example-blinky](https://github.com/ARMmbed/mbed-os-example-blinky/commit/880ed00bff6352cea56031d34b9d891ba21059e9) which is using 0063e5de32fc575f061244c96ac60c41c07bd2e6 revision of mbed-os. Then I also checked out latest master (rev. 14b77c9a2e51acbfa7c675e680a516f89696f22c). I compiled and flashed with the following command:
```
mbed compile -t ARMC5 -m UBLOX_EVK_ODIN_W2 --app-config=mbed_app.wlan0_ARM21.json --profile debug --flash
```
I am using ARMC5 compiler version `ARM Compiler 5.06 update 3 (build 300)`

I can see the device is blinking LED at a given rate and I can see frequent serial outputs coming fine. 
Do I need to wait to see the crash or should it occur right away?
Do you have any local changes?

aqib-ublox, 2019-06-19:
@michalpasztamobica can u try to debug a code if you put break point in main() it never get hits and it crashes while debugging

michalpasztamobica, 2019-06-19:
@aqib-ublox , how did you set up your eclipse? 
I tried the most obvious way, but it seems that UBLOX is not supported:
```
$ mbed export -i eclipse_armc5 -m UBLOX_EVK_ODIN_W2 --profile debug
UBLOX_EVK_ODIN_W2 not supported by eclipse_armc5
```
I also don't see stm32f439xl among the supported versions in pyocd-gdbserver.

Have you tried following guidelines from @0xc0170 in mentioned in #9975 ?

To my best understanding the best way to move forward is to try the mbed studio. Have you tried that?

aqib-ublox, 2019-06-19:
> Have you tried following guidelines from @0xc0170 in mentioned in #9975 ?

Yes i tried with eclipse and OPenOcd.

> To my best understanding the best way to move forward is to try the mbed studio

please share a link to perform working with mbed studio.

aqib-ublox, 2019-06-19:
> I tried the most obvious way, but it seems that UBLOX is not supported:

```
$ mbed export -i eclipse_armc5 -m UBLOX_EVK_ODIN_W2 --profile debug
UBLOX_EVK_ODIN_W2 not supported by eclipse_armc5
```
but if you try above with **stm32f429xl** it would say it is not supported but i was able to get it debug thats strange

michalpasztamobica, 2019-06-19:
Not sure if there is any instructions page, but what I did was quite trivial:
1) Go to https://os.mbed.com/studio/ download and install mbed studio with a double click
2) File -> New Program... -> choose blinky from the example programs list
3) Click build and wait (I can see that ARMC6 is used by default. Not sure how to change this to ARMC5 - is this important for you, @aqib-ublox ?)
4) Click Debug

After that the program was stopped at main entry point, I could step through it or play and pause the program without crashes or errors.
Please, let us know if this worked for you.

michalpasztamobica, 2019-06-19:
@aqib-ublox , I found that mbed studio does not support ARMC5, sorry. 
I hope the newer ARMC6 works fine for you, just like it worked for me?

aqib-ublox, 2019-06-20:
@michalpasztamobica it is useful but i need support against ARMC5 and I have also experienced multiple times even if you don't export your project for particular IDE but you compiled your project with debug profile it can be added to eclipse and debugging code work fine without any issue.

So just a side note you can try this with UBLOX_EVK_ODIN_W2 as well and try putting break point in vector table or at system reset sequence it must get hit but point where it tries to load main address and try branching to main system crashes. To be it seems like symbol resolution problem.

michalpasztamobica, 2019-06-21:
@aqib-ublox , I see your point. It seems however that Eclipse doesn't have official support from mbed side, especially for an older version of ARMC5.

Following advice from @evedon I tried to use the official tools - Keil uvision5. The `mbed cli` exporter refused to export a uvision5 project for UBLOX and adjusting uvision6 export was quite trouble some. Instead, I run the [mbed online compiler](https://ide.mbed.com), imported the blinky example for UBLOX on-line and there the Export button gave me a zip file, containing a project I could successfully load and compile in Keil uvision5. 
I did come across a compilation issue as I was missing linker script - solution is [here](https://os.mbed.com/questions/82654/Keil-uVision-compilation-problem/).
Once compiled I also had to [downgrade DFP for STM32F4xx to v 2.10](http://www.keil.com/forum/63409/keil-stm32f4xx-dfp-v2-11-0-to-v2-12-0-upgrade-issue/), make sure that the [programming is done under reset](http://www.keil.com/support/docs/4082.htm) and [configure the Programming algorithm](http://www.keil.com/support/docs/2831.htm).
After that I managed to get the debugger running with an armc5-compiled program.

Breakpoints and instruction stepping work fine.

@aqib-ublox , I suggest that you try this approach, if ARMC5 is important for you.
If you really want to use Eclipse, you may try getting an export from the on-line compiler, I didn't manage to get it working, but perhaps if you check it against your set-up you will see the possible problem.

evedon, 2019-06-24:
@aqib-ublox I hope that you find the information from @michalpasztamobica useful. 
As there is no official support in Eclipse for mbed and we are not able to reproduce the issue using the mbed online compiler with ARMC5, please consider contributing if you find the cause of the issue that you reported. 
I will close this issue for now as we can not offer any more support for the Eclipse environment.

closed 2019-06-24 by evedon